espressif/esp-idf #6167: esp_http_client use results in stack overflow when sending simple text data to the server

rjam-xav, 2020-11-26:
Hello, I am trying to post a simple query to the server but there is always stack overflow every time a call to esp_err_t err = esp_http_client_perform(client); is made.

The log:
> I (3077) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
> I (3077) HTTP_CLIENT: Last esp error code: 0x103
> I (3080) HTTP_CLIENT: Last mbedtls failure: 0x0
> I (3085) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
> I (3090) HTTP_CLIENT: Last esp error code: 0x103
> I (3096) HTTP_CLIENT: Last mbedtls failure: 0x0
> ***ERROR*** A stack overflow in task main has bee

The code:

>   esp_http_client_config_t config = {
>         .host = "127.0.0.1",
>         .path = "/api/cognitive/voice_interaction_wav",
>         .port=8000,
>         .query = "POST",
>         .event_handler = _http_event_handler,
>         .user_data = local_response_buffer,        // Pass address of local buffer to get response
>     };
>     esp_http_client_handle_t client = esp_http_client_init(&config);
> 
>     // POST
>     ESP_LOGW(PTAG,"configuered");
> 
>     
>     const char* post_data = "{\"voiceFile\":\"UklGRiRwDQBXQVZFZm10IBAAAAABAAIARKwAABCxA\"}";
>     esp_http_client_set_url(client, "https://centralindia.stt.speech.microsoft.com/speech/recognition/conversation/cognitiveservices/v1?language=en-US");
>     esp_http_client_set_method(client, HTTP_METHOD_POST);
>     esp_http_client_set_header(client, "Content-Type", "audio/wav; codecs=audio/pcm; samplerate=16000");
>     esp_http_client_set_header(client, "Connection", "keep-alive");
>     esp_http_client_set_header(client, "Accept", "application/json");
>     esp_http_client_set_header(client, "Expect", "100-continue");
>     esp_http_client_set_post_field(client, post_data, strlen(post_data));
>     esp_err_t err = esp_http_client_perform(client);
>     if (err == ESP_OK) {
>         ESP_LOGI(PTAG, "HTTP POST Status = %d, content_length = %d",
>             esp_http_client_get_status_code(client),
>             esp_http_client_get_content_length(client));
>     }
>     else {
>         ESP_LOGE(PTAG, "HTTP POST request failed: %s", esp_err_to_name(err));
>     }
>     ESP_LOG_BUFFER_CHAR(PTAG, local_response_buffer, strlen(local_response_buffer));
>

Alvin1Zhang, 2020-11-27:
Thanks for reporting, we will look into.

mahavirj, 2020-11-27:
@rjam-xav 

Please post entire sample code along with `sdkconfig`. Please see github issue template for more information.

pdemianczuk07, 2020-11-27:
You can set buffor for transmision in config structure
esp_http_client_config_t config = {
...
.buffer_size = XXZ, //size in Bytes
.buffer_size_tx = YYZ,
};
buffer is set by default on #define DEFAULT_HTTP_BUF_SIZE (512)

if you sure that problem is according to stack, you might try menuconfig
Menu path: (Top) -> Component config -> LWIP    
Name: LWIP_TCPIP_TASK_STACK_SIZE                                                                                                                                      
Prompt: TCP/IP Task Stack Size                                                                                                                                        
Type: int                                                                                                                                                             
Value: 3072  
but I don't have that is this option.

shubhamkulkarni97, 2020-11-30:
@rjam-xav, From the logs it seems esp_http_client_perform() is called from main task. Stack size for main task is 3584 by default, which is not enough for https request. To perform https request, it is recommended to have stack size of 8192 bytes.

You can refer to [esp_http_client example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/esp_http_client), where requests are performed from http_test_task with stack size of 8192.

rjam-xav, 2020-12-01:
Increasing the stack size significantly to 8192 did indeed work. thank you.

closed 2020-12-01 by rjam-xav