espressif/esp-idf #6169: Unexpected system restart when applying network traffic

pyadvichuk, 2020-11-26:
## Environment

- Development Kit:      [none]
- Kit version:
- Module or chip used:  [ESP32-WROVER-B]
- IDF version: v4.2-beta1-332-g511965b26
- Build System:         [idf.py]
- Compiler version:  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- environment type:
- Using an IDE?: [docker based build,  image - espressif/idf:release-v4.2]
- Power Supply:         [external 3.3V]

## Problem Description
Unexpected system restart when applying network traffic (wifi or ethernet no difference)

//Detailed problem description goes here.

small custom application including wifi and ethernet setup + httpd-server.
httpd-server contains single REST handler - get for main.js.
The 'js' file itself is included into project as a byte array.

during tests, I've applied continuous network load including
- 5..8 threads simple ping
- 5-10 threads UPD packets on different ports (used and unused)
- 3.6 threads   curl get main.js

after an unpredictable time, the system restarts with followed messages (no any other more or less related to)

```
.........
ASSERT_PARAM(-218959118 0), in arch_main.c at line 343
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
.........
```
also tried to run such application/load with current stable release (4.1) but observed memory leak somewhere inside wifi-lwip-http-server.

### Expected Behavior
the system is alive, probably lost some requests

### Actual Behavior
the system unexpectedly restarted at a random point of time

### Steps to reproduce

1. apply continuous network traffic like it was described above
2. wait

### Code to reproduce this issue

minimal application for issue demonstrating
https://github.com/pyadvichuk/esp32-leak-net-httpd

in /qa directory attached simple traffic generator.

## Debug Logs

.........
ASSERT_PARAM(-218959118 0), in arch_main.c at line 343
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
.........
```

## Other items if possible

https://github.com/pyadvichuk/esp32-leak-net-httpd

Alvin1Zhang, 2020-11-27:
Thanks for reporting, we will look into.

david-cermak, 2020-11-30:
Hi @pyadvichuk 

this means that some really serious issue happened, since the interrupt watchdog triggered reset without the panic handler to process it. Could you please also share the logs few lines before and few lines after the 
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
```
Specifically the later, as after the reset, it should print the last address of the PC, when the WDT triggered. This could help giving some pointer for debugging.

Please also share your ESP32 revision (printed in startup). I've been trying to reproduce the issue with your project and the shell script, but no luck, so far.

pyadvichuk, 2020-11-30:
```
I (5014981) UPDATE: Writing to partition subtype 16 at offset 0x40000
ASSERT_PARAM(-218959118 0), in arch_main.c at line 343
ets Jun  8 2016 00:22:57

 

rst:0x8 (TG1WDT_SYS_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6904
load:0x40078000,len:13976
load:0x40080400,len:4052
0x40080400: _init at ??:?

 

entry 0x40080688
I (172) boot: ESP-IDF v4.1-273-g94cb7e8b8 2nd stage bootloader
I (172) boot: compile time 13:37:32
I (173) boot: chip revision: 1
I (177) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (184) boot.esp32: SPI Speed      : 40MHz
I (189) boot.esp32: SPI Mode       : DIO
I (194) boot.esp32: SPI Flash Size : 4MB
W (198) boot.esp32: PRO CPU has been reset by WDT.
W (204) boot.esp32: WDT reset info: PRO CPU PC=0x4008d99d
0x4008d99d: coex_core_release at ??:?

 

W (210) boot.esp32: WDT reset info: APP CPU PC=0x400979f9
0x400979f9: xTaskResumeAll at C:/Users/ukr.asolomentsev/esp/esp4/components/freertos/tasks.c:2298

 

I (216) boot: Enabling RNG early entropy source...
I (222) boot: Partition Table:
I (225) boot: ## Label            Usage          Type ST Offset   Length
I (233) boot:  0 nvs              WiFi data        01 02 00010000 00006000
I (240) boot:  1 phy_init         RF data          01 01 00016000 00001000
I (248) boot:  2 varkeys          NVS keys         01 04 00017000 00001000
I (255) boot:  3 var0             WiFi data        01 02 00018000 00012000
I (263) boot:  4 var1             WiFi data        01 02 0002a000 00012000
I (270) boot:  5 otadata          OTA data         01 00 0003c000 00002000
I (278) boot:  6 ota_0            OTA app          00 10 00040000 001e0000
I (285) boot:  7 ota_1            OTA app          00 11 00220000 001e0000
I (293) boot: End of partition table
I (297) boot_comm: chip revision: 1, min. application chip revision: 0
```

one more restart observed from our QA team. the first line in log - from our application. 
This restart is reproducible, each time when start writing flash during update procedure.

david-cermak, 2020-12-01:
@pyadvichuk Thanks for sharing the logs!
```
W (204) boot.esp32: WDT reset info: PRO CPU PC=0x4008d99d
0x4008d99d: coex_core_release at ??:?
```
This indicates the issue is coming somewhere from the BT/WiFi coexistence logic. Interestingly the application linked doesn't use BT at all, correct? Nor it uses OTA. Or is the app only supposed to reproduce memory leak issue? In that case, i would suggest raising a separate issue.

> This restart is reproducible, each time when start writing flash during update procedure.

This is a perfect, but would need steps to reproduce this and possibly also the application, as it seems the linked project was meant to check memory issues only.

With that said, might I ask you to track only the memory leaks in this issue and submit a new issue with all the steps to reproduce and the app possibly to track the crashes in the coexistence? resulting in the WDT resets, such as
```
ASSERT_PARAM(-218959118 0),
```

pyadvichuk, 2020-12-20:
it can be closed as I not able to reproduce it on 4.2rc and higher

Alvin1Zhang, 2020-12-21:
Thanks for sharing updates.

closed 2020-12-21 by Alvin1Zhang