espressif/esp-idf #6170: btc config parse crash after flash_erase

fariouche, 2020-11-26:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): ESP32-D0WDQ6 (revision 1)
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-1720-g494a124d9
- Build System:         [Make]
- Power Supply:         [USB||external 3.3V]

## Problem Description
I've changed the partitions, after I've done a make erase_flash followed by a make flash
After that, the wifi is working, but bluetooth BLE is not, it is crashing inside btc_config because it cannot find the config in the nvs.

At first I was thinking that the issue was a faulty power supply. I'm now using a 3.3v external power supply 3A, and still crashing.

//Detailed problem description goes here.

The backtrace is the following:
0x400fc329: heap_caps_alloc_failed at esp-idf/components/heap/heap_caps.c:63
0x400851e7: heap_caps_malloc at esp-idf/components/heap/heap_caps.c:155
0x40085201: heap_caps_malloc_default at esp-idf/components/heap/heap_caps.c:177
0x40095f10: _calloc_r at esp-idf/components/newlib/heap.c:72
0x40095f34: calloc at esp-idf/components/newlib/heap.c:36
0x4014cf1f: config_parse at esp-idf/components/bt/common/osi/config.c:557
0x4014d1e8: config_new atesp-idf/components/bt/common/osi/config.c:102 (discriminator 3)
0x4012ad08: btc_config_init at esp-idf/components/bt/host/bluedroid/btc/core/btc_config.c:74
0x4012a34f: btc_init_bluetooth at esp-idf/components/bt/host/bluedroid/btc/core/btc_main.c:66
0x4012a3f8: btc_main_call_handler atesp-idf/components/bt/host/bluedroid/btc/core/btc_main.c:114
0x4012a205: btc_thread_handler at esp-idf/components/bt/common/btc/core/btc_task.c:177
0x4014e1f5: osi_thread_run at esp-idf/components/bt/common/osi/thread.c:66
0x4009174d: vPortTaskWrapper at esp-idf/components/freertos/xtensa/port.c:170

The reason is that the config does not exist, so get_config_size_from_flash() returns zero.
then there is a malloc(zero)... and this is the reason of the crash.

Here is a possible patch: just defer the allocation just after having checked that the size is not zero.
Bluetooth is again working after that.

diff --git a/components/bt/common/osi/config.c b/components/bt/common/osi/config.c
index b3b881603..e11aef8e1 100644
--- a/components/bt/common/osi/config.c
+++ b/components/bt/common/osi/config.c
@@ -550,10 +550,11 @@ static void config_parse(nvs_handle_t fp, config_t *config)
     const size_t keyname_bufsz = sizeof(CONFIG_KEY) + 5 + 1; // including log10(sizeof(i))
     char *keyname = osi_calloc(keyname_bufsz);
     int buf_size = get_config_size_from_flash(fp);
-    char *buf = osi_calloc(buf_size);
+    char *buf = NULL;
     if(buf_size == 0) { //First use nvs
         goto error;
     }
+    buf = osi_calloc(buf_size);
     if (!line || !section || !buf || !keyname) {
         err_code |= 0x01;
         goto error;

Hope this help

Regards

Alvin1Zhang, 2020-11-27:
Thanks for reporting, we will look into.

fariouche, 2020-11-27:
I haven't seen that the patch I've included in the comment got corrupted!
Here is a proper patch file, more readable

[esp32-btconfig.patch.txt](https://github.com/espressif/esp-idf/files/5606539/esp32-btconfig.patch.txt)

xulongzhuang, 2021-06-22:
hi @fariouche 
I can't reproduce the problem you encountered. What changes have been made to the contents of your partition two times before and after? Can you provide a detailed step or demo that can reproduce the problem, such as using our demo to make changes, thanks

fariouche, 2021-07-15:
Hello.
I've done nothing special... juste erase flash and start my application that init wifi and BLE.... and it crashed in BLE init.
The reason is that get_config_size_from_flash(fp) is returning zero (why, I don't know, maybe a race condition?), so the osi_calloc(buf_size) is failling...

So adding a condition to not do the alloc if buf_size is zero solved the problem

closed 2022-01-07 by espressif-bot