espressif/esp-idf #6171: Failed to get the IP address

rony91jp, 2020-11-27:
Hello,

I am currently using ESP32-POE board and tried to build an Ethernet Connection.
Following is the result.
[image]
Below is the example configuration& ethernet configuration
[image]
[image]

Thank you

suda-morris, 2020-11-27:
RMII clock output mode is not always work on esp32, we have two GPIO (16 and 17, they're inverted), in my experience, if GPIO17 can't work, then GPIO16 could work. Many causes can lead to such issues, like PCB trace (remember RMII clock is 50MHz) length.
I can't reproduce your issue with my LAN8720 breakout board and some wire jumpers. Maybe you should go to the [forum](https://esp32.com/index.php) to ask for help.

ryankurte, 2020-11-29:
we're seeing a similar issue with a custom board + LAN8710, the MAC comes up, connect / disconnect events work, and data is received / transmitted, but, we never receive an IP. 
setting the debug level to `VERBOSE` shows incoming ICMP packets that appear to be corrupt (MAC addresses that do not exist on the network etc.) so we're working on the basis this is an RMII / clock signal integrity issue. do y'all have any advice for further diagnosing this?

rony-91, 2020-11-29:
Hello,

Regarding the above issue, I was able to get the static IP address using the Arduino ethernet example. I used the below command to assign my static IP.
ETH.begin();

ETH.config(IPAddress(192, 168, 1, 101),IPAddress(192, 168, 1, 1),IPAddress(255, 255, 255, 0),IPAddress(192, 168, 1, 1), IPAddress(192, 168, 1, 1));

Output:
image

My question is, I want to disable the DHCP and want to assign the static IP above the same way in the esp-idf example. Where should I change in the esp-idf ethernet basic example program?

Thank you

suda-morris, 2020-11-30:
@rony-91 
To set static IP address, firstly you need to stop the dhcpc by `esp_netif_dhcpc_stop`, then set IP into by `esp_netif_set_ip_info`

More info, please refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html

rony91jp, 2020-11-30:
@suda-morris 

I need your little help. Why i did not get the static ip from below program?

#include <string.h>
#include "sdkconfig.h"
#include "esp_event.h"
#include "esp_wifi.h"
#include "esp_wifi_default.h"

#if CONFIG_EXAMPLE_CONNECT_ETHERNET
#include "esp_eth.h"
#endif

#include "esp_log.h"
#include "driver/gpio.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "lwip/err.h"
#include "lwip/sys.h"

#include "ethernet_connect.h"

#define GOT_IPV4_BIT BIT(4)
#define GOT_IPV6_BIT BIT(1)
#define CONNECTED_BITS (GOT_IPV4_BIT)

static EventGroupHandle_t s_connect_event_group;
static esp_ip4_addr_t s_ip_addr;
static const char *s_connection_name;
static esp_netif_t *s_example_esp_netif = NULL;

static const char *TAG = "online_connection";
static void start(void);
static void stop(void);

static void on_got_ip(void *arg, esp_event_base_t event_base,
                      int32_t event_id, void *event_data)
{

    ESP_LOGI(TAG, "Got IP event!");
    ip_event_got_ip_t *event = (ip_event_got_ip_t *)event_data;
    memcpy(&s_ip_addr, &event->ip_info.ip, sizeof(s_ip_addr));
    xEventGroupSetBits(s_connect_event_group, GOT_IPV4_BIT);
}

esp_err_t example_connect(void)
{
    if (s_connect_event_group != NULL)
    {
        return ESP_ERR_INVALID_STATE;
    }

    s_connect_event_group = xEventGroupCreate();
    start();
    ESP_ERROR_CHECK(esp_register_shutdown_handler(&stop));
    ESP_LOGI(TAG, "Waiting for IP");
    ESP_LOGI(TAG, "Connected to %s", s_connection_name);
    ESP_LOGI(TAG, "IPv4 address: " IPSTR, IP2STR(&s_ip_addr));
    xEventGroupWaitBits(s_connect_event_group, CONNECTED_BITS, true, true, portMAX_DELAY);
    ESP_LOGI(TAG, "IPv4 address: " IPSTR, IP2STR(&s_ip_addr));
    return ESP_OK;
}

esp_err_t example_disconnect(void)
{
    if (s_connect_event_group == NULL) {
        return ESP_ERR_INVALID_STATE;
    }
[... 109 lines cut ...]
 * @brief Configure Wi-Fi or Ethernet, connect, wait for IP
 *
 * This all-in-one helper function is used in protocols examples to
 * reduce the amount of boilerplate in the example.
 *
 * It is not intended to be used in real world applications.
 * See examples under examples/wifi/getting_started/ and examples/ethernet/
 * for more complete Wi-Fi or Ethernet initialization code.
 *
 * Read "Establishing Wi-Fi or Ethernet Connection" section in
 * examples/protocols/README.md for more information about this function.
 *
 * @return ESP_OK on successful connection
 */
esp_err_t example_connect(void);

/**
 * Counterpart to example_connect, de-initializes Wi-Fi or Ethernet
 */
esp_err_t example_disconnect(void);
esp_netif_t *get_example_netif(void);

#ifdef __cplusplus
}
#endif

rony91jp, 2020-12-01:
The above problem has solved. Thank you all for your support.

closed 2020-12-01 by rony91jp