espressif/esp-idf #6173: esp_netif_destroy is not working

Zeni241, 2020-11-28:
If a user didn't sent correct SSID and PW in first attempt, he resends the credentials and "wifi_init_sta(void)" is executed again.
But the second time core panics. Reason most probably is that the first instance of "esp_netif" is not destroyed.
I know "esp_netif_init()" should be fired only once and if firing second time, it should be destroyed first. But question is how to destroy :?: :?:

Bellow is my code which is not working  

```
esp_netif_t *my_esp_netif;
'
'
'
'
'
if(my_esp_netif){
        esp_netif_destroy(my_esp_netif);
      }
  
   my_esp_netif= esp_netif_init();
   ESP_ERROR_CHECK(esp_event_loop_create_default()); //     This is the line at which core panics.
    esp_netif_create_default_wifi_sta();
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

```

david-cermak, 2020-11-30:
Hi @Zeni241 

Yes, correct, the API [`esp_netif_init()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html#_CPPv414esp_netif_initv) should be called only once, to initialize the TCPIP stack.
Then, the interface itself is created as a new instance using [`esp_netif_new()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html#_CPPv413esp_netif_newPK18esp_netif_config_t).
This instance, a pointer to `esp_netif_t` object, could be then passed to the API  `esp_netif_destroy()`.

Please check the [ESP-NETIF programmer's manual](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html#esp-netif-programmer-s-manual) for more details.

In case of [these Wifi examples](https://github.com/espressif/esp-idf/tree/178b122c145c19e94ac896197a3a4a9d379cd618/examples/wifi/getting_started), another API is used to setup and create a default wifi network interfaces:
``` c
esp_netif_create_default_wifi_sta();
```
which also returns a netif pointer (on success), too, but some examples just ignore it: https://github.com/espressif/esp-idf/blob/178b122/examples/wifi/getting_started/station/main/station_example_main.c#L73
so you can update that code to
``` c
    esp_netif_init();
     ....
    esp_netif_t *my_esp_netif = esp_netif_create_default_wifi_sta();
     ....
    esp_netif_destroy(my_esp_netif);
```
to correctly destoy the created netif

Zeni241, 2020-11-30:
@david-cermak thanks a lot for your reply.
Here is my code now:‎
I call `esp_netif_init();` in `app_main(void)` so that it is called only once.

```
void wifi_init_sta(void)‎
‎{‎
ESP_ERROR_CHECK(esp_event_loop_create_default());  ‎
‎    esp_netif_t *my_esp_netif = esp_netif_create_default_wifi_sta();‎

wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();‎
‎    ESP_ERROR_CHECK(esp_wifi_init(&cfg));‎
‎    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, ‎&wifi_event_handler, NULL));‎
‎    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, ‎&wifi_event_handler, NULL));‎
‎‘‎
‎‘‎
‎‘‎
‎‘‎
ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, ‎&wifi_event_handler));‎
‎    ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, ‎&wifi_event_handler));‎
‎     esp_netif_destroy(my_esp_netif);‎
‎    return;‎
‎}‎
```
It works well for first call. But if I call **void wifi_init_sta(void)‎** second time, the core panic.‎

david-cermak, 2020-11-30:
@Zeni241 Please check the WiFi library documentation, as well:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#api-reference

The same principle applies, please deinit the wifi before calling the init again with [`esp_wifi_deinit()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv415esp_wifi_deinitv)

Zeni241, 2020-11-30:
@david-cermak thanks again. I will go through it thoroughly as soon as get the time. For the time being I have improvised my code as follows and its working fine.

`int wifiCnt=0;`
'
'
'
'
'

```
void wifi_init_sta(void)
{
f(wifiCnt==0){
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_sta();
    wifiCnt=1;
}
   
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &wifi_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &wifi_event_handler, NULL));
'
'
'
'
}
   ```
Please let me know if you see any pitfall in above code.

david-cermak, 2020-12-01:
@Zeni241 There's no problem in initializing certain module only once and reusing it. Moreover it's generally a preferable approach  in embedded environments rather than creating and deleting the objects.

Could you please close the issue, since the `esp_netif_destroy()` apparently works as expected. Also please use the [ESP32-Forum](https://esp32.com/viewforum.php?f=23&sid=3d1e0b4dfaf07164c25e8be78cb892b8) for general questions like this one.

closed 2020-12-01 by Zeni241