espressif/esp-idf #6175: Use a secure element for EAP-TLS

fgervais, 2020-11-29:
**Is your feature request related to a problem? Please describe.**

No

**Describe the solution you'd like**

I'd like to be able to use a secure element (ATECC608A, SE050) to do the EAP-TLS authentication. This would prevent having to add the private key embedded in the firmware.

**Describe alternatives you've considered**

The alternative right now is to do like here:

https://github.com/espressif/esp-idf/blob/master/examples/wifi/wpa2_enterprise/main/wpa2_enterprise_main.c

But in this case the private key is in the firmware and could possibly be extracted and used by a malicious third party to log-in to the wifi network.

**Additional context**

Using a pkcs11 secure element is supported by wpa_supplicant:

```
# OpenSSL Engine support
# These options can be used to load OpenSSL engines in special or legacy
# modes.
# The two engines that are supported currently are shown below:
# They are both from the opensc project (http://www.opensc.org/)
# By default the PKCS#11 engine is loaded if the client_cert or
# private_key option appear to be a PKCS#11 URI, and these options
# should not need to be used explicitly.
# make the opensc engine available
#opensc_engine_path=/usr/lib/opensc/engine_opensc.so
# make the pkcs11 engine available
#pkcs11_engine_path=/usr/lib/opensc/engine_pkcs11.so
# configure the path to the pkcs11 module required by the pkcs11 engine
#pkcs11_module_path=/usr/lib/pkcs11/opensc-pkcs11.so
```

Example to using this feature in linux with the SE050: https://www.nxp.com/docs/en/application-note/AN12661.pdf

Alvin1Zhang, 2020-11-30:
Thanks for raising this feature request.

sagb2015, 2021-04-15:
@fgervais  We will take this feature request but it will take some time to be available in IDF mainline.

Currently we do have support for using ATECC secure element through for  TLS handshake though.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html#atecc608a-secure-element-with-esp-tls

jclab-joseph, 2021-09-11:
This is a good feature request.
I also look forward to the update.

jclab-joseph, 2024-04-29:
https://github.com/espressif/esp-idf/blob/d4cd437ede613fffacc06ac6d6c93a083829022f/components/wpa_supplicant/esp_supplicant/src/crypto/tls_mbedtls.c#L167

I think it would be better to use mbedtls.
Below is an example of integrating atecc806 into mbedtls.

https://github.com/MicrochipTech/mbedtls-examples