espressif/esp-idf #6176: ADC1 Channel 0 conversion not working properly

francis2tm, 2020-11-30:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (v3]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-1720-g494a124d9
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description
Hello,
I suspect it has something to do with the build in hall sensor. I'm trying to use channels 0 as normal adc1 channels (I don't use the hall sensor) and I can't get proper adc conversions like I do in the other channels. The reading keeps oscillating in a high frequency manner between 1.5V and 2.1V (while the correct reading should be a DC constant 1.62V). I use exactly the same firmware to do it and the pin corresponding to channel 0 (GPIO36)  don't have any additional electronics that may affect the reading.
All other channels of ADC1 are working properly (including channel 3 that's also used for the hall sensor).

cardtap, 2020-12-04:
Hi there, I think I am experiencing a similar problem. I am attempting to use GPIO36 as an analog input, and although the voltage on the pin is around 0.8V-2.5V (measured with Fluke multimeter), the ADC always outputs 4095 (3.3V reading) which is false. Have you discovered a fix yet? It's been over two weeks on my post, and I've had no luck getting responses.

francis2tm, 2020-12-04:
Hey @cardtap , nop, not yet. No news from espressif and I still don't know if the problem is solvable, maybe it's an hardware problem.

francis2tm, 2020-12-11:
Hi, I just realized the problem, it was a pcb design error. Sorry for the trouble

closed 2020-12-11 by francis2tm