espressif/esp-idf #6177: Problems building the "coap_server" CoAP example in a mixed Arduino/IDF project.

ameerhazo, 2020-11-30:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (DevKitC): v2
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.3-dev-1720-g494a124d9
- Build System:  CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:   Windows
- (Windows only) environment type: ESP Command Prompt.
- Using an IDE?: VSCode Extension
- Power Supply:  n/a

## Problem Description
Building the "coap_server" example (coap IDF component) works fine in a pure IDF project but returned with errors when built in a mixed Arduino/IDF project.
### Expected Behavior
coap_server example building fine as no configuration of it were altered.
### Actual Behavior
Errors in the lwip header files. Errors in the coap2/str.h header file.
### Steps to reproduce

1. Add Arduino as an ESP-IDF component (https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md)
2. copy the "coap_server" example into the test file. Change the file to .cpp instead of .c because and add extern "C" before void app_main();
3. Build the project.

### Code to reproduce this issue

```cpp
/#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <sys/socket.h>

#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include "freertos/event_groups.h"

#include "esp_log.h"
#include "esp_wifi.h"
#include "esp_event.h"

#include "nvs_flash.h"
//WiFi
#include "protocol_examples_common.h"

#if 1
/* Needed until coap_dtls.h becomes a part of libcoap proper */
#include "libcoap.h"
//#include "coap_dtls.h"
#endif
#include "coap.h"

#include "Arduino.h"
#include "sdkconfig.h"

/*the response handlers are written up here, theyre not affcted*/

//CoAP_Server Task
static void coap_example_server(void *p)
{
    coap_context_t *ctx = NULL;
    coap_address_t serv_addr;
    coap_resource_t *resource = NULL;
   

    snprintf(espressif_data, sizeof(espressif_data), INITIAL_DATA);
    espressif_data_len = strlen(espressif_data);

   

[... 171 lines cut ...]
../../../../components/coap/libcoap/include/coap2/str.h:98:10: note: candidate: 'constexpr libcoap::CoAPStrConst::CoAPStrConst(libcoap::CoAPStrConst&&)'
../../../../components/coap/libcoap/include/coap2/str.h:98:10: note:   candidate expects 1 argument, 2 provided
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
## Possible causes
Compiler doesnt know if the macro "coap_make_str_const" to use is C or C++? I've even used extern "C" on the "coap_example_server" task to test if it would build but the errors still persist.
```
// from C:\Users\USER\Desktop\esp-idf\examples\get-started\sample_project\components\arduino\tools\sdk\esp32\include\coap\libcoap\include\coap2\str.h

#ifdef __cplusplus
namespace libcoap {
  struct CoAPStrConst : coap_str_const_t {
    operator coap_str_const_t *() { return this; }
  };
}
#define coap_make_str_const(CStr)                                       \
  libcoap::CoAPStrConst{sizeof(CStr)-1, reinterpret_cast<const uint8_t *>(CStr)}
#else /* __cplusplus */
#define coap_make_str_const(string)                                     \
  (&(coap_str_const_t){sizeof(string)-1,(const uint8_t *)(string)})
#endif  /* __cplusplus */

```
Any help would be greatly appreciated. Thank you.

me-no-dev, 2020-11-30:
I can help you on the first issue. Problem is that Arduino defines `IPADDR_NONE` as a different value than IDF. Separate the Arduino code in a separate CPP file and export just the needed functions through a header. You could also separate the coap code the same way, then you will not need to include lwip and coap and arduino in the same file and you will not have the issues that you are seeing. Maybe moving the coap code to C file will help you with the other issue.

ameerhazo, 2020-11-30:
Hi @me-no-dev thank you for helping me with the issues. Apparently the first problem from my issue was solved by another Github user who posted about the problem on esp32-arduino Github page ( #4405)
But I'm not sure what he changed to fix the issue as the solution he linked was an issue on a different board? or core?
_

ameerhazo, 2020-11-30:
Issue solved thanks to @me-no-dev :D

closed 2020-11-30 by ameerhazo