espressif/esp-idf #6178: Relase V4.1 ou V4.2-RC memory not being freed, inside a Task with Multicastudp support

filzek, 2020-11-30:
We have found out a huge memory leak in the compiler but we cant explain how to solve or where it crates the caveat to not release the memory by the free call.

We have split all the part outside the SDK and run it in GCC with valgrind for trace the memory leak if it was within any of our code, and the codes runs smootly and perfect without any problem so far, but once the code runs inside the esp-idf allocated in a Task with multicastudp support the FREE simple doesnt seen to work for the code and create a HUGE problem.

Steps taken to check the problem.

Mbded base64 decode
Mbded aes decrypt 256

call 5 inline functions

return to the main task and free the object.

and keep running the task on udp received socket.

This creates the problem and the free not being called.

We will try to trace tomorrow to understand WHERE it is not being free, so, we will call a memory check prior to alloc and after free and will list how deep it occurs.

The intrigating question is that outside the multicast task the same code runs perfect and not a problem to free the it, so only occurs to do not free inside the multicast.

This is an odd behavior.

Valgrind runs smoothly and show no error so far:
==1549== HEAP SUMMARY:
==1549==     in use at exit: 0 bytes in 0 blocks
==1549==   total heap usage: 7 allocs, 7 frees, 6,033 bytes allocated
==1549== 
==1549== All heap blocks were freed -- no leaks are possible
==1549== 
==1549== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
==1549== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

but inside the Multicast task every loop consumes 1300 bytes that NEVER has been free.

YouDONG-ESP, 2020-12-01:
Hi filzek, if Im not miss understanding this issue, it's a memory leak issue for UDP multicast server, for unicast there is no such issue, right?

filzek, 2020-12-02:
We have found what is causing the problem.

Currently using mbedtls_aes_crypt_cbc and the padding used for the block was " " a space.
This throw a memory leak as the routine used in mbedtls_aes_crypt_cbc  to crypt and decrypt while using space as padding is corrupting the final crypt.

So, this is happen eveywhere and not only in Multicast udp, we have tested the same in all other parts of the SDK and the result is the same problem with the decrypt of it.

I will close it, but need to open the new ticket to explain the problem inside the ESP-IDF framework as this error doesnt happen in the GCC/X86 version of mbedtls.

closed 2020-12-02 by filzek

Alvin1Zhang, 2020-12-03:
https://github.com/espressif/esp-idf/issues/6197