espressif/esp-idf #6179: Device resets instead of waking up from deep sleep

Totalus, 2020-11-30:
## Problem Description

After deep sleep timeout, the device sometimes boots up as a reset `rst:0x1 (POWERON)` instead of waking up from deep sleep `rst:0x5 (DSLEEP)` as it should. I've observed that behavior with other wake up sources as well (e.g RTC GPIOs).

## Environment

- Development Kit:      ESP32-S2-Saola-1 v1.2 (based on ESP32-S2-WROOM)
- Module or chip used:  ESP32-S2-WROOM
- IDF version: `v4.2-rc-5-g511965b26`
- Build System: `idf.py`
- Compiler version: `xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0`
- Operating System: Windows
- Environment type: MSYS2 mingw64
- Using an IDE: No
- Power Supply: USB

### Expected Behavior

Device waking up from deep sleep `rst:0x5 (DSLEEP)`.

### Actual Behavior

Device often waking up in reset `rst:0x1 (POWERON)`, which causes all the RTC memory to be lost (time is reset).

### Steps and Code to reproduce this issue

Start from the [hello_world](https://github.com/espressif/esp-idf/tree/release/v4.2/examples/get-started/hello_world) example project and replace the main code by the following.

```cpp
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "esp_log.h"
#include "esp_sleep.h"

void app_main(void)
{
    esp_reset_reason_t resetReason = esp_reset_reason();

    ESP_LOGI("test", "--- Woke up ---\n");

    if(resetReason != ESP_RST_DEEPSLEEP)
    {
        ESP_LOGE("test", "ERROR: Device reset\n");
    }

    ESP_LOGI("test", "--- Going to sleep ---\n");
    esp_deep_sleep(5*1000000UL); // 5 seconds
}
```
Flash this on the IC and check the logs. Once in a while I the `ERROR: Device reset` prints out (see log below).

- The wake up happens after the deep sleep period (5 seconds) even when it is a reset, so it not likely caused by some other external event (supply glitch or else).
- I don't have to wait very long to see it happen. It sometimes happens twice in a row. There do not seem to be a regular pattern to the occurrences.

## Debug Logs

I won't attach all of the log because its the same thing over an over, but here's a snapshot where you can see some normal wake up `rst:0x5 (DSLEEP)` and a reset `rst:0x1 (POWERON)`. 

See highlighted lines:

```markdown
[...]

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
#rst:0x5 (DSLEEP),boot:0x8 (SPI_FAST_FLASH_BOOT)
[... 144 lines cut ...]
I (180) boot: Loaded app from partition at offset 0x10000
I (180) boot: Disabling RNG early entropy source...
I (186) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (194) cpu_start: Pro cpu up.
I (197) cpu_start: Application information:
I (202) cpu_start: Project name:     hello-world
I (207) cpu_start: App version:      v4.2-rc-5-g511965b26-dirty
I (214) cpu_start: Compile time:     Nov 29 2020 20:44:37
I (220) cpu_start: ELF file SHA256:  21c27e11a7b7efe5...
I (226) cpu_start: ESP-IDF:          v4.2-rc-5-g511965b26-dirty
I (233) cpu_start: Single core mode
I (237) heap_init: Initializing. RAM available for dynamic allocation:
I (244) heap_init: At 3FFC0438 len 0003BBC8 (238 KiB): DRAM
I (250) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (256) cpu_start: Pro cpu start user code
I (314) spi_flash: detected chip: generic
I (314) spi_flash: flash io: dio
W (314) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (324) cpu_start: Starting scheduler on PRO CPU.
I (330) test: --- Woke up ---

I (330) test: --- Going to sleep ---

[...]
```

igrr, 2020-11-30:
@Totalus Could you please try turning off the brownout detector in menuconfig, under Component config -> ESP32-S2 specific, and see the problem still exists?

Totalus, 2020-11-30:
@igrr It solved the issue. Nice catch !

How do you explain that the brown out detector triggers this easily ? I wouldn't expect it to do so on a simple boot of the IC.

igrr, 2020-11-30:
@Totalus Thanks for the confirmation. This might be a hardware issue. I'll follow up internally to see whether this is something we can work around in ESP-IDF.

closed 2020-11-30 by Totalus

Totalus, 2020-12-02:
I'm wondering, why does it not reset with the `ESP_RST_BROWNOUT` flag instead, since it seems to be a brownout detection issue ?

I think I also had brownout issues elsewhere, but they were never flagged as brownouts, just as simple resets, which makes it harder to figure out.

usmanmehmood55, 2022-03-01:
> @Totalus Could you please try turning off the brownout detector in menuconfig, under Component config -> ESP32-S2 specific, and see the problem still exists?

Can you kindly guide me on where exactly this file is in an Espressif IDF project? 
I am facing the same problem with the ESP32 C3, but I cannot find this file.