espressif/esp-idf #6180: Router connectivity issue

shaqlain137, 2020-11-30:
**Environment**

    Development Kit: [ESP32-DevKitV1]
    Module or chip used: [ESP32-WROOM-32]
    IDF version: V4.0
    IDE : VScode
    Operating System: [Linux]
    Power Supply: [USB]

**Problem Description**

Thankyou for seeing my issue,

we have have created a simple program that connects esp32 to MQTT broker,but problem is that it is working fine with **TPLINK** router but when i upload same code for the other router which is **Fortinet FortiAP 221E** ,it requires multiple reset to connect (sometimes it connects with network sometimes not) or else it is giving MQTT disconnected even if the network is fine. although router connects with network but still it gives MQTT disconnected error.While the TP-LINK router just working fine without any issue.

**These are the following errors while this issue**

Router SSID is : SURATi iLABRouter Password is : Smc@13579I (1183) wifi:wifi driver task: 3ffc6fe4, prio:23, stack:6656, core=0
I (1186) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1195) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1224) wifi:wifi firmware version: aa5336b
I (1224) wifi:config NVS flash: enabled
I (1225) wifi:config nano formating: disabled
I (1225) wifi:Init dynamic tx buffer num: 32
I (1229) wifi:Init data frame dynamic rx buffer num: 32
I (1234) wifi:Init management frame dynamic rx buffer num: 32
I (1239) wifi:Init management short buffer num: 32
I (1244) wifi:Init static rx buffer size: 1600
I (1248) wifi:Init static rx buffer num: 10
I (1252) wifi:Init dynamic rx buffer num: 32
I (1355) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1357) wifi:mode : sta (3c:71:bf:fd:03:a8)
I (1359) wifi softAP: wifi_init_sta finished.
I (1482) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (1484) wifi:state: init -> auth (b0)
I (1487) wifi:state: auth -> assoc (0)
I (1493) wifi:state: assoc -> run (10)
I (1504) wifi:connected with SURATi iLAB, aid = 6, channel 11, BW20, bssid = e8:1c:ba:28:9d:90
I (1504) wifi:security: WPA2-PSK, phy: bgn, rssi: -61
I (1510) wifi:pm start, type: 1

I (1535) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3184) tcpip_adapter: sta ip: 192.168.241.97, mask: 255.255.255.0, gw: 192.168.241.1
I (3184) wifi softAP: got ip:192.168.241.97
I (3187) wifi softAP: connected to ap SSID:SURATi iLAB password:Smc@13579

URI : mqtt://cloud.avin.com
PORT : 1883
USERNAME : shaqlain
PASSWORD : uVzpuddfdfnpLVI (3200) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (3214) httpHandler: Starting server on port: '80'
I (3220) httpHandler: Registering URI handlers

MQTT_CLIENT: Error transport connect
E (46180) MQTT_CLIENT: Error transport connect
E (75180) MQTT_CLIENT: Error transport connect
E (104180) MQTT_CLIENT: Error transport connect
E (133180) MQTT_CLIENT: Error transport connect
E (162180) MQTT_CLIENT: Error transport connect

Alvin1Zhang, 2020-11-30:
Thanks for reporting

AxelLin, 2020-11-30:
Are you sure the URL is correct?

$ nslookup cloud.avin.com
Server:		127.0.0.1
Address:	127.0.0.1#53

** server can't find cloud.avin.com: NXDOMAIN

shaqlain137, 2020-11-30:
i have just changed it here for privacy reasons and yes it is working fine in TP-LINK router as i menthioned

shaqlain137, 2020-11-30:
I've also tested with this my mobile hotspot and its seems to be working fine ,The issue in only with the router configuration i guess.

YouDONG-ESP, 2020-11-30:
HI @shaqlain137, thanks for reporting, your log shows that it's a MQTT connect error, but your wifi didn't lose connection right? could you provide a packet capture for this issues? It's a little bit inconvenient for us to get this specific router, it may takes tim

shaqlain137, 2020-12-01:
> HI @shaqlain137, thanks for reporting, your log shows that it's a MQTT connect error, but your wifi didn't lose connection right? could you provide a packet capture for this issues? It's a little bit inconvenient for us to get this specific router.

yes,
here a packet capture(i hope its right)
[image]
[image]
[image]

I (1138) wifi softAP: ESP_WIFI_MODE_STA
Checking monitor "SURATi iLAB"
Checking monitor "Smc@13579"
Router SSID is : SURATi iLABRouter Password is : Smc@13579I (1145) wifi:wifi driver task: 3ffc6ff0, prio:23, stack:6656, core=0
I (1149) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1159) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1187) wifi:wifi firmware version: aa5336b
I (1188) wifi:config NVS flash: enabled
I (1188) wifi:config nano formating: disabled
I (1188) wifi:Init dynamic tx buffer num: 32
I (1192) wifi:Init data frame dynamic rx buffer num: 32
I (1197) wifi:Init management frame dynamic rx buffer num: 32
I (1203) wifi:Init management short buffer num: 32
I (1207) wifi:Init static rx buffer size: 1600
I (1211) wifi:Init static rx buffer num: 10
I (1215) wifi:Init dynamic rx buffer num: 32
I (1320) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1321) wifi:mode : sta (3c:71:bf:fd:03:a8)
I (1323) wifi softAP: wifi_init_sta finished.
I (1446) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1447) wifi:state: init -> auth (b0)
I (1558) wifi:state: auth -> assoc (0)
I (1568) wifi:state: assoc -> run (10)
I (1581) wifi:connected with SURATi iLAB, aid = 1, channel 1, BW20, bssid = e8:1c:ba:28:96:58
I (1582) wifi:security: WPA2-PSK, phy: bgn, rssi: -70
I (1587) wifi:pm start, type: 1

I (1616) wifi:AP's beacon interval = 102400 us, DTIM period = 1

AxelLin, 2020-12-05:
> > HI @shaqlain137, thanks for reporting, your log shows that it's a MQTT connect error, but your wifi didn't lose connection right? could you provide a packet capture for this issues? It's a little bit inconvenient for us to get this specific router.
> 
> yes,
> here a packet capture(i hope its right)
Please provide raw wireshark log (.pcap) instead of screenshot.

YouDONG-ESP, 2020-12-07:
@shaqlain137 we bought this router for futher debugging, delivery might take some time.

YouDONG-ESP, 2020-12-08:
Hi @shaqlain137, we got the router, could you provide a code to reproduce this issue?

HarveyRong-Esp, 2020-12-21:
Hello @shaqlain137, we try to reproduce this issue in the local environment, but Fortinet FortiAP 221E needs to be configured with a firewall device and cannot be used alone. Therefore, we have no environment. Can only rely on you to provide information.
Generally speaking, about MQTT Related issues need to provide both LOG + ESP equipment and router wireless packages and router and server wired packages, it is difficult to analyze without these.

HarveyRong-Esp, 2021-01-18:
Hi @shaqlain137, Are there any updates?

HarveyRong-Esp, 2021-01-28:
Hi @shaqlain137, Has your issue been solved?
I built the `Fortinet FortiAP 221E` routing environment locally and tried to use the [example](https://github.com/espressif/esp-idf/tree/release/v4.0/examples/protocols/mqtt/tcp)  to reproduce, Unfortunately it does not disconnect. Can you provide minimal reproduce code?

**Note:** Once the code is published here, the code will be open source.

HarveyRong-Esp, 2021-02-22:
Hi @shaqlain137 
Are there any updates?  If not, I will close issue. If there is issue, please open it at any time.

Alvin1Zhang, 2021-08-05:
Will close due to short of updates and feedback, feel free to reopen.

closed 2021-08-05 by Alvin1Zhang