espressif/esp-idf #6181: Python 3.7.3 Requirements are not satisfied on Windows 10 x64

aznlilchino103, 2020-11-30:
The following Python requirements are not satisfied:

gdbgui==0.13.2.0

pygdbmi<=0.9.0.2

reedsolo>=1.5.3,<=1.5.4

bitstring>=3.1.6

ecdsa>=0.16.0

To install the missing packages, please run "C:\Users\xxxx.xxxx\ESP\esp-idf\install.bat"

Diagnostic information:

    IDF_PYTHON_ENV_PATH: C:\Users\xxxx.xxxx\.espressif\python_env\idf4.1_py3.7_env

    Python interpreter used: C:\Users\xxxx.xxxx\.espressif\python_env\idf4.1_py3.7_env\Scripts\python.exe

ESP-IDF v4.2-rc-5-g511965b26

Don't understand why it's showing up as the "4.1" vs what I've installed as "4.2"

dobairoland, 2020-11-30:
Hi @aznlilchino103, 

Do you also have a `C:\Users\xxxx.xxxx\.espressif\python_env\idf4.2_py3.7_env` directory? Do you have more than ESP-IDFs on your computer? Do you set globally an IDF_PATH variable?

`echo %IDF_PATH%` in ESP-IDF console prints the right directory? What is printed by `git describe --tags`?

I'm sorry for the lot of questions.

dobairoland, 2020-12-08:
@aznlilchino103 Do you have any update on this issue?

DanielPLongo, 2020-12-18:
I have the same issue. 

_Checking if Python packages are up to date...
The following Python requirements are not satisfied:
click>=5.0
future>=0.15.2
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
esp-windows-curses; sys_platform == 'win32'
To install the missing packages, please run "C:\Users\Daniel\esp\esp-idf\install.bat"

C:\Users\Daniel\esp\esp-idf>echo %IDF_PATH%
C:\Users\Daniel\esp\esp-idf

C:\Users\Daniel\esp\esp-idf>git describe --tags
v4.1_

directory:  C:\Users\Daniel\esp\.espressif\python_env\idf4.1_py3.9_env
No matter how many times I run install.bat and export.bat, I get the same error.

Thanks!

dobairoland, 2020-12-18:
@DanielPLongo Please examine the content of your PATH. It seems that you have there permanently set some IDF tools which are incompatible. I suspect that there might be more than one Pythons and that is why you are getting this behavior. Did you installed a Python with the Windows tools installer? Do you have others which you use for other purposes?

You can clean your path by removing all IDF tools from there. Unused Pythons could be uninstalled and removed from the PATH.

DanielPLongo, 2020-12-19:
I have tried a clean install on a machine that didn't have Python and I am getting the same error. I am using IDF 4.1 and I recall somewhere seeing a GIT issue thread about the json file needing to be replaced. Is that still an issue? Anyway, here is a screenshot of my PATH on this new machine.

[image]

DanielPLongo, 2020-12-19:
@dobairoland In an attempt to customize my folder paths I made errors. They have been addressed and it works now. I am sorry for wasting your time. Thank you for the quick replies.

dobairoland, 2020-12-19:
@DanielPLongo Great! I'm happy that you solved the issue.

dobairoland, 2020-12-19:
I'm closing this issue for lack of activity from the original reporter.

closed 2020-12-19 by dobairoland