espressif/esp-idf #6182: esp-idf git repo is always dirty on Windows

saloid, 2020-11-30:
## Environment

- IDF version:  4.0, 4.1, 4.2
- Operating System:     Windows

## Problem Description

Git repo is always dirty after cloning because there some file permissions, which don't have analogs in NTFS (as I understood):
```
old mode 100755
new mode 100644
```

### Expected Behavior

No changes at `git status`.

### Actual Behavior

git status shows bunch of modifications. And there is no way to discard them.

### Steps to reproduce

1. Clone or install via .exe installer toolchain on windows pc
2. `git status` show bunch of modifications

### Possible solutions

#### Best way

Change "problem files" permissions to corresponding ones in NTFS.
or
Maybe it's possible to do something with `.gitattributes` file to prevent permissions tracking.

#### User-side way
`git config core.fileMode false` disables file permissions change tracking.

saloid, 2020-11-30:
This issue helps me to resolve my case:
https://stackoverflow.com/questions/14564946/git-status-shows-changed-files-but-git-diff-doesnt

georgik, 2020-12-02:
@saloid Thank you for reporting the problem. The ZIP file with git repository was created on Linux and unzipping on Windows will create files and directories with different permission (e.g. 100644).
To solve the problem, please use following command in the directory with repository:
```
git config --local core.filemode false
```
The command will change .git/config:
```
[core]
    ...
    filemode = false
```
You may also need to apply the change to submodules. Please use command:
```
git submodule foreach git config --local core.filemode fals
```

saloid, 2020-12-02:
@georgik, thank you for reply. But my proposal is to make it clean automatically after installation, without extra actions.
One way is to change this permissions to those that will be compatible with Windows (and still be suitable for Linux, of course).
Other way is to disable permission change tracking by default in `.gitconfig` file in repo root, but as far as I know git will not apply settings from this file automatically (need to manually run `git config --local include.path ../.gitconfig`)
Third way is to run this command `git config --local core.filemode false` automatically in idf installer.

As for, first variant would be the best.

georgik, 2020-12-02:
@saloid Sure, I agree. The repo should be clean after installation.
Please consider the sugegsted commands as workaround. Tests for updated installer are in progress.

georgik, 2021-02-02:
@saloid The suggested solution is part of new installers
- Online installer (30 MB) : https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
- Offline installer - contains v4.2 and v4.1.1 (810 MB): https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.5.exe

closed 2021-02-03 by Alvin1Zhang