espressif/esp-idf #6184: ***ERROR*** A stack overflow in task sys_evt has been detected.

xukaijie111, 2020-11-30:
- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v2
- Module or chip used: ESP32-WROOM-32U
- IDF version : v4.3-dev-1901-g178b122c1
- Build System:         idf.py
- Compiler version :xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:    macOS
- Using an IDE?: VSCODE
- Power Supply:       USB

## Problem Description

I use esp-idf/examples/protocols/esp_http_client . and in the example ,I edit the code,the code is blow,the wifi can be connected , after the wifi connetced, I test the http function,but scrash a error.

### Error Show

ELF file SHA256: ee2d4dfa1b3eb545

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6968
load:0x40078000,len:14332
load:0x40080400,len:3672
0x40080400: _init at ??:?

entry 0x40080678
I (29) boot: ESP-IDF v4.3-dev-1901-g178b122c1 2nd stage bootloader
I (29) boot: compile time 21:30:02
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1c18c (115084) map
I (157) esp_image: segment 1: paddr=0x0002c1b4 vaddr=0x3ffb0000 size=0x03e64 ( 15972) load
I (164) esp_image: segment 2: paddr=0x00030020 vaddr=0x400d0020 size=0x8d840 (579648) map
0x400d0020: _stext at ??:?

I (385) esp_image: segment 3: paddr=0x000bd868 vaddr=0x3ffb3e64 size=0x0016c (   364) load
I (385) esp_image: segment 4: paddr=0x000bd9dc vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /Users/macuser/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1730

I (392) esp_image: segment 5: paddr=0x000bdde8 vaddr=0x40080404 size=0x14aec ( 84716) load
I (449) boot: Loaded app from partition at offset 0x10000
I (449) boot: Disabling RNG early entropy source...
I (461) cpu_start: Pro cpu up.
I (461) cpu_start: Starting app cpu, entry point is 0x400811ac
0x400811ac: call_start_cpu1 at /Users/macuser/esp/esp-idf/components/esp_system/port/cpu_start.c:126

I (447) cpu_start: App cpu up.
I (475) cpu_start: Pro cpu start user code
I (476) cpu_start: cpu freq: 160000000
I (476) cpu_start: Application information:
I (480) cpu_start: Project name:     esp-http-client-example
I (486) cpu_start: App version:      1
I (491) cpu_start: Compile time:     Nov 30 2020 21:59:22
[... 268 lines cut ...]
        },
    };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());
}

void app_main(void)
{
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    fast_scan();
   

    // xTaskCreate(&http_test_task, "http_test_task", 8192, NULL, 5, NULL);
}

igrr, 2020-11-30:
The issue is likely related to calling `http_rest_with_url` function from `event_handler` function. The latter function runs on the sys_evt task, which has limited stack size. Instead of calling http_rest_with_url directly from the event task, you can either call xTaskCreate to create a new task, as in the original example; or, use a semaphore to send an event to the main task, and start the HTTP request from there.

xukaijie111, 2020-12-01:
> The issue is likely related to calling `http_rest_with_url` function from `event_handler` function. The latter function runs on the sys_evt task, which has limited stack size. Instead of calling [...]

thanks for you reply. I do not understand why should call xTaskCreate for the http_rest_with_url function?

igrr, 2020-12-01:
The `event_handler` function is executed as a callback in sys_evt task. It is not recommended to perform lengthy operations in such callbacks. This is because until one callback returns, other event handlers will not run. So running an HTTP request from sys_evt task will prevent e.g. a Wi-Fi disconnect event handler from being executed, should you loose Wi-Fi connection.

xukaijie111, 2020-12-01:
> lengthy

> The `event_handler` function is executed as a callback in sys_evt task. It is not recommended to perform lengthy operations in such callbacks. This is because until one callback returns, other event [...]

thanks , I understand this idea .the task is for the event_handler.when an event is comming  the task should promise one process can handle the callback. But I see the function http_test_task release the task in the function end . so the event_handler can be excuted?

igrr, 2020-12-01:
In this case, http_test_task returns, so eventually control will return to the event loop in sys_evt task, and it can process further events.

However this is not good practice to rely on this: as mentioned above, some events may be delayed, which will cause incorrect operation. Consider the case that Wi-Fi gets disconnected while the application is waiting for HTTP response. Normally, in this case the event should be handled as follows:

* Wi-Fi stack sends "disconnected" event to the event loop
* event loop (in sys_evt) task calls all handlers registered for this event
* the event handler registered by esp_netif layer gets called, and brings the corresponding network interface "down"
* this releases the blocking `read` operation on the socket, making it return with an error code
* application can handle the error code and return from the HTTP request function

If the sys_evt task is blocked inside the HTTP request function, then the Wi-Fi disconnected event will not be handled, and the blocking "read" operation on the socket will not be interrupted. Depending on the configuration, it might eventually time out, but in this case it will take longer to handle Wi-Fi disconnection than it should, normally.

@xukaijie111 if you have further questions about application design the use of tasks and the event loops, you are welcome to raise such questions on the esp32.com forum. Please close this issue if you think it is resolved.

closed 2020-12-02 by xukaijie111