espressif/esp-idf #6185: registered_heaps_write_lock will never be NULL from heap_caps_init.c

samuelbles07, 2020-11-30:
## Environment

- Development Kit:  ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.3-dev-1197-g8bc19ba89
- Build System: Make
- Compiler version: (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux - ubuntu
- Using an IDE?: No
- Power Supply: USB

## Problem Description

I am trying to use this [repository](https://github.com/loboris/ESP32_TFT_library), i only change flash size to 4MB (expected by partition table total size) then directly `make` the project. An error happens that i think not from the TFT project but from idf  (CMIIW). 

### Expected Behavior

Build success

### Actual Behavior

```
/home/samuel/tools/esp/esp-idf/components/heap/heap_caps_init.c:230:44: error: the comparison will always evaluate as 'true' for the address of 'registered_heaps_write_lock' will never be NULL [-Werror=address]
     MULTI_HEAP_LOCK(&registered_heaps_write_lock);
                                            ^~
/home/samuel/tools/esp/esp-idf/components/heap/heap_caps_init.c:232:45: error: the comparison will always evaluate as 'true' for the address of 'registered_heaps_write_lock' will never be NULL [-Werror=address]
     MULTI_HEAP_UNLOCK(&registered_heaps_write_lock);

```

### Steps to reproduce

1. Clone [repository](https://github.com/loboris/ESP32_TFT_library
2. Change flash size to 4MB
3. Compile using `make`

igrr, 2020-11-30:
This seems to be due to this line: https://github.com/loboris/ESP32_TFT_library/blob/aa21772f54a71887ec08b2e8bbaef9e304009891/Makefile#L8
Could you try removing it?

Note that there are a couple other issues that need to be fixed in order to get the library to build: there is a missing `#include "driver/gpio.h"` in one file and a couple of misleading-indentation errors in another. That said, ESP32_TFT_library project hasn't been maintained since 2018, so even if it compiles, it might not work due to API changes. PR https://github.com/loboris/ESP32_TFT_library/pull/79 seems to have the changes to get the library to work with ESP-IDF v4.0, which is a bit more up to date.

samuelbles07, 2020-12-03:
Hi thank you for your response. Yes, i decided to use the PR that you mention. Thank you

closed 2020-12-03 by samuelbles07