espressif/esp-idf #6186: Current "Fix" for UART corrupted data issue?

TomMakovi, 2020-11-30:
There has been previous postings ( such as https://github.com/espressif/esp-idf/issues/1202 ) on the data from the ESP32's UARTs getting "corrupted" in the UART hardware.  And I'm not sure but the following ( from https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf ) might be "related".

[image]

In my design, I have the ESP32 sending data out UART1 to a second processor, and then "re-configuring" UART1 to receive data.  When the data is received by the ESP32, the bytes are "out of order" as shown below where I receive 27 parameters of 5 bytes each.  The first 7 parameters ( k = 0 to k = 6 ) shows the data is received correctly where the parameter values are "only" in the "bottom" two or three bytes.  Then the data "gets scrambled" in "k = 7" onward.

So, what is the "current best fix" for this issue?  ( And resetting the ESP32 is not an option for my design. )

3347 DETECTED CORRUPTED UART Data!
3349 Data = 
k = 0 bytes = 0 0 3 4E 1D
k = 1 bytes = 0 0 2 C8 2D
k = 2 bytes = 0 0 1 EB 86
k = 3 bytes = 0 0 1 1C 2B
k = 4 bytes = 0 0 0 37 DA
k = 5 bytes = 0 0 0 21 CE
k = 6 bytes = 0 0 0 85 11
k = 7 bytes = 88 A1 73 C0 9B
k = 8 bytes = CF 36 DA 50 0
k = 9 bytes = BB 24 65 1D 32
k = 10 bytes = 29 4E E8 4 58
k = 11 bytes = 22 46 F2 9C 4D
k = 12 bytes = C DE 75 6C C9
k = 13 bytes = 0 0 0 A 64
k = 14 bytes = 0 13 F4 0 8E
k = 15 bytes = 0 F 82 0 0
k = 16 bytes = 0 D 72 0 0
k = 17 bytes = 0 A DC 0 0
k = 18 bytes = 0 9 4A 0 0
k = 19 bytes = 0 8 F1 0 0
k = 20 bytes = 0 8 28 0 0
k = 21 bytes = 0 8 F 0 0
k = 22 bytes = 0 8 67 0 0
k = 23 bytes = 0 9 C 0 0
k = 24 bytes = 0 A 25 0 0
k = 25 bytes = 0 B 8E 0 0
k = 26 bytes = 0 D 42 0 0
3358 CORRUPTED DATA

AxelLin, 2020-12-01:
A question about commit eea38d76986f366d07fa781d19e28e7fda5d4ca8,
I'm wondering if fifo_cnt is not credible, why it's safe to do
  len = fifo_cnt > 0 ? 128 : 0;

negativekelvin, 2020-12-02:
Linking to possibly related
https://github.com/espressif/esp-idf/issues/6152
https://github.com/espressif/esp-idf/issues/5101

AshUK, 2021-02-17:
Is this issue resolved in c3/s2/s3 models ?

AxelLin, 2021-07-14:
> A question about commit [eea38d7](https://github.com/espressif/esp-idf/commit/eea38d76986f366d07fa781d19e28e7fda5d4ca8),
> I'm wondering if fifo_cnt is not credible, why it's safe to do
> len = fifo_cnt > 0 ? 128 : 0;

This issue is still open so I'm wondering how is the status.
I'm wondering if someone can clarify my previous question:
Why it's safe to check fifo_count when rx_status.wr_addr == rx_status.rd_addr?
If fifo_count is not credible, are you sure "len = fifo_cnt > 0 ? 128 : 0;" is always correct?

@o-marshmallow @negativekelvin

ammaree, 2021-07-14:
I believe the UART support in general needs some improvement.
The hardware FIFO is 1024 bytes big but only 256 bytes are used by default, 128 for RX and 128 for TX.
Many ESP32 projects (majority?) uses only a single UART and I would guess very very few uses more than 2 UART ports.
IDF does not provide any API to change the use of the FIFO buffer to better use the full 1024 bytes available.
In fact, the code snippet copied above shows that the usage is hardcode restricted to 128 bytes per buffer.

In my view, not a good idea and restriction that has forced us to avoid the UART APIs, and provide our own (limited) single UART support for buffers of TX/896 and RX/128 bytes, thus getting the benefit of the full 1024 bytes and 2Mbps speeds, without any SRAM based buffering.

negativekelvin, 2021-07-14:
>If fifo_count is not credible, are you sure "len = fifo_cnt > 0 ? 128 : 0;" is always correct?

When write ptr == read ptr the fifo is either empty or full and it is hard to believe if the fifo is full that fifo_cnt could be 0 (didn't count any bytes).  The probability of an error that big has to be insignificant.

AxelLin, 2021-07-15:
> When write ptr == read ptr the fifo is either empty or full and it is hard to believe if the fifo is full that fifo_cnt could be 0 (didn't count any bytes). The probability of an error that big has [...]

So you still think fifo_count is credible.
e.g. there is no chance "write ptr == read ptr" && fifo_count != 0 && fifo_count != 128?

negativekelvin, 2021-07-15:
>"write ptr == read ptr" && fifo_count != 0 && fifo_count != 128

Not if write ptr and read ptr are credible

AxelLin, 2021-07-19:
> > "write ptr == read ptr" && fifo_count != 0 && fifo_count != 128
> 
> You may have a point that if fifo_count << 128 it would be strange to assume it is full

I added below debug code for testing:
https://gist.github.com/AxelLin/154f07f8f1c0aa4b481987fac8546541

Then I setup a timer to print the u_dbg and u_dbg_fcnt variables in my test application.

I do observe the case "write ptr == read ptr" && fifo_count != 0 && fifo_count != 128" can happen.
The fifo_count I got in my testss including below values:
91, 122, 126, 56, 116

Anyway, I just want to point out this could happen and when it happen I received broken data.

negativekelvin, 2021-07-19:
What was the actual state of the fifo in those cases?

AxelLin, 2021-07-28:
> I believe the UART support in general needs some improvement.
> The hardware FIFO is 1024 bytes big but only 256 bytes are used by default, 128 for RX and 128 for TX.
> Many ESP32 projects (majority?) [...]

Hi @ammaree 
It's indeed strange to hard-coded  128 for FIFO buffer.
Can you create a new issue ? (to make it easier to trace status).

AxelLin, 2021-08-21:
> What was the actual state of the fifo in those cases?

Hi @negativekelvin 
Regarding https://github.com/espressif/esp-idf/issues/6186#issuecomment-882344686
It's difficult to reproduce.
But I notice calling vTaskList() and vTaskGetRunTimeStats() may cause uart receiving wrong data.
(Seems uart rx fifo full due to cpu is too busy or something wrong with above calls.)