espressif/esp-idf #6188: menuconfig overwrites CONFIG_IDF_TARGET.

nopnop2002, 2020-12-01:
## Environment
```
freeRTOS version:V10.2.1
NEWLIB version:3.0.0
lwIP version:2-1-3-0
ESP-IDF version:v4.3-dev-1901-g178b122c1-dirty
```

## Problem Description
menuconfig rewrites CONFIG_IDF_TARGET.

### Steps to reproduce

1.Step1
Delete current sdkconfig.
```
$ rm sdkconfig
```

2.Step2   
Create sdkconfig.defaults.

```
$ cat sdkconfig.defaults
#
CONFIG_IDF_TARGET="esp32s2"
```

3.Step3   
Execute make menuconfig.

4.Step4   
Check sdkconfig.
```
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000
```

dobairoland, 2020-12-01:
Hi @nopnop2002. 

Have you also removed the build directory, i.e. `rm -r build`?

A better solution is to use `idf.py set-target esp32s2` but if you want to have an ESP32-S2-only project then I agree that you need to use `sdkconfig.defaults`.

Are you still getting the issue after you've removed the build directory?

nopnop2002, 2020-12-01:
@dobairoland 

I tried the following:

```
$ rm -r build
$ rm sdkconfig
$ make menuconfig
make flash monitor
A fatal error occurred: This chip is ESP32-S2 not ESP32. Wrong --chip argument?
/home/nop/esp-idf/components/esptool_py/Makefile.projbuild:104: recipe for target 'flash' failed
make: *** [flash] Error 2

```

dobairoland, 2020-12-01:
@nopnop2002 I've missed one point. `make` is deprecated and only the `esp32` target is working there. You need to use `idf.py` (or directly `cmake`) for `esp32s2`.

nopnop2002, 2020-12-01:
@dobairoland 

idf.py does not use a custom sdkconfig.

```
$ idf.py menuconfig
$ idf.py set-target esp32s2 flash monitor
```

dobairoland, 2020-12-01:
@nopnop2002 `idf.py set-target` cleans any config you've made in menuconfig. The correct sequence is

```bash
rm -r build sdkconfig   # probably not necessary but I recommend since you invoked make earlier
idf.py set-target esp32s2
idf.py menuconfig
idf.py flash monitor
```

dobairoland, 2020-12-01:
for reference: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/index.html#step-7-configure & https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/build-system.html#selecting-the-target

nopnop2002, 2020-12-01:
@dobairoland 

I will read [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/build-system.html#selecting-the-target) again.

closed 2020-12-01 by nopnop2002

dobairoland, 2020-12-01:
So if you have `sdkconfig.defaults` with `CONFIG_IDF_TARGET="esp32s2"` then the correct sequence which should work is: 
```bash
rm -r build sdkconfig 
idf.py menuconfig
idf.py flash monitor
```

Please let me know if the above is not working and we can reopen this issue.