espressif/esp-idf #6190: mdns_query_a() can't find host

nopnop2002, 2020-12-01:
## Environment

```
freeRTOS version:V10.2.1
NEWLIB version:3.0.0
lwIP version:2-1-3-0
ESP-IDF version:v4.3-dev-1901-g178b122c1-dirty
Problem Description
mDNS does not return a response after 10 minutes.
```

## Problem Description
mdns_query_a() cannot find the host.

## Steps to reproduce   

1.Step1
build [this](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mdns) example and run.
mdns hostname set to: [esp32-mdns-05C634]

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (605) system_api: Base MAC address is not set
I (605) system_api: read default base MAC address from EFUSE
I (615) mdns-test: mdns hostname set to: [esp32-mdns-05C634]
I (615) mdns-test: EXAMPLE_MDNS_INSTANCE=ESP32 with mDNS
```

2.Step2
I execute the following program on another ESP32.
mdns_query_a() work fine.
esp32-mdns-05C634.local resolved to: 192.168.10.132

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (594) wifi station: ESP_WIFI_MODE_STA
I (624) wifi:wifi driver task: 3ffb9510, prio:23, stack:3584, core=0
I (624) system_api: Base MAC address is not set
I (624) system_api: read default base MAC address from EFUSE
I (644) wifi:wifi firmware version: b9e8721
I (644) wifi:wifi certification version: v7.0
I (644) wifi:config NVS flash: enabled
I (644) wifi:config nano formating: disabled
I (654) wifi:Init data frame dynamic rx buffer num: 32
I (654) wifi:Init management frame dynamic rx buffer num: 32
I (664) wifi:Init management short buffer num: 32
I (664) wifi:Init dynamic tx buffer num: 32
I (674) wifi:Init static rx buffer size: 1600
I (674) wifi:Init static rx buffer num: 10
I (674) wifi:Init dynamic rx buffer num: 32
I (684) wifi_init: rx ba win: 6
I (684) wifi_init: tcpip mbox: 32
I (694) wifi_init: udp mbox: 6
I (694) wifi_init: tcp mbox: 6
I (694) wifi_init: tcp tx win: 5744
I (704) wifi_init: tcp rx win: 5744
I (704) wifi_init: tcp mss: 1440
I (704) wifi_init: WiFi IRAM OP enabled
I (714) wifi_init: WiFi RX IRAM OP enabled
I (714) wifi:Set ps type: 0

I (814) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (814) wifi:mode : sta (30:ae:a4:ca:e1:40)
I (814) wifi:enable tsf
I (944) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (944) wifi:state: init -> auth (b0)
I (964) wifi:state: auth -> assoc (0)
I (1004) wifi:state: assoc -> run (10)
W (1014) wifi:<ba-add>idx:0 (ifx:0, f8:b7:97:36:de:52), tid:5, ssn:0, winSize:64
W (1094) wifi:<ba-add>idx:1 (ifx:0, f8:b7:97:36:de:52), tid:0, ssn:2, winSize:64
I (1094) wifi:connected with aterm-d5a4ee-g, aid = 11, channel 1, BW20, bssid = f8:b7:97:36:de:52
I (1104) wifi:security: WPA2-PSK, phy: bgn, rssi: -47
I (1104) wifi:pm start, type: 0

I (1124) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2094) esp_netif_handlers: sta ip: 192.168.10.130, mask: 255.255.255.0, gw: 192.168.10.1
I (2094) wifi station: got ip:192.168.10.130
I (2094) wifi station: connected to ap SSID:aterm-d5a4ee-g password:
I (2104) wifi station: wifi_init_sta finished.
I (2104) wifi station: connect to ap SSID:aterm-d5a4ee-g
I (2114) wifi station: Query A: esp32-mdns-05C634.local
I (3074) wifi station: Query A: esp32-mdns-05C634.local resolved to: 192.168.10.132
```

```
static void query_mdns_host(const char * host_name)
{
        ESP_LOGI(TAG, "Query A: %s.local", host_name);

        struct esp_ip4_addr addr;
        addr.addr = 0;

        esp_err_t err = mdns_query_a(host_name, 2000,  &addr);
        if(err){
                if(err == ESP_ERR_NOT_FOUND){
                        ESP_LOGW(TAG, "%s: Host was not found!", esp_err_to_name(err));
                        return;
                }
                ESP_LOGE(TAG, "Query Failed: %s", esp_err_to_name(err));
                return;
        }

        ESP_LOGI(TAG, "Query A: %s.local resolved to: " IPSTR, host_name, IP2STR(&addr));
}

void app_main()
{
        //Initialize NVS
        esp_err_t ret = nvs_flash_init();
        if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
                ESP_ERROR_CHECK(nvs_flash_erase());
                ret = nvs_flash_init();
        }
        ESP_ERROR_CHECK(ret);

        ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
        wifi_init_sta();

        //initialize mDNS
        ESP_ERROR_CHECK( mdns_init() );
        query_mdns_host("esp32-mdns-05C634");

}

```

3.Step3   
I execute the same program again.
mdns_query_a() don't work.
Host was not found!

```
I (0) cpu_start: Starting scheduler on APP CPU.
I (594) wifi station: ESP_WIFI_MODE_STA
I (624) wifi:wifi driver task: 3ffb9510, prio:23, stack:3584, core=0
I (624) system_api: Base MAC address is not set
I (624) system_api: read default base MAC address from EFUSE
I (644) wifi:wifi firmware version: b9e8721
I (644) wifi:wifi certification version: v7.0
I (644) wifi:config NVS flash: enabled
I (644) wifi:config nano formating: disabled
I (654) wifi:Init data frame dynamic rx buffer num: 32
I (654) wifi:Init management frame dynamic rx buffer num: 32
I (664) wifi:Init management short buffer num: 32
I (664) wifi:Init dynamic tx buffer num: 32
I (674) wifi:Init static rx buffer size: 1600
I (674) wifi:Init static rx buffer num: 10
I (674) wifi:Init dynamic rx buffer num: 32
I (684) wifi_init: rx ba win: 6
I (684) wifi_init: tcpip mbox: 32
I (694) wifi_init: udp mbox: 6
I (694) wifi_init: tcp mbox: 6
I (694) wifi_init: tcp tx win: 5744
I (704) wifi_init: tcp rx win: 5744
I (704) wifi_init: tcp mss: 1440
I (704) wifi_init: WiFi IRAM OP enabled
I (714) wifi_init: WiFi RX IRAM OP enabled
I (714) wifi:Set ps type: 0

I (814) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (814) wifi:mode : sta (30:ae:a4:ca:e1:40)
I (814) wifi:enable tsf
I (944) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (944) wifi:state: init -> auth (b0)
I (974) wifi:state: auth -> assoc (0)
I (994) wifi:state: assoc -> run (10)
W (1004) wifi:<ba-add>idx:0 (ifx:0, f8:b7:97:36:de:52), tid:5, ssn:0, winSize:64
W (1104) wifi:<ba-add>idx:1 (ifx:0, f8:b7:97:36:de:52), tid:0, ssn:1, winSize:64
I (1104) wifi:connected with aterm-d5a4ee-g, aid = 11, channel 1, BW20, bssid = f8:b7:97:36:de:52
I (1104) wifi:security: WPA2-PSK, phy: bgn, rssi: -48
I (1114) wifi:pm start, type: 0

I (1134) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2094) esp_netif_handlers: sta ip: 192.168.10.130, mask: 255.255.255.0, gw: 192.168.10.1
I (2094) wifi station: got ip:192.168.10.130
I (2094) wifi station: connected to ap SSID:aterm-d5a4ee-g password:
I (2104) wifi station: wifi_init_sta finished.
I (2104) wifi station: connect to ap SSID:aterm-d5a4ee-g
I (2114) wifi station: Query A: esp32-mdns-05C634.local
W (4214) wifi station: ESP_ERR_NOT_FOUND: Host was not found!
```

Alvin1Zhang, 2020-12-02:
Thanks for reporting, we will look into.

david-cermak, 2020-12-09:
@nopnop2002 Thanks for making a separate issue for this problem and sorry for such a late response.
Indeed there's a regression in Espressif mDNS library. 

Could you please set the MDNS_STRICT_MODE to 1 in:
https://github.com/espressif/esp-idf/blob/master/components/mdns/private_include/mdns_private.h#L34
and test again? This is just to be sure we're experiencing the same problem. The issue needs to be fixed!

closed 2021-01-05 by espressif-bot