espressif/esp-idf #6193: [esp32-S2] download flash tool cant flash S2 over USB

chegewara, 2020-12-01:
I am trying to flash custom PCB with esp32 S2 wrover module. I do have both buttons so i can easy enter download mode.
I am trying to use native USB to flash on windows and after some time this is the log:
```
no log file output ...
test offset :  65536 0x10000
case ok
test offset :  32768 0x8000
case ok
test offset :  57344 0xe000
case ok
test offset :  4096 0x1000
case ok
======
CONNECT BAUD: 115200
============
Uploading stub...
Running stub...
[2020-12-01 16:08:13,358][ESP8266Loader_spi[1]][espDownloader.py][line:554][ERROR]: ESP32S2 Chip stub error esp_stub_and_set_baud.
no log file output ...
```

With S2 devkit and UART there is no problem to flash.

igrr, 2020-12-01:
@chegewara are you able to flash with esptool.py provided in ESP-IDF?

chegewara, 2020-12-01:
Now even more weird part. After connecting USB to UART to my custom PCB i am getting this message from flash download tool when i am trying to erase it:
```
.Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
[2020-12-01 16:24:18,009][ESP8266Loader_spi[1]][espDownloader.py][line:1767][ERROR]: ESP32 efuse secure check fail:Already secure boot, NOT Allow download bootloader again!!!
[2020-12-01 16:24:18,009][ESP8266Loader_spi[1]][espDownloader.py][line:528][ERROR]: ESP32S2 Secure Encrypt check error flash_erase_func.
```
and here we can see all my efuses:
```
Connecting...
Detecting chip type... ESP32-S2
espefuse.py v3.0
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
TEMP_SENSOR_CAL (BLOCK2)                 Temperature calibration                            = -8.3 R/W (0b101010011)
ADC1_MODE0_D2 (BLOCK2)                   ADC1 calibration 1                                 = -104 R/W (0x9a)
ADC1_MODE1_D2 (BLOCK2)                   ADC1 calibration 2                                 = -100 R/W (0x99)
ADC1_MODE2_D2 (BLOCK2)                   ADC1 calibration 3                                 = -104 R/W (0x9a)
ADC1_MODE3_D2 (BLOCK2)                   ADC1 calibration 4                                 = -112 R/W (0x9c)
ADC2_MODE0_D2 (BLOCK2)                   ADC2 calibration 5                                 = -16 R/W (0x84)
ADC2_MODE1_D2 (BLOCK2)                   ADC2 calibration 6                                 = -12 R/W (0x83)
ADC2_MODE2_D2 (BLOCK2)                   ADC2 calibration 7                                 = -12 R/W (0x83)
ADC2_MODE3_D2 (BLOCK2)                   ADC2 calibration 8                                 = 4 R/W (0x01)
ADC1_MODE0_D1 (BLOCK2)                   ADC1 calibration 9                                 = -32 R/W (0b101000)
ADC1_MODE1_D1 (BLOCK2)                   ADC1 calibration 10                                = -24 R/W (0b100110)
ADC1_MODE2_D1 (BLOCK2)                   ADC1 calibration 11                                = -20 R/W (0b100101)
ADC1_MODE3_D1 (BLOCK2)                   ADC1 calibration 12                                = -12 R/W (0b100011)
ADC2_MODE0_D1 (BLOCK2)                   ADC2 calibration 13                                = -16 R/W (0b100100)
ADC2_MODE1_D1 (BLOCK2)                   ADC2 calibration 14                                = -8 R/W (0b100010)
ADC2_MODE2_D1 (BLOCK2)                   ADC2 calibration 15                                = -8 R/W (0b100010)
ADC2_MODE3_D1 (BLOCK2)                   ADC2 calibration 16                                = -4 R/W (0b100001)

Config fuses:
DIS_RTC_RAM_BOOT (BLOCK0)                Disables boot from RTC RAM                         = False R/W (0b0)
DIS_ICACHE (BLOCK0)                      Disables ICache                                    = False R/W (0b0)
DIS_DCACHE (BLOCK0)                      Disables DCache                                    = False R/W (0b0)
DIS_DOWNLOAD_ICACHE (BLOCK0)             Disables Icache when SoC is in Download mode       = False R/W (0b0)
DIS_DOWNLOAD_DCACHE (BLOCK0)             Disables Dcache when SoC is in Download mode       = False R/W (0b0)
DIS_FORCE_DOWNLOAD (BLOCK0)              Disables forcing chip into Download mode           = False R/W (0b0)
DIS_CAN (BLOCK0)                         Disables the TWAI Controller hardware              = False R/W (0b0)
DIS_BOOT_REMAP (BLOCK0)                  Disables capability to Remap RAM to ROM address sp = False R/W (0b0)
                                         ace
FLASH_TPUW (BLOCK0)                      Configures flash startup delay after SoC power-up, = 0 R/W (0x0)
                                          unit is (ms/2). When the value is 15, delay is 7.
                                         5 ms
DIS_LEGACY_SPI_BOOT (BLOCK0)             Disables Legacy SPI boot mode                      = False R/W (0b0)
UART_PRINT_CHANNEL (BLOCK0)              Selects the default UART for printing boot msg     = UART0 R/W (0b0)
DIS_USB_DOWNLOAD_MODE (BLOCK0)           Disables use of USB in UART download boot mode     = False R/W (0b0)
UART_PRINT_CONTROL (BLOCK0)              Sets the default UART boot message output mode     = Enabled R/W (0b00)
FLASH_TYPE (BLOCK0)                      Selects SPI flash type                             = 4 data lines R/W (0b0)
FORCE_SEND_RESUME (BLOCK0)               Forces ROM code to send an SPI flash resume comman = False R/W (0b0)
                                         d during SPI boot
BLOCK_USR_DATA (BLOCK3)                  User data
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS (BLOCK0)                          Disables programming of individual eFuses          = 0 R/W (0x00000000)
RD_DIS (BLOCK0)                          Disables software reading from BLOCK4-10           = 0 R/W (0b0000000)

Identity fuses:
BLOCK0_VERSION (BLOCK0)                  BLOCK0 efuse version                               = 0 R/W (0b00)
SECURE_VERSION (BLOCK0)                  Secure version (used by ESP-IDF anti-rollback feat = 0 R/W (0x0000)
                                         ure)
MAC (BLOCK1)                             Factory MAC Address
   = 7c:df:a1:01:76:a4: (OK) R/W
[... cut ...]
Flash voltage (VDD_SPI) determined by GPIO45 on reset (GPIO45=High: VDD_SPI pin is powered from internal 1.8V LDO
GPIO45=Low or NC: VDD_SPI pin is powered directly from VDD3P3_RTC_IO via resistor Rspi. Typically this voltage is 3.3 V).
```

I can easy flash with arduino, with both modes, or from esp-idf. The only issue is flash download tool.

All 4 bootstrap pins (including pin26 - psram) are kept not connected, except IO0 to button.
Any help @projectgus 

Thanks

PS i managed to flash with USB to UART module, but its not easy

chegewara, 2020-12-01:
> @chegewara are you able to flash with esptool.py provided in ESP-IDF?

With esp-idf and arduino, using UART or USB, even dfu.

chegewara, 2020-12-01:
There is one more thing, which i will just mention here, but i think it requires to open new issue. When i eventually flashed S2 with tool i am getting this message, which didnt appear when i flash with arduino IDE:
```
load:0x3ffe6108,len:0x570
load:0x4004c000,len:0xb0c
load:0x40050000,len:0x25a4
SHA-256 comparison failed:
Calculated: f0f97a4a55bc2315c896597accd6a483c70f87997eda7dfae11029f3603c0017
Expected: 28f622ecd5b38a7f0d09f7835c477d1ccc24c9e1591844b3faabc39b1235a77a
Attempting to boot anyway...
entry 0x4004c180
```

igrr, 2020-12-01:
> With esp-idf and arduino, using UART or USB, even dfu.

Sorry, did you mean that you *are* able to flash using these options, or you *are not*?

If the issue is only with Flash download tools, I am sorry but this is out of scope of ESP-IDF project issue tracker.

chegewara, 2020-12-01:
Sorry @igrr 
yes, i am able to flash using all those option. Only issue i have is flash over USB using flash download tool.

Where i can open issue with tool then, sorry for wrong issue tracker?

closed 2023-03-23 by chegewara