espressif/esp-idf #6194: Ethernet is not working with ESP32-WROOM-32U/UE modules

rmshub, 2020-12-02:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32U | ESP32-WROOM-32UE
- IDF version : v4.2-beta1
- Build System: idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:  Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE:  Microsoft Visual Studio Code
- Power Supply: USB

## Problem Description

Ethernet interface with LAN7820A is not working with ESP32-WROOM-32U and ESP32-WROOM-32UE modules and the following errors are observed always

E (1766) emac_esp32: emac_esp32_init(308): reset timeout
E (1766) esp_eth: esp_eth_driver_install(198): init mac failed
 
The firmware is working properly with ESP32-WROOM-32D module. While capturing MDC (GPIO23) and MDIO (GPIO18) using logic analyzer able to see clock and data with ESP32-WROOM-32D. However, No clock & data is observed for ESP32-WROOM-32U/32UE modules. It seems like there a problem in GPIO Mux.

### Expected Behavior
Ethernet interface with LAN8720A should work with ESP32-WROOM-32U/UE modules
### Actual Behavior
Ethernet is not working
### Steps to reproduce

1.  Basic ethernet example code

rmshub, 2020-12-02:
@me-no-dev  @igrr @mahavirj 
Can you please help me on this issue. We are in production stage.

suda-morris, 2020-12-02:
The module works fine in my side. Most of the "emac reset timeout" failure is caused by invalid or non-exist RMII clock. For how to configure ethernet according to your board design, please read this chapter: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_eth.html#configure-mac-and-phy

rmshub, 2020-12-03:
@suda-morris RMII clock is available in my design and verified by CRO. Is it possible to share RMII clock, MDIO and MDC signals routing reference.

suda-morris, 2020-12-03:
I'm using a LAN8720 break-out board for testing. RMII clock is generated by ESP32 and output from GPIO17. SMI interface is the default setting we used in all Ethernet examples. Have you configured the RMII clock correctly in the menuconfig?

rmshub, 2020-12-03:
I'm using 50MHz crystal oscillator as reference clock for ESP32 GPIO0 and LAN8720. It is working with ESP32-WROOM-32D but not working ESP32-WROOM-32U/UE modules.

suda-morris, 2020-12-03:
have you checked the 50MHz from GPIO0, instead of from the oscillator side? Also pay attention to the voltage level of that signal.

rmshub, 2020-12-03:
Yes, Measured at GPIO0 and observed 50MHz clock. If it is problem with RMII clock, it won't work with ESP32-WROOM-32D module. May be there is difference between ESP32-WROOM-32D & ESP32-WROOM-32U/UE ethernet specifications in terms of voltage levels and timings.

suda-morris, 2020-12-03:
Can you report your issue to our support team with more information about your testing information? (e.g. clock wave, schematic, basic reproduce code) https://www.espressif.com/en/contact-us/technical-inquiries/hardware-issues

rmshub, 2020-12-04:
RMII clock out mode is working properly using either GPIO16 or GPIO17 with ESP32-WROOM-32U. However, RMII clock input mode (GPIO0) is not working. It seems there is an issue with silicon or change in hardware specifications. Kindly help me on this, how to make it work with GPIO0.

Alvin1Zhang, 2020-12-04:
@rmshub Thanks for reporting, will you please kindly help write to sales@espressif.com? Thanks.

sauttefk, 2020-12-05:
We have a similar sounding problem with a new design with a 20 pieces sample run (same circuit we used before, except for a change to a ESP32-WROOM-32UE-16MB, LAN8720A to LAN8720Ai and a different RJ45 connector with integrated magnetics). RMII clock is derived from GPIO17.
The problem vanished when we exchanged the LAN8720Ai with ~2 year old LAN8720A we had left in stock.
The only difference between LAN8720A and LAN8720Ai I found is an extended temperature range of the "i" version.
Still investigating if there is some other issue causing our problems.

Alvin1Zhang, 2021-04-14:
Thanks for reporting, feel free to reopen.

closed 2021-04-14 by Alvin1Zhang

hagenbuch, 2022-03-07:
I seem to have the same problem with micropython:

I am using an Olimex ESP32-POE Revision B board. With the same board, I have been successful to get Ethernet running on micropython 1.14 which uses  IDF3, but whenever I try one of the newer versions (like 1.18) that have been made with IDF4, I get those debug messages

`E (2589) emac_esp32: emac_esp32_init(308): reset timeout
E (2589) esp_eth: esp_eth_driver_install(198): init mac failed`

in the code 

`# lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), power = machine.Pin(12), phy_type = network.PHY_LAN8720, phy_addr = 0, clock_mode = network.ETH_CLOCK_GPIO17_OUT)
lan = network.LAN(mdc = machine.Pin(23), mdio = machine.Pin(18), power = machine.Pin(12), phy_type = network.PHY_LAN8720, phy_addr = 0)
time.sleep_ms(500)
lan.active(True)`

The code that DID work on 1.14 is seen in the comment above

Any help would be greatly appreciated!