espressif/esp-idf #6197: mbedtls_aes_crypt_cbc  error padding with SPACES in all ESP-IDF version

filzek, 2020-12-02:
Hi, 

We have found out a problematic bug that cause the crypted info with mbedtls_aes_crypt_cbc  when " " SPACE padding is being used.

The error happen randonly with some kind of padding sizes and contents, so, once that kind happen again the error will be throw with a invalid results.

So, maybe this shall be add as a warning and report to mbed to fix it.

The erro do not happen on GCC/X86 compiler.

Alvin1Zhang, 2020-12-03:
@filzek Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

AdityaHPatwardhan, 2020-12-03:
@filzek Thank you for reporting the issue

Can you please provide some sample code or something with which I could be able to reproduce the invalid result case ? 

Also can you please check if "CONFIG_MBEDTLS_HARDWARE_AES" is enabled in your case ( Component config->Mbedtls-> Enable hardware AES acceleration) . If you have not changed the setting it should be enabled by default.

filzek, 2020-12-03:
Its happen in all 4 and up SDK version. Maybe happen in the entire sdk project as well.

We have built in 4.1 release and 4.2-rc both with CONFIG_MBEDTLS_HARDWARE_AES enabled as default.

With external psiram and not using external psiram did the same result as well.

as the problem happen only with specific arrays of chars, we can send a example we have create for it:

the is an example on how the problem happen:

`
size_to_be_crypted_t = strlen("{\"destiny\":\"d3ad563b-b5c7-4a6e-ba4e-71bd2a07adbf\",\"CommandControl\":677,\"source\":\"dc0fb241-c104-4ca5-970f-880c84f3e90f\",\"queueContent\":[{\"port\":\"0008\",\"command\":\"5004\",\"data\":\"50\"}]}");
old_size_block = size_to_be_crypted_t;
mod_aes_block = old_size_block % 16;

if (mod_aes_block!=0){
aes_block_missing = AES_BLOCK_SIZE - mod_aes_block;
size_to_be_crypted_t+= aes_block_missing;
}
size_to_be_crypted_t+=16;

unsigned char sAdjust_str[size_to_be_crypted_t+1];
memset(sAdjust_str,0,size_to_be_crypted_t+1);
memset(sAdjust_str,' ',size_to_be_crypted_t);
memcpy(sAdjust_str, "{\"destiny\":\"d3ad563b-b5c7-4a6e-ba4e-71bd2a07adbf\",\"CommandControl\":677,\"source\":\"dc0fb241-c104-4ca5-970f-880c84f3e90f\",\"queueContent\":[{\"port\":\"0008\",\"command\":\"5004\",\"data\":\"50\"}]}",old_size_block);
mbedtls_aes_context aes256;
mbedtls_aes_init(&aes256);

ret = mbedtls_aes_setkey_enc(&aes256, (const unsigned char *)"3sHaZQE134mR9k+8A0oQ81hzYsD0Fy+r", 256);
if(ret < 0) {
mbedtls_aes_free(&aes256);
printf("ERROR MBEC SETKEY\n");
}
uint8_t cbc[16];
volatile unsigned char 			is_IV[17] 						= { 0 };
memcpy(is_IV, "1NM6J7wHLVK8QDWn", strlen("1NM6J7wHLVK8QDWn"));
os_memcpy(cbc, is_IV, AES_BLOCK_SIZE);
memcpy(cbc, is_IV, 16);
ret = mbedtls_aes_crypt_cbc(&aes256, MBEDTLS_AES_ENCRYPT, size_to_be_crypted_t, cbc, sAdjust_str, sAdjust_str);
mbedtls_aes_free(&aes256);
`

and now just decrypt it back and the problem is show!

`ret = mbedtls_aes_crypt_cbc(&aes256, MBEDTLS_AES_DECRYPT, size_to_be_crypted_t, cbc, sAdjust_str, sAdjust_str);

`

just print the final result in sAdjust_str and the problem is show, corruption in the crypt algotithm.

AdityaHPatwardhan, 2020-12-04:
@filzek I can confirm the error in decrypted message. I will try to find out the cause of the problem and let you know.

AdityaHPatwardhan, 2020-12-04:
> @filzek I can confirm the error in decrypted message. I will try to find out the cause of the problem (and hopefully a fix ) and let you know.

Sorry for this hasty reply, I take that back. it was silly mistake on my part. 
The decrypted result is fine.

AdityaHPatwardhan, 2020-12-15:
@filzek any updates, If the problem is resolved can we close the issue ? 
Thank you

AdityaHPatwardhan, 2021-01-05:
Hi @filzek is the issue resolved ?

If it would help here is my output after encryption and decryption of the same string. It exactly matches.

```
I (334) MBEDTLS_ENC_DEC:
string before mbedtls encrypt:
 {"destiny":"d3ad563b-b5c7-4a6e-ba4e-71bd2a07adbf","CommandControl":677,"source":"dc0fb241-c104-4ca5-970f-880c84f3e90f","queueContent":[{"port":"0008","command":"5004","data":"50"}]}
I (354) MBEDTLS_ENC_DEC:
string after mbedtls decrypt: 
 {"destiny":"d3ad563b-b5c7-4a6e-ba4e-71bd2a07adbf","CommandControl":677,"source":"dc0fb241-c104-4ca5-970f-880c84f3e90f","queueContent":[{"port":"0008","command":"5004","data":"50"}]}
```

 If it is resolved, can you please close the issue ?
Thank you

filzek, 2021-01-05:
Hi, it is closed!

Thanks!

closed 2021-01-05 by filzek