espressif/esp-idf #6198: Warnings caused by security features

KaeLL, 2020-12-02:
## Environment

- IDF version: v4.2-rc
- Build System: Make

## Problem Description

When enabling the following configurations
[image]
[image]
the following warning is displayed after closing `menuconfig`
```
warning: SECURE_ENABLE_SECURE_ROM_DL_MODE (defined at /esp-idf/components/bootloader/Kconfig.projbuild:749) has direct dependencies IDF_TARGET_ESP32S2 && !SECURE_DISABLE_ROM_DL_MODE with value n, but is currently being y-selected by the following symbols:
 - SECURE_FLASH_ENCRYPTION_MODE_RELEASE (defined at /esp-idf/components/bootloader/Kconfig.projbuild:604), with value y, direct dependencies <choice SECURE_FLASH_ENCRYPTION_MODE> (value: y), and select condition <choice SECURE_FLASH_ENCRYPTION_MODE> (value: y)
Loaded configuration '/home/dev07/Documents/VSCode/AR/TestBench/ESP32-W5100/sdkconfig'
warning: SECURE_ENABLE_SECURE_ROM_DL_MODE (defined at /esp-idf/components/bootloader/Kconfig.projbuild:749) has direct dependencies IDF_TARGET_ESP32S2 && !SECURE_DISABLE_ROM_DL_MODE with value n, but is currently being y-selected by the following symbols:
 - SECURE_FLASH_ENCRYPTION_MODE_RELEASE (defined at /esp-idf/components/bootloader/Kconfig.projbuild:604), with value y, direct dependencies <choice SECURE_FLASH_ENCRYPTION_MODE> (value: y), and select condition <choice SECURE_FLASH_ENCRYPTION_MODE> (value: y)
No changes to save (for '/home/dev07/Documents/VSCode/AR/TestBench/ESP32-W5100/sdkconfig')
warning: SECURE_ENABLE_SECURE_ROM_DL_MODE (defined at /esp-idf/components/bootloader/Kconfig.projbuild:749) has direct dependencies IDF_TARGET_ESP32S2 && !SECURE_DISABLE_ROM_DL_MODE with value n, but is currently being y-selected by the following symbols:
 - SECURE_FLASH_ENCRYPTION_MODE_RELEASE (defined at /esp-idf/components/bootloader/Kconfig.projbuild:604), with value y, direct dependencies <choice SECURE_FLASH_ENCRYPTION_MODE> (value: y), and select condition <choice SECURE_FLASH_ENCRYPTION_MODE> (value: y)
```
### Expected Behavior

No warnings

### Steps to reproduce

Enable the mentioned options on your sdkconfig with idf v4.2-rc

projectgus, 2020-12-03:
Hi @KaeLL,

Thanks for letting us know about this.

You can ignore the warnings, this config item is only relevant on ESP32-S2 and doesn't do anything on ESP32, so there should be no incorrect behaviour.

However we will fix this ASAP.

yannpom, 2021-01-21:
Hello,

I just bricked 2 of my production boards with ESP32-S2 because of this problem.

I set both `SECURE_FLASH_ENCRYPTION_MODE_RELEASE=y` and `SECURE_DISABLE_ROM_DL_MODE=y` to flash my first production device (with Secure boot v2 + Encryption), but `SECURE_ENABLE_SECURE_ROM_DL_MODE` kept the value `y`.

Then in `startup.c:252` I got stuck (infinite boot / reset) on this assert on "Failed to enable Secure Download mode":
```
#if CONFIG_SECURE_ENABLE_SECURE_ROM_DL_MODE
    err = esp_efuse_enable_rom_secure_download_mode();
    assert(err == ESP_OK && "Failed to enable Secure Download mode");
#endif
```

My understanding is that `SECURE_ENABLE_SECURE_ROM_DL_MODE` **cannot** be `y` when `SECURE_DISABLE_ROM_DL_MODE=y`.

I managed to workaround this problem and finally flashed successfully my ESP-S2 devices with Secure boot v2 + Encryption + `SECURE_DISABLE_ROM_DL_MODE` by modifying `components/bootloader/Kconfig.projbuild`.

I can create a pull request if needed.

closed 2021-05-05 by mahavirj