espressif/esp-idf #6199: httpd_stop() hangs

sz-po, 2020-12-03:
## Environment

- Development Kit: ESP32-DevKitC
- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.1
- Build System: CMake
- Compiler version: crosstool-NG esp-2020r2) 8.2.0
- Operating System: macOS
- Using an IDE?: Yes, VSCode + IDF plugin 
- Power Supply: external 5V

## Problem Description

After calling httpd_stop() program just hangs, and never goes through this function. Http server still running and http requests are handled.

### Expected Behavior

Http server should stop and program flow should just go.

### Actual Behavior

Program hangs on httpd_stop() function.

### Steps to reproduce

1. Just run code below.

### Code to reproduce this issue

```cpp
#include <stdio.h>
#include "sdkconfig.h"
#include "esp_log.h"
#include "esp_event.h"
#include "nvs_flash.h"
#include "esp_http_server.h"
#include "lwip/sys.h"
#include "esp_wifi.h"

static httpd_handle_t test_httpd_server_handle = NULL;
static httpd_config_t test_httpd_config = HTTPD_DEFAULT_CONFIG();

void app_main(void)
{
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    // Initialization of NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_ERROR_CHECK(esp_netif_init());

    esp_netif_create_default_wifi_sta();

    wifi_init_config_t wifi_init_config = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&wifi_init_config));

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_start());

    vTaskDelay(2000 / portTICK_PERIOD_MS);
    esp_wifi_connect();

    for (;;) {      
        ESP_LOGI("main", "httpd_start: before");
        ESP_ERROR_CHECK(httpd_start(&test_httpd_server_handle, &test_httpd_config));
        ESP_LOGI("main", "httpd_start: after");

        vTaskDelay(10000 / portTICK_PERIOD_MS);

        ESP_LOGI("main", "httpd_stop: before");
        ESP_ERROR_CHECK(httpd_stop(&test_httpd_server_handle));
        // Program NEVER reach this place, i'm still able to call http server
        ESP_LOGI("main", "httpd_stop: after");

        vTaskDelay(10000 / portTICK_PERIOD_MS);
    }

}
```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7008
ho 0 tail 12 room 4
load:0x40078000,len:13076
load:0x40080400,len:3896
0x40080400: _init at ??:?

entry 0x40080688
I (31) boot: ESP-IDF v4.1-dirty 2nd stage bootloader
I (31) boot: compile time 00:30:18
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00011000 00006000
I (79) boot:  1 phy_init         RF data          01 01 00017000 00001000
I (86) boot:  2 factory          factory app      00 00 00020000 00180000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x19bf8 (105464) map
I (154) esp_image: segment 1: paddr=0x00039c20 vaddr=0x3ffbdb60 size=0x03dbc ( 15804) load
I (161) esp_image: segment 2: paddr=0x0003d9e4 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /Users/szymonpodeszwa/esp/esp-idf/components/freertos/xtensa_vectors.S:1778

I (162) esp_image: segment 3: paddr=0x0003ddf0 vaddr=0x40080404 size=0x02228 (  8744) load
I (174) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0x859d0 (547280) map
0x400d0020: _stext at ??:?

I (387) esp_image: segment 5: paddr=0x000c59f8 vaddr=0x4008262c size=0x17610 ( 95760) load
0x4008262c: esp_cpu_in_ocd_debug_mode at /Users/szymonpodeszwa/esp/esp-idf/components/soc/esp32/cpu_util.c:54

I (443) boot: Loaded app from partition at offset 0x20000
I (443) boot: Disabling RNG early entropy source...
I (443) cpu_start: Pro cpu up.
I (447) cpu_start: Application information:
I (452) cpu_start: Project name:     esp32-idf-robot-firmware
I (458) cpu_start: App version:      1
I (462) cpu_start: Compile time:     Dec  3 2020 00:29:52
I (468) cpu_start: ELF file SHA256:  914c72d8a14f1c16...
I (474) cpu_start: ESP-IDF:          v4.1-dirty
I (480) cpu_start: Starting app cpu, entry point is 0x40081374
0x40081374: call_start_cpu1 at /Users/szymonpodeszwa/esp/esp-idf/components/esp32/cpu_start.c:271

I (0) cpu_start: App cpu up.
I (490) heap_init: Initializing. RAM available for dynamic allocation:
I (497) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (503) heap_init: At 3FFC71F0 len 00018E10 (99 KiB): DRAM
I (509) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (516) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (523) heap_init: At 40099C3C len 000063C4 (24 KiB): IRAM
I (528) cpu_start: Pro cpu start user code
I (547) spi_flash: detected chip: generic
I (548) spi_flash: flash io: dio
I (548) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (671) wifi:wifi driver task: 3ffb949c, prio:23, stack:6656, core=0
I (671) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (671) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (711) wifi:wifi firmware version: 3ea4c76
I (711) wifi:config NVS flash: enabled
I (711) wifi:config nano formating: disabled
I (711) wifi:Init dynamic tx buffer num: 32
I (711) wifi:Init data frame dynamic rx buffer num: 32
I (711) wifi:Init management frame dynamic rx buffer num: 32
I (721) wifi:Init management short buffer num: 32
I (721) wifi:Init static rx buffer size: 1600
I (731) wifi:Init static rx buffer num: 10
I (731) wifi:Init dynamic rx buffer num: 32
I (841) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (841) wifi:mode : sta (24:62:ab:dc:ae:54)
I (2851) main: httpd_start: before
I (2851) main: httpd_start: after
I (2971) wifi:new:<9,1>, old:<1,0>, ap:<255,255>, sta:<9,1>, prof:1
I (2971) wifi:state: init -> auth (b0)
I (2971) wifi:state: auth -> assoc (0)
I (2981) wifi:state: assoc -> run (10)
I (3001) wifi:connected with citygarden.io, aid = 1, channel 9, 40U, bssid = 4a:8f:5a:41:0e:ab
I (3001) wifi:security: WPA2-PSK, phy: bgn, rssi: -43
I (3011) wifi:pm start, type: 1

I (3031) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (3031) esp_netif_handlers: sta ip: 192.168.88.217, mask: 255.255.255.0, gw: 192.168.88.1
W (6011) httpd_uri: httpd_uri: URI '/' not found
W (6011) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
W (7341) httpd_uri: httpd_uri: URI '/' not found
W (7341) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
W (22391) httpd_uri: httpd_uri: URI '/' not found
W (22391) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
I (32851) main: httpd_stop: before
**BELLOW REQUESTS HANDLED AFTER CALLING httpd_stop()**
W (34481) httpd_uri: httpd_uri: URI '/' not found
W (34481) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
W (36731) httpd_uri: httpd_uri: URI '/' not found
W (36731) httpd_txrx: httpd_resp_send_err: 404 Not Found - This URI does not exist
```

## Other items if possible
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/5632772/sdkconfig.zip)

AxelLin, 2020-12-03:
Check if setting CONFIG_ESP_NETIF_LOOPBACK=y helps.

boarchuz, 2020-12-03:
httpd_stop(test_httpd_server_handle); // (no &)

mahavirj, 2020-12-03:
@sz-po As suggested by @boarchuz please update your test code from:

```
ESP_ERROR_CHECK(httpd_stop(&test_httpd_server_handle));
```

to

```
ESP_ERROR_CHECK(httpd_stop(test_httpd_server_handle));
```

This should fix the problem.

sz-po, 2020-12-03:
@AxelLin @mahavirj  @boarchuz Thank you, guys. Of course you are right. Thank your for support.

closed 2020-12-03 by sz-po