espressif/esp-idf #6201: switching from ble to wifi

lky2431, 2020-12-03:
Hi,

I am writing a project using ESP32-WROVER. One of the feature is that i have to send a wifi and password from a mobile phone to the esp32 through ble gatt server. The wifi information is sucessfully received. However, there is "lock_acquire_generic" bug or sometimes "corrode heap" bug when the esp is setting the wifi information.

The code I use to receive the wifi information like:
static void gatts_profile_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param)
{.....
}else if (res == SPP_IDX_SPP_WIFI_VAL){
ESP_LOGI(GATTS_TABLE_TAG,"message come");
.....
sta_on();

The code I use to switch to wifi like:
void sta_on(){
ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
ESP_ERROR_CHECK(esp_wifi_start() );
}
The wifi is already initialized and set to STA before. It work completely fine if I set using wifi but not bluetooth for first program.

If anyone have similar experience before, could you share some to me.
I would be grateful about it.
Thank you

TimXia, 2021-03-05:
@lky2431  Could you please provide a demo to reproduce the issue? You can also try the blufi example in IDF. It does the same thing as you expected.

xulongzhuang, 2021-04-07:
@lky2431 ,Please provide more detailed information, such as IDF version and commit ID, if there is a demo that can reproduce the problem, it would be better

Alvin1Zhang, 2021-08-05:
@lky2431 Thanks for reporting, would you please help share if any updates for the issue? Thanks.

jack0c, 2022-05-23:
@lky2431 It looks like a heap issue，there are several heap issues fixed，please update the latest release version

closed 2022-07-18 by Alvin1Zhang