espressif/esp-idf #6202: CONFIG_FMB_TCP_CONNECTION_TOUT_SEC not defined

pilnikov, 2020-12-03:
CONFIG_FMB_TCP_CONNECTION_TOUT_SEC
not defined in \esp-idf\components\freemodbus\tcp_slave\port\port_tcp_slave.c

(void)vMBPortClose(); instead (void)vMBTCPPortClose(); in \esp-idf\components\freemodbus\tcp_slave\modbus_controller\mbc_tcp_slave.c

in \esp-idf\components\freemodbus\tcp_master\port\port_tcp_master.c

i change sequence - first vMBMasterTCPPortDisable(void), after vMBMasterTCPPortClose(void)

void
vMBMasterTCPPortDisable(void) //first
{
    for (USHORT ucCnt = 0; ucCnt < MB_TCP_PORT_MAX_CONN; ucCnt++) {
        MbSlaveInfo_t* pxInfo = xMbPortConfig.pxMbSlaveInfo[ucCnt];
        if (pxInfo) {
            xMBTCPPortMasterCloseConnection(pxInfo);
            if (pxInfo->pucRcvBuf) {
                free(pxInfo->pucRcvBuf);
            }
            free(pxInfo);
            xMbPortConfig.pxMbSlaveInfo[ucCnt] = NULL;
        }
    }
}

void
vMBMasterTCPPortClose(void) //second
{
    (void)vTaskDelete(xMbPortConfig.xMbTcpTaskHandle);
    (void)vMBMasterTCPPortDisable(); // this must be first
    free(xMbPortConfig.pxMbSlaveInfo);
    vQueueDelete(xMbPortConfig.xConnectQueue);
    vMBMasterPortTimerClose();
    // Release resources for the event queue.
    vMBMasterPortEventClose();
}

alisitsyn, 2020-12-04:
@pilnikov,

1. The CONFIG_FMB_TCP_CONNECTION_TOUT_SEC is defined in `sdkconfig.h` file which is included by `port.h -> mbconfig.h` and actually defined in the port_tcp_slave.c. I agree this it is better to make it clear.
2. The `vMBPortClose()` can not be called here because it is for serial port but the vMBTCPPortClose() missing the call to the `vMBPortTimerClose(  )` for example and some resources are not released correctly. I will check and fix the destroy  sequence for TCP master and slave.
3. In general the order of the functions does not matter because the prototypes of these functions are defined in  `freemodbus\modbus\include\mbport.h` which is included from `freemodbus\tcp_slave\port\port_tcp_slave.c`.

Thanks.

alisitsyn, 2020-12-07:
Hi @pilnikov,

I checked the inconsistencies you mentioned above. I also checked the destroy sequence of master and slave TCP port. It looks good for me and required resources are released. So, according to information above I can not confirm your issue. There are may be subjective things in the code but they do not prevent to work properly. If you disagree with me please clarify your point of view. Would you agree to close issue?

closed 2020-12-07 by pilnikov