espressif/esp-idf #6203: Coexistence of ESP8266-RTOS-SDK and ESP-IDF Programming Environments on same computer

colesnicov, 2020-12-03:
Hi. I have set the Environment for programming ESP8266-RTOS-SDK in my computer. Now I would like to start programming ESP32. But if I understand correctly, then when calling these two commands:
```
cd ~/esp/esp-idf
./install.sh
```
, so it will scatter my Environment for programming ESP8266. Correctly?
What are the options for programming both chips on the same computer? Thank you.

Kiznoh, 2020-12-03:
Hi @colesnicov, 
both ESP-IDF and ESP8266-RTOS-SDK may use the same environment variables. If they mix up, you may encounter weird problems. To avoid that only source one export.sh script in a single terminal session. Based on ESP8266-RTOS-SDK documentation, you may already have the export.sh script in your shell's profile, the IDF_PATH set or PATH altered. Just make sure to only use one of the development environments in a single terminal session.

If you have any follow up questions, don't hesitate and comment here.

colesnicov, 2020-12-04:
Good day. Thank you for a quick response. I'll try it.

closed 2020-12-04 by colesnicov