espressif/esp-idf #6204: SPI driver transaction crash

veelox321, 2020-12-03:
Hi,

I am trying to write an SPI driver for my project, but cant seems to work. I've tried to use spi_device_transmit, spi_device_polling_transmit, queue_trans, get_trans but i never seem to be able to get it working.

`
void initializeSPI(void)
{
    esp_err_t err;
      
    spi_bus_config_t buscfg;

    buscfg.miso_io_num=12; //PIN_MTDI_HSPI_Q_MISO        GPIO_NUM_12
    buscfg.mosi_io_num=23; //PIN_IMU_SPI_MOSI            GPIO_NUM_23
    buscfg.sclk_io_num=19; //PIN_IMU_SPI_MISO            GPIO_NUM_19
    buscfg.quadwp_io_num=-1;
    buscfg.quadhd_io_num=-1;  
    buscfg.max_transfer_sz=4094; //Maximum transfer size, in bytes. Defaults to 4094 if 0.

    err=spi_bus_initialize(SPI3_HOST, &buscfg, 2);
    printf("spi initialize err : %s\r\n", esp_err_to_name(err));

    //SPI_PORT.begin(PIN_IMU_SPI_CLK,PIN_IMU_SPI_MISO,PIN_IMU_SPI_MOSI);
    //SPI_SdCard.begin(PIN_MTMS_HSPI_CLK,PIN_MTDI_HSPI_Q_MISO,PIN_MTCK_HSPI_D_MOSI,-1);  
}

ICM_20948_Status_e ICM_20948_SPI::begin(uint8_t csPin, uint32_t SPIFreq)
{
#if DEBUG
    printf ("__FUNCTION__ = %s\n", __FUNCTION__);
#endif
    esp_err_t err;
    //spi_host_device_t host = SPI2_HOST; //between 1 and 3

    spi_device_interface_config_t devcfg;
    devcfg.clock_speed_hz=SPIFreq;
    devcfg.mode=ICM_20948_SPI_DEFAULT_MODE;
    devcfg.spics_io_num=_csPin_;
    devcfg.queue_size=1;
    devcfg.flags = SPI_DEVICE_HALFDUPLEX;             
    devcfg.command_bits = 0;       // no command bits used
    devcfg.address_bits = 0;       // register address is first byte in MOSI
    devcfg.dummy_bits = 0;          // no dummy bits used
    devcfg.pre_cb =NULL;

    err=spi_bus_add_device(SPI3_HOST, &devcfg, &spi);
    printf("icm 20948 add device err : %s\r\n", esp_err_to_name(err)); // return ESP_OK

    // Set pinmodes
    gpio_set_direction((gpio_num_t)_csPin_, GPIO_MODE_OUTPUT);

    // Set pins to default positions
    gpio_set_level((gpio_num_t)_csPin_, 0x1);

    // Set up the serif
    _serif.write = ICM_20948_write_SPI;
    _serif.read = ICM_20948_read_SPI;
    _serif.user = (void *)this; // refer to yourself in the user field

    // Link the serif
    _device._serif = &_serif;

    // Perform default startup
    status = startupDefault();
    if (status != ICM_20948_Stat_Ok)
    {
        return status;
    }

    return ICM_20948_Stat_Ok;
}

// serif functions for the I2C and SPI classes
/*ICM_20948_Status_e ICM_20948_write_I2C(uint8_t reg, uint8_t *data, uint32_t len, void *user)
{
    if (user == NULL)
    {
        return ICM_20948_Stat_ParamErr;
    }
    TwoWire *_i2c = ((ICM_20948_I2C *)user)->_i2c; // Cast user field to ICM_20948_I2C type and extract the I2C interface pointer
    uint8_t addr = ((ICM_20948_I2C *)user)->_addr;
    if (_i2c == NULL)
    {
        return ICM_20948_Stat_ParamErr;
    }

    _i2c->beginTransmission(addr);
    _i2c->write(reg);
    _i2c->write(data, len);
    _i2c->endTransmission();

    // for( uint32_t indi = 0; indi < len; indi++ ){
    //     _i2c->beginTransmission(addr);
    //     _i2c->write(reg + indi);
    //     _i2c->write(*(data + indi) );
    //     _i2c->endTransmission();
    //     delay(10);
    // }

    // delay(10);

    return ICM_20948_Stat_Ok;
}

ICM_20948_Status_e ICM_20948_read_I2C(uint8_t reg, uint8_t *buff, uint32_t len, void *user)
{
    if (user == NULL)
    {
        return ICM_20948_Stat_ParamErr;
    }
    TwoWire *_i2c = ((ICM_20948_I2C *)user)->_i2c;
    uint8_t addr = ((ICM_20948_I2C *)user)->_addr;
    if (_i2c == NULL)
    {
        return ICM_20948_Stat_ParamErr;
    }

    _i2c->beginTransmission(addr);
    _i2c->write(reg);
    _i2c->endTransmission(false); // Send repeated start

    uint32_t offset = 0;
    uint32_t num_received = _i2c->requestFrom(addr, len);
    // while(_i2c->available()){
    //     if(len > 0){
    //         *(buff + offset) = _i2c->read();
    //         len--;
    //     }else{
    //         break;
    //     }
    // }

    if (num_received == len)
    {
        for (uint8_t i = 0; i < len; i++)
        {
            buff[i] = _i2c->read();
        }
        return ICM_20948_Stat_Ok;
    }
    else
    {
        return ICM_20948_Stat_NoData;
    }

    if (len != 0)
    {
        return ICM_20948_Stat_NoData;
    }
    return ICM_20948_Stat_Ok;
}
*/

ICM_20948_Status_e ICM_20948_write_SPI(uint8_t reg, uint8_t *data, uint32_t len, void *user)
{
#if DEBUG
    printf ("__FUNCTION__ = %s\n", __FUNCTION__);
#endif
    gpio_set_level((gpio_num_t)_csPin_, 0x0);

    static uint8_t mosi[64];

    mosi[0] = reg;

    for (int i = 0; i < len; i++)
        mosi[i+1] = data[i];

    if (!spi_transfer_pf (mosi, NULL, len+1))
    {
        return ICM_20948_Stat_Err;
    }

    gpio_set_level((gpio_num_t)_csPin_, 0x1);

    return ICM_20948_Stat_Ok;
}

ICM_20948_Status_e ICM_20948_read_SPI(uint8_t reg, uint8_t *buff, uint32_t len, void *user)
{
#if DEBUG
    printf ("__FUNCTION__ = %s\n", __FUNCTION__);
#endif
    if (user == NULL)
    {
        return ICM_20948_Stat_ParamErr;
    }

    static uint8_t mosi[64];
    static uint8_t miso[64];

    memset (mosi, 0xff, 64);
    memset (miso, 0xff, 64);

    mosi[0] = reg;

    gpio_set_level((gpio_num_t)_csPin_, 0x0);

    if (!spi_transfer_pf (mosi, miso, len+1))
    {
        gpio_set_level((gpio_num_t)_csPin_, 0x1);
        return ICM_20948_Stat_Err;
    }
    // shift data one by left, first byte received while sending register address is invalid
    for (int i=0; i < len; i++)
      buff[i] = miso[i+1];

    gpio_set_level((gpio_num_t)_csPin_, 0x1);

    return ICM_20948_Stat_Ok;
}

size_t spi_transfer_pf (const uint8_t *mosi, uint8_t *miso, uint16_t len)
{
#if DEBUG
    printf ("__FUNCTION__ = %s\n", __FUNCTION__);
#endif
    spi_transaction_t spi_trans;

    memset(&spi_trans, 0, sizeof(spi_trans)); // zero out spi_trans;
    spi_trans.tx_buffer = mosi;
    spi_trans.rx_buffer = miso;
    spi_trans.length=len*8;

    if (spi_device_transmit(spi, &spi_trans) != ESP_OK)
        return 0;

    return len;
}
`

Now the error i get is 
`
IMU1...__FUNCTION__ = begin
icm 20948 add device err : ESP_OK
__FUNCTION__ = startupDefault
__FUNCTION__ = checkID
__FUNCTION__ = ICM_20948_check_id
__FUNCTION__ = ICM_20948_get_who_am_i
__FUNCTION__ = ICM_20948_set_bank
__FUNCTION__ = ICM_20948_execute_w
__FUNCTION__ = ICM_20948_write_SPI
__FUNCTION__ = spi_transfer_pf
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000001  PS      : 0x00060031  A0      : 0x8008410c  A1      : 0x3ffb0c50
A2      : 0x3ffbad94  A3      : 0x3ffc70a0  A4      : 0x800864f2  A5      : 0x3ffb0c60
A6      : 0x3ffb0ca8  A7      : 0x00000001  A8      : 0x80083f58  A9      : 0x3ffb0c20
A10     : 0x3ffc70a0  A11     : 0x00000000  A12     : 0x80087f4f  A13     : 0x3ffb0c30
A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
Core 1 was running in ISR context:
EPC1    : 0x00000001  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

ELF file SHA256: 85155fe13a0a5c14

Backtrace: 0xfffffffe:0x3ffb0c50 |<-CORRUPTED

Rebooting...
ets Jun  8 2016 00:22:57
`

KaeLL, 2020-12-03:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#instrfetchprohibited

Btw, why are you controlling the CS pin manually?

veelox321, 2020-12-03:
SparkFun is doing it on his arduino implementation, i tought it was the same i just migrate digitalWrite to gpio_set_level :
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary/tree/master/src
```
ICM_20948_Status_e ICM_20948_write_SPI(uint8_t reg, uint8_t *data, uint32_t len, void *user)
{
    if (user == NULL)
    {
        return ICM_20948_Stat_ParamErr;
    }
    SPIClass *_spi = ((ICM_20948_SPI *)user)->_spi; // Cast user field to ICM_20948_SPI type and extract the SPI interface pointer
    uint8_t cs = ((ICM_20948_SPI *)user)->_cs;
    SPISettings spisettings = ((ICM_20948_SPI *)user)->_spisettings;
    if (_spi == NULL)
    {
        return ICM_20948_Stat_ParamErr;
    }

    // 'Kickstart' the SPI hardware. This is a fairly high amount of overhead, but it guarantees that the lines will start in the correct states even when sharing the SPI bus with devices that use other modes
    _spi->beginTransaction(spisettings);
    _spi->transfer(0x00);
    _spi->endTransaction();

    digitalWrite(cs, LOW);
    // delayMicroseconds(5);
    _spi->beginTransaction(spisettings);
    _spi->transfer(((reg & 0x7F) | 0x00));
    //  SPI.transfer(data, len); // Can't do this thanks to Arduino's poor implementation
    for (uint32_t indi = 0; indi < len; indi++)
    {
        _spi->transfer(*(data + indi));
    }
    _spi->endTransaction();
    // delayMicroseconds(5);
    digitalWrite(cs, HIGH);

    return ICM_20948_Stat_Ok;
}

ICM_20948_Status_e ICM_20948_read_SPI(uint8_t reg, uint8_t *buff, uint32_t len, void *user)
{
    if (user == NULL)
    {
        return ICM_20948_Stat_ParamErr;
    }
    SPIClass *_spi = ((ICM_20948_SPI *)user)->_spi;
    uint8_t cs = ((ICM_20948_SPI *)user)->_cs;
    SPISettings spisettings = ((ICM_20948_SPI *)user)->_spisettings;
    if (_spi == NULL)
    {
        return ICM_20948_Stat_ParamErr;
    }

    // 'Kickstart' the SPI hardware. This is a fairly high amount of overhead, but it guarantees that the lines will start in the correct states
    _spi->beginTransaction(spisettings);
    _spi->transfer(0x00);
    _spi->endTransaction();

    digitalWrite(cs, LOW);
    //   delayMicroseconds(5);
    _spi->beginTransaction(spisettings);
    _spi->transfer(((reg & 0x7F) | 0x80));
    //  SPI.transfer(data, len); // Can't do this thanks to Arduino's stupid implementation
    for (uint32_t indi = 0; indi < len; indi++)
    {
        *(buff + indi) = _spi->transfer(0x00);
    }
    _spi->endTransaction();
    //   delayMicroseconds(5);
    digitalWrite(cs, HIGH);

    return ICM_20948_Stat_Ok;
}
```

KaeLL, 2020-12-03:
Regarding the original error, as mentioned in the link I posted, it's probably a function pointer you forgot to initialize or did so with something other than an actual function.  
As for your SPI code, the spi_master library takes care of GPIO management. Just provide it with the pins.

veelox321, 2020-12-03:
I have added buffer space to :
`    static uint8_t mosi[64];
    static uint8_t miso[64];

    memset (mosi, 0xff, 64);
    memset (miso, 0xff, 64);
`
and now its not crashing anymore. But the other problem is one i already had. It stays almost indefinitely inside 
`
   if (spi_device_transmit(spi, &spi_trans) != ESP_OK)
        return 0;
`
I am doing multi threading and the function seems alive and able to reset the watchdog since i have no warning.

veelox321, 2020-12-05:
This seems to have fixed it:
```
spi_device_interface_config_t devcfg = {};
```

closed 2020-12-05 by veelox321