espressif/esp-idf #6205: LWIP crashes or tiT task hits WDT on unexpected wifi connect/disconnect

sebirdman, 2020-12-04:
## Environment

- IDF version: 4.0.2
- Build System:  CMAKE
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux

## Problem Description

Recently we started moving our application over to using poll and selects with longer timeouts to prevent extra loops. When my wifi disconnects suddently, my application often panics or gets stuck in a WDT from the tiT task

### Expected Behavior

The device does not crash or WDT

### Actual Behavior

The device WDT's or crashes

### Steps to reproduce

1. create socket and poll on it
2. kick the device off wifi (on my unifi router, i just press "reconnect" on the client)
3. repeat

I've got two sockets actively polling, after doing this at most, two times, i'm able to reproduce either the above coredump, or the WDT. 

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4014e99c  PS      : 0x00060230  A0      : 0x8014eb60  A1      : 0x3ffbd7d0  
0x4014e99c: select_check_waiters at /usr/src/work/goat/build_esp32_app/../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/sockets.c:2670
A2      : 0x00000036  A3      : 0x00000000  A4      : 0x00000039  A5      : 0x00000001  
A6      : 0x3ffc6714  A7      : 0x00000059  A8      : 0x00000059  A9      : 0x3ffbd7b0  
A10     : 0x00000001  A11     : 0x00000036  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000055  LBEG    : 0x400917a0  LEND    : 0x400917bc  LCOUNT  : 0xffffffff  
0x400917a0: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:175
0x400917bc: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:197
ELF file SHA256: 5c989c682ad53a4c
Backtrace: 0x4014e999:0x3ffbd7d0 0x4014eb5d:0x3ffbd800 0x40166b45:0x3ffbd820 0x4015976d:0x3ffbd840 0x4015a3ed:0x3ffbd870 0x4016017e:0x3ffbd8a0 0x40165731:0x3ffbd8c0 0x40152399:0x3ffbd8e0 0x40152418:0x3ffbd900 0x400962f9:0x3ffbd930
0x4014e999: select_check_waiters at /usr/src/work/goat/build_esp32_app/../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/sockets.c:2669
0x4014eb5d: event_callback at /usr/src/work/goat/build_esp32_app/../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/sockets.c:2620
0x40166b45: lwip_netconn_do_connected at /usr/src/work/goat/build_esp32_app/../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/api_msg.c:1394 (discriminator 1)
0x4015976d: tcp_process at /usr/src/work/goat/build_esp32_app/../dependencies/espressif/esp-idf/components/lwip/lwip/src/core/tcp_in.c:903 (discriminator 1)
0x4015a3ed: tcp_input at /usr/src/work/goat/build_esp32_app/../dependencies/espressif/esp-idf/components/lwip/lwip/src/core/tcp_in.c:438
0x4016017e: ip4_input at /usr/src/work/goat/build_esp32_app/../dependencies/espressif/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:784
0x40165731: ethernet_input at /usr/src/work/goat/build_esp32_app/../dependencies/espressif/esp-idf/components/lwip/lwip/src/netif/ethernet.c:186
0x40152399: tcpip_thread_handle_msg at /usr/src/work/goat/build_esp32_app/../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/tcpip.c:180 (discriminator 4)
0x40152418: tcpip_thread at /usr/src/work/goat/build_esp32_app/../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/tcpip.c:154
0x400962f9: vPortTaskWrapper at /usr/src/work/goat/build_esp32_app/../dependencies/espressif/esp-idf/components/freertos/port.c:143
```

## Other items if possible

I've done some digging and have found that these seem to both be semi-related to each other (the crash and the WDT) both are cases where we enter the for loop:

```
  for (scb = select_cb_list; scb != NULL; scb = scb->next) {
```

in select_check_waiters. The WDT is triggered because we do not exit this loop, while the coredump appears to be from uninitialized data placed in the select_cb_list buffer.

Alvin1Zhang, 2020-12-04:
Thanks for the report, we will look into.

sebirdman, 2020-12-04:
Thanks for the quick response @Alvin1Zhang 

I should also note that both of the polls are in different unrelated tasks.

sebirdman, 2020-12-04:
I'm also seeing the following crash in these cases:

```
assertion "select_cb->prev == NULL" failed: file "../dependencies/espressif/esp-idf/components/lwip/lwip/src/api/sockets.c", line 1865, function: lwip_unlink_select_cb
abort() was called at PC 0x401eb73c on core 0
0x401eb73c: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)
```

```
invoke_abort() (/home/bird/deako/goat/dependencies/espressif/esp-idf/components/esp32/panic.c:157)
abort() (/home/bird/deako/goat/dependencies/espressif/esp-idf/components/esp32/panic.c:175)
__assert_func(const char * file, int line, const char * func, const char * failedexpr) (/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62)
lwip_unlink_select_cb(struct lwip_select_cb * select_cb) (/home/bird/deako/goat/dependencies/espressif/esp-idf/components/lwip/lwip/src/api/sockets.c:1865)
lwip_select(int maxfdp1, _types_fd_set * readset, _types_fd_set * writeset, _types_fd_set * exceptset, struct timeval * timeout) (/home/bird/deako/goat/dependencies/espressif/esp-idf/components/lwip/lwip/src/api/sockets.c:2183)
telnet_deako_check_for_connection() (/home/bird/deako/goat/dependencies/espressif/esp-idf/components/esp32/task_wdt.c:79)
prv_telnet_handle_connection() (/home/bird/deako/goat/dependencies/espressif/esp-idf/components/esp32/task_wdt.c:79)
vPortTaskWrapper(TaskFunction_t pxCode, void * pvParameters) (/home/bird/deako/goat/dependencies/espressif/esp-idf/components/freertos/port.c:143)
```

sebirdman, 2020-12-07:
@Alvin1Zhang I've done some further digging on this and have found it to primarily be an issue when you try to close a currently poll/select'd socket.

If i make my socket only poll for some short time, and ensure that the close does not happen during the polling/selecting timeframe, then this does not occur.

david-cermak, 2020-12-17:
@sebirdman Could you please share some parts of your code where the issue could be reproducible (ideally a minimal failing example)?
I've been trying to exercise closing listener's socket while two other threads polling on it, including disconnects from WiFi, but no issue so far. Here's my test code based on the `tcp-server` example: https://gist.github.com/david-cermak/19c1b9eba361accc54113b9b98c9cb75 (two threads simultaneously polling one or both sockets while another thread closing one or the other socket after a random delay; both client & server connected to a softAP which restarts after a random delay)

sebirdman, 2020-12-17:
@david-cermak thanks for the response

I also cannot reproduce with the simple example you've got there (with some small changes to make it more like what i've got) I'm beginning to suspect that like the other ticket you responded to: https://github.com/espressif/esp-idf/issues/6283 there may be some memory corruption going on here. 

i'll see if i can narrow things down further and if i can , will get you a small example.

sebirdman, 2020-12-18:
@david-cermak I'm beginning to dig into the corruption/error a bit more and i've identified that the `scb` variable mentioned in the initial post is coming back as 0xfefefefe, which appears to mean that the heap has been freed and we're trying to use it after free.

not sure where this would be happening, but it certainly doesn't seem to be tied directly to the opening or closing of sockets.

sebirdman, 2020-12-18:
I've finally figured out what i was doing wrong, heres my order of events and why it was breaking things:

1) so lwip_select and lwip_poll create a callback that it registers with a linked list in the socket.c file
2) after creating the callback, they wait on the semaphore for things to happen
3) that callback exists in STACK for the task that is doing the polling
4) we were deleting the task, and then trying to close the socket, this freed the callback in the list -> turning on heap memory debugging lead to that pointer resulting in 0xfefefefe

closed 2020-12-18 by sebirdman