espressif/esp-idf #6206: ethernet example cannot set GPIO of SPI of esp32s2

nopnop2002, 2020-12-04:
```
freeRTOS version:V10.2.1
NEWLIB version:3.0.0
lwIP version:2-1-3-0
ESP-IDF version:v4.3-dev-1901-g178b122c1-dirty
Problem Description
mDNS does not return a response after 10 minutes.
```

## Problem Description

ethernet example cannot set GPIO of SPI of esp32s2

## Steps to reproduce

1.Step1

Run the menuconfig in [this](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic) example using esp32s2.

2.Step2
In esp32s2, it is necessary to set these GPIOs, but they cannot be set because they are out of range.

[image]

```
        config EXAMPLE_DM9051_SCLK_GPIO
            int "SPI SCLK GPIO number"
            range 0 33
            default 19
            help
                Set the GPIO number used by SPI SCLK.

        config EXAMPLE_DM9051_MOSI_GPIO
            int "SPI MOSI GPIO number"
            range 0 33
            default 23
            help
                Set the GPIO number used by SPI MOSI.

        config EXAMPLE_DM9051_MISO_GPIO
            int "SPI MISO GPIO number"
            range 0 33
            default 25
            help
                Set the GPIO number used by SPI MISO.

        config EXAMPLE_DM9051_CS_GPIO
            int "SPI CS GPIO number"
            range 0 33
            default 22
            help
                Set the GPIO number used by SPI CS.
```

Alvin1Zhang, 2020-12-04:
Thanks for reporting, we will look into.

suda-morris, 2020-12-04:
Ah yes, the GPIO number limitation is still the ESP32 version, maybe we need to update it. For now, you can change the example Kconfig to expand the GPIO range or just override the Kconfig value in example main.c

closed 2021-04-27 by espressif-bot