espressif/esp-idf #6207: Show total time needed for compilation / build.

kifantidis, 2020-12-04:
Hello there !

I would like to request a feature. I would like to be able to see the total time taken for my project's compilation/ build.
I'm working with eclipse IDE + idf-eclipse pluggin. 
I've done some search before posting this and I've found nothing related to this. I've tried also to see if there is a way for me to implement it inside esptool.py but I'm not that familiar with the compilation tools. 

Best regards, K.Ifantidis.

Alvin1Zhang, 2020-12-07:
Thanks for raising this feature request.

ESP-Marius, 2022-08-31:
This was implemented in the idf-eclipse plugin with https://github.com/espressif/idf-eclipse-plugin/pull/162

If building with commandline then there are several options to achieve this, e.g. `time idf.py build` on linux, so I dont think we will be adding this to esp-idf itself.

Marking this as closed.

closed 2022-08-31 by ESP-Marius