espressif/esp-idf #6211: UART RS485 half-duplex intermittent DE line

MightyPork, 2020-12-05:
Platform:
- IDF v4.1
- ESP32

I'm investigating a bug with the DE line in UART RS485 mode. A customer complained of losing data, I'm not exactly sure what was happening. Nonetheless...

I discovered that the DE line becomes intermittent when the baud rate is roughly above 200000 bps. 9600, 150000 and 115200 are OK. It doesn't make any sense to me, so I thought it's worth a report. Maybe it's a driver bug? Or something in the hardware peripheral?

**Here are some of the waveforms:**

<details>
  <summary>Oscilloscope screenshots</summary>

## 115200, all OK
[image]
9600 is also OK.

## 200000, groups of 2 bytes
[image]

## 250000, each byte sent separately
[image]

## 500000, increasingly strange
[image]
  
The gap has 40us.

</details>

Excerpt from the config:

```c
    uart_config_t uart_config = {
        .baud_rate = 200000,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
        .source_clk = UART_SCLK_REF_TICK
    };
    ESP_ERROR_CHECK(uart_param_config(uart_num, &uart_config));
    ESP_ERROR_CHECK(uart_driver_install(uart_num, 512, 512, 10, &event_queue, 0));
    ESP_ERROR_CHECK(uart_set_mode(UART_NUM_2, UART_MODE_RS485_HALF_DUPLEX));
```

Bytes are sent using `uart_write_bytes()` from a heap allocated buffer.

appreciate any ideas

AxelLin, 2020-12-05:
.source_clk = UART_SCLK_REF_TICK can only support up to 115200 baudrate.
Use .source_clk = UART_SCLK_APB for higher baudrate.

MightyPork, 2020-12-05:
Thanks, unfortunately that did not change anything.

Here is 1Mbps with UART_SCLK_APB:

[image]

same problem down to 200000 baud, like before.

Here is my sdkconfig, in case something is wrong there:

<details>
  <summary>SDKCONFIG</summary>
  
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
# CONFIG_SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS is not set
# end of SDK tool configuration

#
# Build type
#
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
CONFIG_APP_BUILD_GENERATE_BINARIES=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_APP_BUILD_USE_FLASH_SECTIONS=y
# end of Build type

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16
# end of Application manager

#
# Bootloader config
#
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_DEBUG is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF is not set
# CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
[... 1093 lines cut ...]
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options

```
  
</details>

Alvin1Zhang, 2020-12-07:
@MightyPork Thanks for the detailed report and sharing the waveforms, we will look into.

alisitsyn, 2020-12-07:
Hello @MightyPork,

Thank you for the issue and good details. The UART driver uses the interrupt driven RS485 control using TX_EMPTY and TX_DONE interrupt. This approach was selected because the automatic direction control feature on ESP32 is available only on UART0 and does not work on other UARTs. The interrupts have the latency time up to ~10us that may prevent to process packets properly on higher speed. There is the kconfig value CONFIG_UART_ISR_IN_IRAM to place the UART interrupt handler into IRAM that can help to decrease the latency. In your config the interrupt handler is placed in FLASH memory. If your code uses tasks with nvs, spi_flash access API this may increase latency even more because these functions disable the CPU cache.

I investigate the issue, try to check solutions and will provide the ideas/solutions little bit later. Please try the key above for now to decrease latency for now.

Thanks.

AxelLin, 2020-12-07:
> Hello @MightyPork,
>
> Thank you for the issue and good details. The UART driver uses the interrupt driven RS485 control using TX_EMPTY and TX_DONE interrupt. This approach was selected because the [...]

Hi @alisitsyn
I'm wondering why CONFIG_UART_ISR_IN_IRAM is disabled by default?
Is there any drawback with CONFIG_UART_ISR_IN_IRAM=y?

alisitsyn, 2020-12-07:
Hi @AxelLin,

The option is disabled for the automatic test application to decrease IRAM size. The test application for v4.1 consumes a lot of IRAM and it cause errors. This is drawback of this option.

MightyPork, 2020-12-07:
Folks, I have to apologize for blaming the UART driver too quickly.

I discovered that a library we were using was sending the frame in multiple small pieces. This made delays in the driver visible as RTS (DE) gaps.

Now I have an explanation why this problem was not apparent at slow baud rates: The FIFO was still processing the last chunk when a new one was added, so the driver / peripheral never saw the FIFO empty and did not toggle DE.

I added a buffer to collect the message and send it in one piece. The waveforms are looking much better now.

*Some interesting observations:*

- The impact of `CONFIG_UART_ISR_IN_IRAM=n` is only a delay of 15us between RTS and the first start bit - at 1 Mbps. I think that's negligible if it means we get more RAM.

- UART works fine up to (tested) 1Mbps with UART_SCLK_REF_TICK - I will still keep ABP, but it does not seem to matter.
  
  So this is probably a myth:
  > .source_clk = UART_SCLK_REF_TICK can only support up to 115200 baudrate

  We are using `CONFIG_FREERTOS_HZ=1000`, maybe it's a problem with `CONFIG_FREERTOS_HZ=100`.

- Bulk of the delays I was seeing was caused by the RTS (DE) switching in the driver. 
  - When I changed UART mode to APP_CTRL and did *not* toggle the RTS (for a test), the time needed to send the frame has shrunk dramatically. That is *before* I fixed our bug with sending the frame as many small pieces. 
  - Adding manual `uart_set_rts()` calls made it slow again. Something to do with the critical sections? I'm not sure.

Again thanks for the help

closed 2020-12-07 by MightyPork

alisitsyn, 2020-12-08:
@MightyPork,

> Bulk of the delays I was seeing was caused by the RTS (DE) switching in the driver.

The ESP32 chip has an error in UART peripheral and is not able to control RTS line automatically in RS485 mode. There is workaround to mitigate this issue in UART driver which uses the TX_DONE and TX_EMPTY interrupts to de/assert RTS line manually and flush buffer to clear zero bytes related to switching of transceiver. This adds delays for interrupt latency. However the hardware of UART0 on ESP32 allows to control the DE pin of transceiver over DTRN_OUT signal. Other UARTs of ESP32 do not support this feature. This feature is available in ESP32S2 chip UARTs but is not supported by ESP-IDF UART driver yet. I can propose you to try this feature. However no any guaranties while it is not officially supported.
[image]
The UART should be configured in APP_CTRL mode and the code below will allow to configure the DTRN_OUT pin of ESP32:
```
#include "driver/gpio.h"
#include "soc/gpio_sig_map.h" //for DTRN pin configuration
#include "esp_rom_gpio.h"
// The DTRN_OUT hw signal will be assigned to this pin in RS485 Half-Duplex Mode (Only on UART_NUM_0)
#define ECHO_TEST_DTR  (19) // The GPIO19 shall be used to control RS485 transceiver DE pin

static void uart_conf_dtrn_out(int dtr_io_num)
{
    PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[dtr_io_num], PIN_FUNC_GPIO);
    gpio_set_direction(dtr_io_num, GPIO_MODE_OUTPUT);
    esp_rom_gpio_connect_out_signal(dtr_io_num, U0DTR_OUT_IDX, 0, 0);
}
....
// app_main: uart initialization code addition:
uart_set_pin(EX_UART_NUM, DMX_TEST_TXD, DMX_TEST_RXD, DMX_TEST_RTS, UART_PIN_NO_CHANGE);
uart_conf_dtrn_out(DMX_TEST_DTR); // Configure the GPIO DMX_TEST_DTR pin number for DTRN_OUT signal on UART0
```