espressif/esp-idf #6212: Build with CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT set fails

crwolff, 2020-12-05:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:     ESP32-S2-Saola-1
- Kit version (for WroverKit/PicoKit/DevKitC): n/a
- Module or chip used:  ESP32-S2-WROOM
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-1901-g178b122
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- (Windows only) environment type: n/a
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

Following the steps in [https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html#signed-app-verification-without-hardware-secure-boot](url) to enable signature checking without enabling secure boot. 

I get the following error and the build fails:
 
~/esp/esp-idf/components/bootloader_support/src/esp_image_format.c: In function 'verify_secure_boot_signature':
~/esp/esp-idf/components/bootloader_support/src/esp_image_format.c:813:5: error: 'sig_block' is used uninitialized in this function [-Werror=uninitialized]
     bootloader_munmap(sig_block);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~

### Expected Behavior

Build should succeed

### Actual Behavior

Build fails

### Steps to reproduce

1. cd ~/esp/esp-idf/examples/get-started/hello_world
2. idf.py set-target esp32s2
3. idf.py menuconfig
    a. Set Security Features -> Require signed app images
    b. Save and exit
4. openssl ecparam -name prime256v1 -genkey -noout -out secure_boot_signing_key.pem
5. idf.py build

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

The issue can be reproduced with the hello_world example.

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/5647266/sdkconfig.zip)

Alvin1Zhang, 2020-12-07:
Thanks for reporting, we will look into.

Darsh-Dev, 2021-03-18:
Hi 

I am using "**ESP32-S2-Saola-1_V1.2**" kit.
I am also facing similar kind of issue. 
I refer below link to enable CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html#signed-app-verification-without-hardware-secure-boot

I closely observe the doc and found that this feature is available for **ESP32** family, when i switch the documents from **ESP32** to **ESP32S2** family then i can't found this type of feature is there, in **ESP32S2** i only see SecureBootV2 doc  https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/secure-boot-v2.html

**My question is** 
Is there any way we can enable CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT feature in ESP32S2 ?

projectgus, 2021-03-21:
Hi @crwolff & @Darsh-Dev,

Sorry for the very long delay in someone getting back to you.

There is currently no support for verifying apps without Secure Boot in ESP32-S2. As mentioned, the docs for Secure Boot in ESP32-S2 are here:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/secure-boot-v2.html

The config option was incorrectly left available on ESP32-S2, this is a bug.

Regarding support for this feature, there are two independent options available on ESP32 when not using secure boot:  "Bootloader verifies app signatures" & "Verify app signature on update"

## Bootloader verifies app signatures (no Secure Boot)

We don't plan to support this option on ESP32-S2, for the following reason:

In ESP32, the Secure Boot V1 scheme makes it difficult to update the bootloader after enabling Secure Boot. This is why we added this support even though it reduces the overall security of the system (an attacker with physical access can rewrite the bootloader and remove any checks if Secure Boot is disabled).

With Secure Boot V2, which is the Secure Boot scheme for ESP32-S2, both the bootloader and the app can be signed with the same trusted signing key and both can be updated by signing a new binary and flashing it. For this reason we considered that "Bootloader verifies app signatures" without Secure Boot was not a good option to support, as enabling this option is very similar in complexity to enabling full Secure Boot but this option fails to protect against most of the threats where Secure Boot is required.

Do you have a use case for verifying the app on boot, but not using Secure Boot? If you do then please let us know, and we'll consider it. 

## Verify app signature on update

This option enables signature verification in the OTA app_update APIs, but not during the boot process. This allows protection against an attacker who is able to compromise the OTA update process but is not able to physically write a new binary to the flash.

This option was not supported for ESP32-S2 in ESP-IDF v4.2, but support will be added starting from ESP-IDF v4.3. The support is not currently available on GitHub will be merged soon, this issue will be updated when it is available.

crwolff, 2021-03-22:
Verify app signature on update without secure boot is sufficient for my application. This prevents a malicious actor from pushing a firmware update through the WiFi. I do not want to enable secure boot since this could potentially brick the hardware due to hardware or software failure.

Darsh-Dev, 2021-03-22:
> Hi @crwolff & @Darsh-Dev,
> 
> Sorry for the very long delay in someone getting back to you.
> 
> There is currently no support for verifying apps without Secure Boot in ESP32-S2. As mentioned, the docs for Secure Boot in ESP32-S2 are here:
> 
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/secure-boot-v2.html
> 
> The config option was incorrectly left available on ESP32-S2, this is a bug.
> 
> Regarding support for this feature, there are two independent options available on ESP32 when not using secure boot: "Bootloader verifies app signatures" & "Verify app signature on update"
> 
> ## Bootloader verifies app signatures (no Secure Boot)
> We don't plan to support this option on ESP32-S2, for the following reason:
> 
> In ESP32, the Secure Boot V1 scheme makes it difficult to update the bootloader after enabling Secure Boot. This is why we added this support even though it reduces the overall security of the system (an attacker with physical access can rewrite the bootloader and remove any checks if Secure Boot is disabled).
> 
> With Secure Boot V2, which is the Secure Boot scheme for ESP32-S2, both the bootloader and the app can be signed with the same trusted signing key and both can be updated by signing a new binary and flashing it. For this reason we considered that "Bootloader verifies app signatures" without Secure Boot was not a good option to support, as enabling this option is very similar in complexity to enabling full Secure Boot but this option fails to protect against most of the threats where Secure Boot is required.
> 
> Do you have a use case for verifying the app on boot, but not using Secure Boot? If you do then please let us know, and we'll consider it.
> 
> ## Verify app signature on update (no Secure Boot)
> This option enables signature verification in the OTA app_update APIs, but not during the boot process. This allows protection against an attacker who is able to compromise the OTA update process but is not able to physically write a new binary to the flash.
> 
> This option was not supported for ESP32-S2 in ESP-IDF v4.2, but support will be added starting from ESP-IDF v4.3. The support is not currently available on GitHub will be merged soon, this issue will be updated when it is available.

Darsh-Dev, 2021-03-22:
Hi @projectgus,

Thank you so much for the reply.
my requirement is also  "Verify app signature on update without secure boot" but yet this code is in progress and not supported in this release i have enable secure boot v2 for my ESP32-S2 in native-ota-example and **i face one another issue while ota update start "native_ota_example: Error: SSL data read error" with secure boot v2 enable**. Below is the steps i have followed.

- In menuconfig “Security Features” set “Enable hardware Secure Boot in bootloader” to enable Secure Boot.
- Change partition_table offset to 0x10000 as secure boot loader size is large.
- create self sign-certificate and copy into server_certs/ directory.
- Build the native_ota_example
- Flash bootloader as secure boot enable need to flash bootlaoder independently.
- Flash partition table, signed native-ota.bin using `$idf.py -p (PORT) flash`   command. and reset the board.
- It run correctly bootloader is verify the native-ota.bin signature also fine.
- Now i sign the hello_word.bin using `espsecure.py sign_data --version 2 --keyfile ./my_signing_key.pem --output ./image_signed.bin image-unsigned.bin
` command
- use the above signed binary for the ota update.
-  start https server at host PC and esp32 give below logs

> I (7859) native_ota_example: Starting OTA example
I (7859) native_ota_example: Running partition type 0 subtype 0 (offset 0x00020000)
I (10119) native_ota_example: Writing to partition subtype 16 at offset 0x120000
I (10119) native_ota_example: New firmware version: v4.4-dev-479-g1067b2870-dirty
I (10119) native_ota_example: Running firmware version: 1
I (10129) native_ota_example: esp_ota_begin succeeded
D (16849) HTTP_CLIENT: esp_transport_read returned:-1 and errno:128 
**E (16849) native_ota_example: Error: SSL data read error**
E (16849) native_ota_example: Exiting task due to fatal error...

Please help me how can i resolve this issue for native_ota example.
Please find attached full log for more details.

**Note:** I have tried above steps secure boot v2 with **simple_ota example** then it's verify the hello_word.bin signature and update the firmware successfully.
Also if i use native_ota example without secure boot v2 then it working fine means **there is some bug with native_ota + secure boot v2 combination.** 
 
[native_ota_secure_boot_v2.log](https://github.com/espressif/esp-idf/files/6182954/native_ota_secure_boot_v2.log)

projectgus, 2021-03-24:
@crwolff @Darsh-Dev Glad to hear that the new option will meet your requirements, will update when this is available.

@Darsh-Dev Sorry to hear that native_ota is not working. The error you're seeing here seems to be coming from the SSL transport layer and not to do with the app_update APIs or Secure Boot itself. In any case, it's not related to this issue - if it's still not working then please open a new issue and fill in the issue template plus provide this information again. Please also give as many details as possible about the HTTPS server setup and any server logs in the new issue, as well as a log from a successful update (without secure boot v2) if that is still working. Thanks.

Darsh-Dev, 2021-04-14:
Is these feature (CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT) is enabled for ESP32-S2 platform?

mahavirj, 2021-04-29:
@Darsh-Dev 

> Is this feature (CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT) is enabled for ESP32-S2 platform?

Yes as part of current `master` or `release/v4.3` branch. See documentation at https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/security/secure-boot-v2.html#how-to-enable-signed-app-verification

projectgus, 2021-05-04:
Yes, support was added to master in commit 95564b4687f0a331afc82a7828804e350393a0cf and release/v4.3 in commit 46e85ed0210ac8d2c1949542ed1122efd2fb0cd1. The [v4.3-beta3 prerelease](https://github.com/espressif/esp-idf/releases/v4.3-beta3) has this feature, as will the final v4.3 release.

Sorry that this wasn't automatically linked to GitHub when it merged.

closed 2021-05-04 by projectgus