espressif/esp-idf #6213: Support for clang static analyzer (scan-build)

Harvie, 2020-12-05:
Hello,
i would love to be able to use scan-build (clang based static analyzer) with ESP-IDF projects.

But it turns out that when i do `. export.sh` and `scan-build idf.py build` the build process gets somewhat mangled... Have you considered to make esp-idf buildsystem compatible with scan-build or even include prebuilt scan-build binary directly to esp-idf, so i can just do `idf.py scan-build` to get my project properly analyzed? These tools are priceless once the project starts to grow. I believe everyone can benefit from having this.

https://clang-analyzer.llvm.org/scan-build.html

david-cermak, 2020-12-07:
Hi @Harvie 

I think you can use `scan-build` with IDF build system, if you generate "Unix makefiles".
Go to the project directory and instead of `scan-build idf.py build` run
``` bash
mkdir build-scan && cd build-scan
cmake -G "Unix Makefiles" ..
scan-build make
```

Yes, it would be a great benefit to have some `static-analysis` target in `idf.py` so users can analyze their codebase. Thanks for this feature request. We cannot provide this feature at the moment, but can share some pointers on the static analysis used internally, which is also based on llvm:

1) Build espressif's platform specific clang-tidy (https://github.com/espressif/llvm-xtensa)
``` bash
 mkdir -p build && cd build && cmake ../llvm/llvm -DCMAKE_BUILD_TYPE=Release \
  -DLLVM_EXTERNAL_CLANG_SOURCE_DIR=...llvm/clang \
  -DLLVM_TARGETS_TO_BUILD="" \
  -DLLVM_EXPERIMENTAL_TARGETS_TO_BUILD=Xtensa \
  -DLLVM_EXTERNAL_CLANG_TOOLS_EXTRA_SOURCE_DIR=...llvm/clang-tools-extra/ \
  && make -j 4 clang-tidy
```
2) Run `idf.py build` to generate `build-commands.json` in the build directory
3) Run `clang-tidy` referencing the json file (or using `run-clang-tidy.py`)

Harvie, 2020-12-07:
it kinda done something when i use idf.py build, but it seemed to call my system-wide gcc instead of xtensa toolchain from esp-idf. i also tried to force xtensa gcc by providing path to gcc and g++ as scan-build parameter, but it didn't worked anyway...

I will try to generate the makefiles to see if it changes something, but i am not really sure it will help. i think the problem is in the environment rather than in the build process.

Harvie, 2020-12-07:
```
$ cmake -G "Unix Makefiles" ..
CMake Error at CMakeLists.txt:86 (include):
  include could not find load file:

    /tools/cmake/project.cmake

-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:90 (target_compile_definitions):
  Cannot specify compile definitions for target "grbl.elf" which is not built
  by this project.

CMake Error at CMakeLists.txt:91 (target_compile_definitions):
  Cannot specify compile definitions for target "grbl.elf" which is not built
  by this project.

CMake Error at CMakeLists.txt:134 (target_add_binary_data):
  Unknown CMake command "target_add_binary_data".

-- Configuring incomplete, errors occurred!
See also "/home/harvie/Temp/grblHAL-test-make/drivers/ESP32/build-scan/CMakeFiles/CMakeOutput.log".
```

And in the log:

```
[harvie@anemophobia build-scan]$ grep Compiler /home/harvie/Temp/grblHAL-test-make/drivers/ESP32/build-scan/CMakeFiles/CMakeOutput.log
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"
The C compiler identification is GNU, found in "/home/harvie/Temp/grblHAL-test-make/drivers/ESP32/build-scan/CMakeFiles/3.19.1/CompilerIdC/a.out"
Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
```

It seems to use /usr/bin/cc  and /usr/bin/c++  i am not sure this is right...

igrr, 2020-12-07:
@Harvie have you executed `. export.sh` to set up an environment?
Could you please run that again, then confirm that environment variables have been set up. `echo $IDF_PATH` should print the path to IDF, and `echo $PATH` should contain some directory related to `xtensa-esp32-elf-gcc`.
If these environment variables are set, delete the build directory (or wipe CMakeCache.txt) and run cmake again.

Harvie, 2020-12-07:
> @Harvie have you executed . export.sh to set up an environment?

yes of course. it builds without scan-build. but i have to confess that i've tried only ESP-IDF, because i tried on project that is not compatible with latest esp-idf...

david-cermak, 2020-12-08:
I think you've missed the `IDF_PATH` part. 

Assuming your project includes the idf cmake file in a similar way as for example https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/CMakeLists.txt#L5 
From the posted log it seems like the expression `$ENV{IDF_PATH}` evaluated to an empty var.
``` bash
CMake Error at CMakeLists.txt:86 (include):
  include could not find load file:
    /tools/cmake/project.cmake
```

Harvie, 2020-12-08:
scan-build does not affect that variable:

```
[harvie@anemophobia ESP32]$ echo $IDF_PATH
/home/harvie/esp/esp-idf
[harvie@anemophobia ESP32]$ scan-build bash
scan-build: Using '/usr/bin/clang-11' for static analysis
[harvie@anemophobia ESP32]$ echo $IDF_PATH
/home/harvie/esp/esp-idf
```

Harvie, 2020-12-08:
Ok, i don't know what changed, but i was able to generate the makefile and build the project using it.
However it still somehow overrided the compiler, so it didn't get analyzed by scan-build...

DCSBL, 2021-07-07:
Hello, I've tried to run clang-build with with the steps from https://github.com/espressif/esp-idf/issues/6213#issuecomment-739872479 and everything builded fine. I see that the app has been build and I got a `scan-build: No bugs found.` message at the end.

The weird part is that I've added a bug the code that should have been detected by the analyser. Is there something that I do wrong?

```c
// <includes>

struct S
{
    int x;
};

void f(struct S s)
{
    printf("%d\n", s.x);
}

void test()
{
    struct S s;
    f(s); // Should trigger a passed-by-value warning, see http://web.cs.ucla.edu/~tianyi.zhang/tutorial.html 'Example 1'
}

void app_main()
{   
    test();
}
```

Harvie, 2021-07-07:
> The weird part is that I've added a bug the code that should have been detected by the analyser. 

I think that scan-build works by diverging executables (perhaps by replacing $PATH ??) so that analyzer wrapper is used instead of calling compiler directly. Problem is that if you have complex build system like ESP-IDF, it might create makefiles using absolute paths or force binary paths in some other way itself. That way no files are actually processed by analyzer.

You can see similar result when i wrap `true` (true = do-nothing) command in scan-build:

```
$ scan-build true
scan-build: Using '/usr/bin/clang-12' for static analysis
scan-build: Analysis run complete.
scan-build: Removing directory '/tmp/scan-build-2021-07-07-113459-959722-1' because it contains no reports.
scan-build: No bugs found.
```

igrr, 2021-07-07:
I would recommend using `build/compile_commands.json` file as in input to clang-tidy, rather than the `scan-build` wrapper. Please see -p argument described in https://clang.llvm.org/extra/clang-tidy/. This is more reliable than the wrapper.

julian-weinert, 2021-08-10:
@igrr I'm currently trying to get this to work. Did you run it successfully already and could provide and example?
It's basically working, but I'm still missing something. It doesn't seem to find system headers, I'm getting `[...]/FreeRTOS.h:77:10: error: 'sys/reent.h' file not found`

david-cermak, 2021-08-16:
@julian-weinert You have to add the path to system headers manually. Just insert these flags to your `compile_commands.json`:

```
-D__XTENSA__ -isystem{}
```

where {} is the root include dir of your toolchain, which could be 
* `...espressif/xtensa-esp32-elf-clang/xtensa-esp32-elf/include/` if you're using espressif xtensa clang
* `...espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include` for the gcc

herculanodavi, 2022-02-18:
@CvRXX i had this same error, but if you run `. {YOUR_IDF_PATH}/export.sh` before running clang, it should work.

gchromik, 2022-04-28:
@herculanodavi  What exactly have you done to get this work? I tried to use steps from [this doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-clang-tidy.html)  but idf.py doesn't recognize command. When I use ```clang-tidy -p build/compile_commands.json main/main.c``` I get exactly the same error like Carlos. I have esp installed from  vscode plugin.

igrr, 2023-01-08:
Clang-tidy can now be launched via `idf.py clang-check`. Please see [the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-clang-tidy.html) for complete instructions.

closed 2023-01-08 by igrr

Harvie, 2023-01-09:
does the clang-tidy provide same results and sensitivity as scan-build ?

igrr, 2023-01-09:
It does, i'll link this answer (slightly outdated, but still essentially true): https://stackoverflow.com/a/70244681

herculanodavi, 2023-01-09:
> @herculanodavi What exactly have you done to get this work? I tried to use steps from [this doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-clang-tidy.html) but [...]

I implemented clang-tidy checks in the CI process of this toy example:
https://github.com/herculanodavi/esp-idf-template

there is also a bot that turns the clang-tidy warnings into pr comments:
https://github.com/herculanodavi/esp-idf-template/pull/1

herculanodavi, 2023-01-09:
@igrr  I see the `idf.py clang-check` is still under development and it might introduce breaking changes. 

- Does this mean it's "experimental" or should I expect it to always work?
- Do you have an idea of the `esp-idf` version that should freeze this interface?

igrr, 2023-01-09:
> there is also a bot that turns the clang-tidy warnings into pr comments:
> https://github.com/herculanodavi/esp-idf-template/pull/1

Very nice, thanks for sharing!

There is also a way to convert the clang-tidy code scanning results into [Github Code Scanning alerts](https://docs.github.com/en/code-security/code-scanning/automatically-scanning-your-code-for-vulnerabilities-and-errors/about-code-scanning), using a workflow like this: https://github.com/espressif/idf-extra-components/blob/25f60c0686c61d3b87689eef3a1214ebf62849ee/.github/workflows/clang-tidy.yml

Github Code Scanning alerts are shown in the "Security" tab of the repo (for collaborators only) and in PRs ([example](https://github.com/espressif/idf-extra-components/pull/96)).

> Does this mean it's "experimental" or should I expect it to always work?
> Do you have an idea of the esp-idf version that should freeze this interface?

There may be changes in clang-check usage between IDF minor releases. For example, in IDF v5.1 and later, the clang toolchain which you need to install using `idf_tools.py` is called `esp-clang` instead of `xtensa-clang` (since we have added support for RISC-V targets.)

Currently there are a couple of feature requests which may require us to change how `idf.py clang-check` command works (https://github.com/espressif/clang-tidy-runner/issues/25, https://github.com/espressif/clang-tidy-runner/issues/8).
However given that this tool is maintained outside of IDF, you should be able to install the specific version of the python package to pin the existing behavior. (E.g. `pip install pyclang==0.2.2`)