espressif/esp-idf #6214: GCC 10 Support

Harvie, 2020-12-05:
Hello,
i would like to use new fancy features available in GCC 10 for ESP32 development. Can you please adopt it soon?

Alvin1Zhang, 2020-12-07:
Thanks for raising this feature request.

mirko, 2021-02-12:
I tried to level up the basic esp32 related patchsets to latest (as of this wrtiting) GCC master/HEAD.
Bumped into no-rtti issues (fixed) and linker issues (not yet fixed), but first attempts of compiling the ESP SDK IDF code succeeded for various versions and configurations, so -fanalyzer is already happily dropping some interesting bits and pieces.
https://github.com/mirko/gcc/commits/wip_gcc11-for-esp32_basedon-85cefe22

galah92, 2021-09-10:
Would very much like that as well

rdeterre, 2021-11-26:
Since GCC 11 is out, should we update this feature request to jump to it directly?

igrr, 2021-11-26:
@rdeterre That's correct, currently work is ongoing to upgrade the toolchain to GCC 11.

strpbrk, 2022-04-04:
Is there any progress? Maybe I could assist in some way?
Thanks!

igrr, 2022-04-04:
Hi @strpbrk, thank you for the kind offer. We are in process of upgrading to GCC 11. Here's the rough status update:

1. We do have a draft of GCC 11 based toolchain which we are testing now.
2. We are fixing various build warnings which are produced in IDF with GCC 11 but not with older GCC versions. Our CI checks that every build succeeds with -Werror, so fixing the warnings is necessary before we can release GCC version upgrade. The recent commits for fixing the warnings are 7cdd1516, 68e5d9d585492163712b9cbb4a488abc82a0b795, https://github.com/espressif/esp-nimble/commit/e6cc38908c58a5d3590bff96a2904fc70d746346, 0d86151b82981b95fbec6d861b0364235b2acd71, e27f1331e48727b783a21c6d2da4900484eb21d8, 3ab9acddc131c7e520f09bed92dd6f34b588982f, 8bb06df68f8fb03d26f829efe1e18324b75335e8, https://github.com/espressif/esp-mqtt/commit/f83d2a911191934082700b9c64108f116bed0cdf. There still a few more warnings to fix.
3. In this toolchain version we are also upgrading Newlib C library to the latest release. This task is in progress right now.
4. We have also noticed a code size regression when using DWARF 5 debug format (used by default in GCC 11, compared to DWARF 4 in GCC8). It needs to be investigated and fixed, or we need to fall back to emitting DWARF 4.
5. GCC 11.2 (which we are upgrading to) has a regression related to designated initializers (https://gcc.gnu.org/bugzilla/show_bug.cgi?id=101767) which is going to be fixed in GCC 11.3. Probably not a blocker, worst case we will backport the fix to our branch.
6. Another large change which we are planning for the GCC 11 based toolchain release is to change the `uint32_t` type in Xtensa compiler from `unsigned int` (as it is now) to `unsigned long`. Upstream GCC uses `unsigned long` for `uint32_t` on Xtensa, RISC-V and other architectures. Both types are 32 bit wide, the change is just for consistency between Xtensa and RISC-V and also with the upstream. This change requires fixing lots of format strings in IDF codebase, since existing cases where `uint32_t` variables are formatted with `%u` or `%x` will now produce warnings.

As you see there is still some work to do, but I'm not sure which of the remaining parts we could ask you to help with. Maybe the DWARF 5 vs DWARF 4 code size issue? @antmak if you have any suggestion, please chime in.

strpbrk, 2022-04-04:
Thank you for detailed information.

I can try to tangle with the DWARF issue, and I can also submit other patches along the way.
However, I was not able to find a crosstool repository with gcc 11.2. Could you please point me to the right direction?
Thank you again.

antmak, 2022-04-14:
Hi! Ah sorry I missed to post a link here. The current state of GCC 11 toolchain I posted there https://github.com/espressif/crosstool-NG/issues/22 . There are links to branches, including IDF dev branch

closed 2022-06-11 by espressif-bot