espressif/esp-idf #6215: Network ioctls

szekelyisz, 2020-12-06:
**Is your feature request related to a problem? Please describe.**

I want to communicate with my devices through COAP via IPv4 multicast. I implemented the necessary support for libcoap, see [this discussion](https://github.com/obgm/libcoap/issues/566). Note here that since ESP-IDF doesn't define `WITH_LWIP`, libcoap can rely on LwIP's socket API to do the job so it has to follow BSD sockets standards.

The decision in libcoap is that the interface name will be passed to the function that joins the multicast group and it will use `ioctl()` to find information about the interface like IPv4 address and index, these are needed for `setsockopt()` to specify which interface to join the multicast group on. Since the ESP32 supports multiple network interfaces, it is critical that the developer can specify the interface in order to achieve the desired multicast setup. Typically `ioctl()` is used to find interface index and IPv4 address to pass to `setsockopt`. I implemented the required ioctls in LwIP. But in ESP-IDF, `ioctl()` is weak-aliased to `esp_vfs_ioctl()` so it will hijack all network-related ioctls as well, which will return an error due to invalid file descriptor in this case, since the FD refers to a socket instead of an open file.

**Describe the solution you'd like**

While overriding the weak alias by implementing `ioctl()` in user code would do the job for now, it would mean that anybody wanting to call LwIP-related ioctls will need to do just that. I'm looking for a more universal solution.

For instance, instead of weak-aliasing `ioctl` directly to `esp_vfs_ioctl`, create another weak alias named `esp_ioctl` for example that will route the call to either the VFS layer or LwIP depending on the request code. Most probably the list of request codes for LwIP has to be maintained manually over releases.

**Describe alternatives you've considered**

As `ioctl` is a standard call, and libcoap has to work on a variety of platforms, it's not possible to call `lwip_ioctl()` from there. As mentioned earlier, there's no way for libcoap to distinguish between Linux and LwIP for example since ESP-IDF doesn't define `WITH_LWIP`. That being said, I don't see any way to do this other than described above.

**Additional context**

I'd be happy to submit a pull request addressing this if a consensus is reached about how to implement this.

szekelyisz, 2020-12-07:
`CONFIG_VFS_SUPPORT_IO` allows you to get to LwIP's `ioctl()`. But you have to sacrifice all VFS weak aliases.

igrr, 2020-12-07:
@szekelyisz could you please describe how to reproduce the issue? VFS normally routes all the calls (read, write, ioctl, etc.) to the FS driver, which in the case of socket FDs is LWIP. If you are seeing that read and write work correctly but ioctl doesn't, it might be a bug somewhere in VFS.

szekelyisz, 2020-12-08:
@igrr Thanks for revisiting this. I'm not sure there's an issue here actually. I just learned that when you call `ioctl(...)`, you will actually call `esp_vfs_ioctl` because of the weak alias, and you can get to LwIP's ioctls by calling `lwip_ioctl`. This is true as long as `CONFIG_VFS_SUPPORT_IO` is enabled. If it's not, then it's the other way around. Since libcoap will have to call `ioctl` on LwIP, simply disabling this option did the trick for me.

This means that you can't use both VFS's ioctls and LwIP's ioctls at the same time callable as `ioctl(...)`. There might be a conflict between two components on this. This is true for all standard syscalls though, not just `ioctl`. In that case you can tell where to forward the call based on the request code, but for other calls you need to know somehow if the FD refers to an open VFS file or a network socket. Since these assigned independently AFAIK, they can even collide so no way to tell whether the VFS function or the LwIP counterpart should be called...

igrr, 2020-12-09:
> Since these assigned independently AFAIK, they can even collide so no way to tell whether the VFS function or the LwIP counterpart should be called...

This is not the case, for LWIP there is a separate range of file descriptors reserved. VFS layer knows if the FD falls into this region and forwards the call to LWIP just as it does for any other VFS.

> This is true for all standard syscalls though, not just ioctl

Could you please tell how to reproduce this? I have run a simple app that creates a socket and calls ioctl, put a breakpoint on lwip_ioctl, and I see that the call is correctly passed to LWIP (ioctl aliased to esp_vfs_ioctl -> calls lwip_ioctl_r_wrapper -> calls lwip_ioctl). This was with CONFIG_VFS_SUPPORT_IO=y.
If you are seeing different behavior, please let me know how to replicate it, and we will investigate.

szekelyisz, 2020-12-11:
@igrr You're right, I didn't dig deep enough. Calling `ioctl(...)` with `CONFIG_VFS_SUPPORT_IO=y` will call `esp_vfs_ioctl` which will end up in `lwip_ioctl`. Nothing to see here.

closed 2020-12-11 by szekelyisz