espressif/esp-idf #6218: HFP AG - wrong size of outgoing data buffer.

ypiotrp, 2020-12-07:
## Environment

- Development Kit:      none
- Kit version:          does not apply
- Module or chip used:  ESP32-WROOM-32D
- IDF version           v4.1-3-g314379eda && v4.2-rc
- Build System:         idf.py
- Compiler version:     8.4.0
- Operating System:     Windows
- Environment type:     PowerShell
- Using an IDE?:        No (only PowerShell with ESP-IDF scripts)
- Power Supply:         USB

## Problem Description
After applying the patch from #4967 on ESP-IDF v4.1 the link between ESP32 and the HFP headset started to work. My project is based on HFP-AG example from the ESP-IDF. I use HCI as a SCO data path, turned off WBS and I changed the `BR/EDR Sync(SCO/eSCO) Max Connections to 1`. Unfortunately after changing the headset for a different model it seems like memory allocated for the `p_buf` buffer submitted to `bt_app_hf_outgoing_cb(uint8_t *p_buf, uint32_t sz)` function is less than `sz`.

The ESP-IDF rev 4.2-rc doesn't need the patch.

### Expected Behavior
Audio data being seemlessly transferred to the HFP Client device.

### Actual Behavior
Using other device after the first call to `bt_app_hf_outgoing_cb` I got core dump. After analysys I made some changes in the ESP_IDF functions. The changes are presesnt in the patch I added as a attachment to this issue. If the value of `sz` in `bt_app_hf_outgoing_cb` is equeal to 120 the the example should work. When I connected to a different device with `sz` equeal to 60 the system crashed.

### Steps to reproduce

1. Pull the ESP-IDF from the repo in rev 4.1 or 4.2-rc.
    * In case of using rev 4.1 apply the patch from #4967.
2. Enter `idf.py menuconfig` and:
    1. In Component config > Bluetooth > Bluetooth controller:
        * Set BR/EDR Sync (SCO/eSCO) Max Connections to 1,
        * Set BR/EDR Sync (SCO/eSCO) default data path to HCI,
    2. In Component config > Bluetooth > Bluedroid Options:
        * Set Hands Free/Handset Profile audio(SCO) data path to HCI,
        * Disable Wide Band Speech
    3. Save the changes and exit `menuconfig`. 
3. In `app_hf_msg_set.c` uncomment line 16 and provide the Bluetooth address of the device you want to connect with.
4. Build the firmware and flash the ESP32.
5. Connect to your headset using `hf con;` and `hf cona;` commands in UART interface.
6. Use different HFP headsets until the project crashes. For me the Bluetooth headphones PaMu Slide Mini worked fine but with the Microchip RN-52 Evaluation Kit it didn't (see crash logs below).

### Code to reproduce this issue
Just original HFP-AG example. I attached my temporary fix as a git patch file.

## Debug Logs

```
W (7011) BT_APPL: new conn_srvc id:5, app_id:0
I (7011) BT_APP_HF: APP HFP event: CONNECTION_STATE_EVT
I (7021) BT_APP_HF: --connection state CONNECTED, peer feats 0x0, chld_feats 0x0
I (7201) BT_APP_HF: APP HFP event: CIND_RESPONSE_EVT
I (7201) BT_APP_HF: --CIND Start.
I (7261) BT_APP_HF: APP HFP event: CONNECTION_STATE_EVT
I (7261) BT_APP_HF: --connection state SLC_CONNECTED, peer feats 0xbf, chld_feats 0x4010
I (7301) BT_APP_HF: APP HFP event: NREC_RESPONSE_EVT
I (7301) BT_APP_HF: --NREC status is: NREC DISABLE.
I (7351) BT_APP_HF: APP HFP event: VOLUME_CONTROL_EVT
I (7351) BT_APP_HF: --Volume Target: SPEAKER, Volume 10
E (7681) BT_BTM: tBTM_SEC_DEV:0x3ffc2104 rs_disc_pending=0

I (7691) BT_APP_HF: APP HFP event: CLCC_RESPONSE_EVT
I (7691) BT_APP_HF: --Calling Line Identification.
I (7811) BT_APP_HF: APP HFP event: UNKNOW_AT_CMD
I (7811) BT_APP_HF: --UNKOW AT CMD: +CSRSF=0,0,0,1,0,0,2
E (11231) CNSL: Command [hf cona;]
Connect Audio
W (11231) BT_BTM: BTM Remote does not support 2-EDR eSCO
W (11231) BT_BTM: BTM Remote does not support 3-EDR eSCO
I (11231) BT_APP_HF: APP HFP event: AUDIO_STATE_EVT
I (11241) BT_APP_HF: --Audio State connecting
E (11271) BT_BTM: btm_sco_connected, handle 180
I (11271) BT_APP_HF: APP HFP event: AUDIO_STATE_EVT
I (11271) BT_APP_HF: --Audio State connected
W (11281) BT_APPL: bta_ag_sco_read_cback: status(2)
CORRUPT HEAP: multi_heap.c:246 detected at 0x3ffbaad8

abort() was called at PC 0x40092943 on core 0
0x40092943: multi_heap_assert at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) merge_adjacent at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/heap/multi_heap.c:246

Backtrace:0x4008f2a3:0x3ffcfcc0 0x4008f941:0x3ffcfce0 0x40093122:0x3ffcfd00 0x40092943:0x3ffcfd70 0x40092bf0:0x3ffcfd90 0x40081442:0x3ffcfdb0 0x4009314d:0x3ffcfdd0 0x400f04a3:0x3ffcfdf0 0x400f0d71:0x3ffcfe10 0x400efa36:0x3ffcfe30 0x400efb8e:0x3ffcfe50 0x400f167a:0x3ffcfe70 0x400edc44:0x3ffcfe90 0x4008f949:0x3ffcfeb0
0x4008f2a3: panic_abort at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/esp_system/panic.c:330

0x4008f941: esp_system_abort at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/esp_system/system_api.c:106

0x40093122: abort at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/newlib/abort.c:46

0x40092943: multi_heap_assert at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) merge_adjacent at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/heap/multi_heap.c:246

0x40092bf0: multi_heap_free_impl at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/heap/multi_heap.c:565

0x40081442: heap_caps_free at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/heap/heap_caps.c:305

0x4009314d: free at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/newlib/heap.c:47

0x400f04a3: bta_ag_sco_event at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/bt/host/bluedroid/bta/hf_ag/bta_ag_sco.c:774

0x400f0d71: bta_ag_ci_sco_data at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/bt/host/bluedroid/bta/hf_ag/bta_ag_sco.c:1669

0x400efa36: bta_ag_sm_execute at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/bt/host/bluedroid/bta/hf_ag/bta_ag_main.c:910 (discriminator 2)

0x400efb8e: bta_ag_hdl_event at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/bt/host/bluedroid/bta/hf_ag/bta_ag_main.c:965 (discriminator 3)

0x400f167a: bta_sys_event at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:499

0x400edc44: osi_thread_run at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/bt/common/osi/thread.c:68

0x4008f949: vPortTaskWrapper at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/freertos/xtensa/port.c:143
```

Alvin1Zhang, 2020-12-08:
Thanks for the very detailed report and sorry for the inconvenience, we will look into.

Wth-Esp, 2020-12-08:
Hi @ypiotrp ,

Thanks for your analysis. And we also find that some device has the buffer size as 60 and some other define it as 120.
My colleague has made some changes to the HFP AG module. I will inform him to give you a patch ASAP.

Hope it wroks.
Thanks

Wth-Esp, 2020-12-08:
Hi @ypiotrp 

Please try this [fix_ag_buf.zip](https://github.com/espressif/esp-idf/files/5656796/fix_ag_buf.zip)
on v4.2.
Thanks

ypiotrp, 2020-12-08:
Hi @Alvin1Zhang and @Wth-Esp

Thank you for your fast response. Today I tested the example with the newest IDF v.4.2 with your patch. First of all the HFP AG example does not build because of the warnings -Warray-bounds in `btc_hf_ag.c`. in several functions the local variable `idx` can be set to -1 if the `btc_hf_idx_by_bdaddr()` failes. Next it is used as an index in eg. `hf_local_param` array which causes Core dump and reboot. To get it working I first turned off the -Werror=all compilation flag but as I said it ended up with Core Dump which is not surprising. Then I made an security check after setting the idx variable. In case it is set to -1 the function is exited with error return value. It helped for the Core Dump but the issue still exists. The effect now is that I can't connect audio with the connected HFP device. 

To reproduce it you just have to turn audio on and off several times. Will there be a time when the HFP AG example will simply work?

Wth-Esp, 2020-12-09:
Hi @ypiotrp 

I didn't face the building warning as yours.
As for the original issue, we will test 60 bytes CVSD. And update fix ASAP.

Thanks

munho-espressif, 2020-12-11:
Hi @ypiotrp

Please try this hfp_patch.zip
on v4.2-rc.
Thanks
[hfp_patch.zip](https://github.com/espressif/esp-idf/files/5676822/hfp_patch.zip)

ypiotrp, 2021-01-04:
Hi @munho-espressif,

I applied your patch. After the configuration I mentioned in "Steps to reproduce" and building the example and trying to connect audio to the device ESP32 dumped core. I present the log below. Have you made any progress on the issue so far? 
Log:
```rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7324
load:0x40078000,len:13212
ho 0 tail 12 room 4
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (30) boot: ESP-IDF v4.2-rc-1-g4ea2679ad8-dirty 2nd stage bootloader
I (30) boot: compile time 13:03:56
I (31) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1b23c (111164) map
I (157) esp_image: segment 1: paddr=0x0002b264 vaddr=0x3ffbdb60 size=0x03518 ( 13592) load
I (163) esp_image: segment 2: paddr=0x0002e784 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (164) esp_image: segment 3: paddr=0x0002eb90 vaddr=0x40080404 size=0x01488 (  5256) load
I (175) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x6fc60 (457824) map
0x400d0020: _stext at ??:?

I (356) esp_image: segment 5: paddr=0x0009fc88 vaddr=0x4008188c size=0x146bc ( 83644) load
0x4008188c: start_cpu0_default at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/esp32/cpu_start.c:486

I (404) boot: Loaded app from partition at offset 0x10000
I (405) boot: Disabling RNG early entropy source...
I (405) cpu_start: Pro cpu up.
I (409) cpu_start: Application information:
I (413) cpu_start: Project name:     hfp_ag
I (418) cpu_start: App version:      1
I (423) cpu_start: Compile time:     Jan  4 2021 13:17:06
I (429) cpu_start: ELF file SHA256:  cbbe88d3e6984bf9...
I (435) cpu_start: ESP-IDF:          v4.2-rc-1-g4ea2679ad8-dirty
I (441) cpu_start: Starting app cpu, entry point is 0x40081904
0x40081904: call_start_cpu1 at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (452) heap_init: Initializing. RAM available for dynamic allocation:
I (459) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (465) heap_init: At 3FFB6BF8 len 00001408 (5 KiB): DRAM
I (471) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (477) heap_init: At 3FFCA208 len 00015DF8 (87 KiB): DRAM
I (483) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (490) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (496) heap_init: At 40095F48 len 0000A0B8 (40 KiB): IRAM
I (502) cpu_start: Pro cpu start user code
I (521) spi_flash: detected chip: generic
I (521) spi_flash: flash io: dio
W (521) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (532) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
[... 120 lines cut ...]
0x400d9f56: hci_host_thread_handler at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/bt/host/bluedroid/hci/hci_layer.c:223

0x400edc44: osi_thread_run at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/bt/common/osi/thread.c:68

0x4008fae5: vPortTaskWrapper at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/freertos/xtensa/port.c:143

Core  1 register dump:
PC      : 0x4013e75a  PS      : 0x00060634  A0      : 0x800d3f86  A1      : 0x3ffbccb0
0x4013e75a: esp_pm_impl_waiti at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d3eda  A9      : 0x3ffbcc80
A10     : 0x00000000  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x00060623
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x4013e757:0x3ffbccb0 0x400d3f83:0x3ffbccd0 0x40090471:0x3ffbccf0 0x4008fae5:0x3ffbcd10
0x4013e757: esp_pm_impl_waiti at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/esp32/pm_esp32.c:484

0x400d3f83: esp_vApplicationIdleHook at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40090471: prvIdleTask at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/freertos/tasks.c:3386 (discriminator 1)

0x4008fae5: vPortTaskWrapper at C:/Users/partyka/Documents/projects/device/esp32/esp-idf/components/freertos/xtensa/port.c:143
```

munho-espressif, 2021-01-22:
Hi @ypiotrp 
Please try this 
[hfp_ag.zip](https://github.com/espressif/esp-idf/files/5854021/hfp_ag.zip)

on v4.2-rc.
Thanks

ypiotrp, 2021-02-03:
Thanks @munho-espressif

I think we can close this issue because the main problem here (size of outgoing buffer) has been solved. I still have problem with HFP-AG example not running stable. I will create a new issue.
By the way I saw in [this comment](https://github.com/espressif/esp-idf/issues/4695#issuecomment-767479785) you are planning to improve the audio applications. Can you tell me more about it? Is there any deadline for that. It it is important for me because I want to create my product using ESP32 and HFP AG.

Regards, Piotr

Alvin1Zhang, 2021-02-04:
@ypiotrp Thanks for reporting and updates, good to know the main problem has been resolved, please create another ticket for the new issue, will close this ticket now.

closed 2021-02-04 by Alvin1Zhang