espressif/esp-idf #6219: Example connection stuck

ThomasGeor, 2020-12-07:
https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/examples/common_components/protocol_examples_common/connect.c#L178

I am using MQTT example which uses the example connection for Wifi. The router i am connecting to provides 192.168.0.xx addresses and it's a TP link. Whenever i run my code it got stuck here after finding my router's information and never got out. By putting a break after the semaphore was taken, my code has finally run correctly. I was wondering if this is a bug. I have also tried another network which provides 192.168.1.xx and it works perfectly.

david-cermak, 2020-12-09:
Hi @ThomasGeor 

Are you also getting an IPv6 address? The example connection code waits for both IPv4 and IPv6 addresses by default. It's possible to disable waiting for the IPv6 if its not supported in your network.
```
(Top) > Example Connection Configuration
...
[ ] Obtain IPv6 address
```

Alvin1Zhang, 2021-02-23:
@ThomasGeor Thanks for reporting, will close due to short of feedback, feel free to reopen if any updates. Thanks.

closed 2021-02-23 by Alvin1Zhang