espressif/esp-idf #6220: WiFi Light Sleep+ WIFI_PS_MAX_MODEM change in power draw

patrick277, 2020-12-07:
Hello,

I have some ESP32's ESP-IDF running in automatic light sleep (WIFI_PS_MAX_MODEM) while remaining connected to the internet. This application is battery based and I notice that occasionally the power draw increases (my battery voltage goes down). The code wakes up from light sleep, publishes mqttt and goes back to light sleep every 60s. I enabled the counters from vTaskGetRunTimeStats and I notice they average:
'IDLE1', 'IDLE0', 'sleep_manager', 'mqtt_task', 'tiT', 'TmrSvc', 'ipc1', 'wifi', 'esp_timer', 'sys_evt', 'ipc0', date
63544921 62496503 24613 158418 115500 0 0 466115 233505 0 0 03/12/2020 12:28:27

and change to:
'IDLE1', 'IDLE0', 'sleep_manager', 'mqtt_task', 'tiT', 'TmrSvc', 'ipc1', 'wifi', 'esp_timer', 'sys_evt', 'ipc0', date
63591049 62878982 24345 149548 94302 0 0 262723 115852 0 0 03/12/2020 12:37:57
when the power draw goes up:
[image]

If anyone has seen this I would love some Ideas! If there is any other info let me know!

YouDONG-ESP, 2020-12-09:
Hi @patrick277 tell me if I miss understanding the issue, your application run the same MQTT task every 60s, in shows that occasionally MQTT and wifi, esptimer runtime will change right? Could you show me how often dose it happens and how long dose it last？
BTW, I think you should mesure the input current to get the exact power draw.

YouDONG-ESP, 2020-12-09:
You said each sample is 63s for the second graph, so this issue last for around 400 ~700 min?

patrick277, 2020-12-09:
> 
> 
> You said each sample is 63s for the second graph, so this issue last for around 400 ~700 min?

That is correct. The way it was resolved initially was by sending a few mqtt publish messages from the server to the device. This seems weird to me. 

It happens randomly, I will try to get a current draw, but from the battery voltage graph, it is ~50mA. 

The operation of the code is as follows:
1. connect to wifi, connect to mqtt wss server, read analogs
2. acquire PM lock
3. publish mqtt message 
4. release PMlock
4b. sleep ~60s
5. wake and acquire PM lock
6. read analogs
7. publish mqtt message
8. release PMlock
9. repeat 4b. thru 8

YouDONG-ESP, 2020-12-09:
Could you tell me your chip model and the idf verion?

patrick277, 2020-12-09:
IDF: ESP-IDF v4.1-dirty
Chip is ESP32D0WDQ6 (revision 1)

patrick277, 2020-12-09:
I ran another test last night 
This time I moved the cpu cores as follows, and as you can see. this time mqtt duty cycle did not change when wifi task and esp_timer task run-time dropped. (the battery also dropped exactly when it happened (second graph pic) it is a 18650 2600mAhr battery .

st_rx_data               core0 ->   core1
IDLE0                         core0
IDLE1               core1
sleep_manager  core0 ->  core 1
mqtt_task          core0 ->   core1
tiT                     core0 ->   core1
Tmr Svc              core0
sys_evt              core0
wifi                    core0
esp_timer           core0
ipc0                   core0
ipc1                  core1

data:
[2020-12-09 overnight test.xlsx](https://github.com/espressif/esp-idf/files/5667369/2020-12-09.overnight.test.xlsx)

battery:
[image]

YouDONG-ESP, 2020-12-11:
@patrick277 whats the different between the green curve and the yellow cuve?
Could you change the abscissa of the data to time? If it's still 63 second/sample, the drop time seems not the same.

YouDONG-ESP, 2020-12-11:
@patrick277 The run time of wifi task in light sleep mode should depand on the environnment. If there are lots of others users, 
sta will wake up more time to compete for the channel.
That means wifi run time and battery consomation may drops during the night because there are less users and the environnment is clean.

patrick277, 2020-12-11:
> @patrick277 The run time of wifi task in light sleep mode should depand on the environnment. If there are lots of others users, sta will wake up more time to compete for the channel.
> That means wifi [...]

Is there any way I can monitor how crowded it is or limit how much it tries to compete? so that I protect the battery supply

YouDONG-ESP, 2020-12-14:
@patrick277 I suggest you do a packet capture to see how crowded the channel is and how many times your device tries to compete.

YouDONG-ESP, 2020-12-16:
@patrick277 Wifi tasks wakes less than 5% of the time is a normal behavior for power safe mode.

closed 2021-01-21 by Alvin1Zhang