espressif/esp-idf #6221: implementing  two different services in gattc client

sowmimalu, 2020-12-07:
Hello,
  In my application i using gattc client for connecting with server(peripheral) , here esp32 is my client and third party is my peripheral ,i am able to connect and also to write data on service characteristics . but i need to use service uuid on central side to connect and write data with different peripherals. how to add one more characteristics in client part ?

xiewenxiang, 2020-12-08:
Hi @sowmimalu 

You can reference the [gattc_gatts_coex](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/coex/gattc_gatts_coex) example.

Alvin1Zhang, 2020-12-08:
Thanks for reporting, for How questions, you could create a post on our forum at https://esp32.com, thanks.

closed 2020-12-08 by Alvin1Zhang