espressif/esp-idf #6222: CPU halted usin I2C

aledoors, 2020-12-08:
This is my code:

#include "bmp280.h"
#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <esp_system.h>
#include <stdbool.h>
#include <string.h>
#include "ssd1306.h"
#include "ssd1306_draw.h"
#include "ssd1306_font.h"
#include "ssd1306_default_if.h"

void Prueba (void *pvParameter)
{
	static const int I2CDisplayAddress = 0x3c;
	static const int I2CDisplayWidth = 128;
	static const int I2CDisplayHeight = 64;
	static const int I2CResetPin = -1;

	struct SSD1306_Device I2CDisplay;

	bool DefaultBusInit( void ) {
		assert( SSD1306_I2CMasterInitDefault( ) == true );
		assert( SSD1306_I2CMasterAttachDisplayDefault( &I2CDisplay, I2CDisplayWidth, I2CDisplayHeight, I2CDisplayAddress, I2CResetPin ) == true );
		return true;
		}

	void SetupDemo( struct SSD1306_Device* DisplayHandle, const struct SSD1306_FontDef* Font ) {
		SSD1306_Clear( DisplayHandle, SSD_COLOR_BLACK );
		SSD1306_SetFont( DisplayHandle, Font );
		}

	void SayHello( struct SSD1306_Device* DisplayHandle, const char* HelloText ) {
		SSD1306_FontDrawAnchoredString( DisplayHandle, TextAnchor_Center, HelloText, SSD_COLOR_WHITE );
		SSD1306_Update( DisplayHandle );
		}

		printf( "Ready...\n" );
		if ( DefaultBusInit( ) == true ) {
			printf( "BUS Init lookin good...\n" );
			printf( "Drawing.\n" );
		SetupDemo( &I2CDisplay, &Font_droid_sans_fallback_24x28 );
        SayHello( &I2CDisplay, "SICCU!" );
        printf( "Done!\n" );
		}
	vTaskDelete(NULL);
}

void bmp280_test(void *pvParamters)
{

    bmp280_params_t params;
    bmp280_init_default_params(&params);
    bmp280_t dev;
    memset(&dev, 0, sizeof(bmp280_t));

	#define SDA_GPIO 4
	#define SCL_GPIO 5

    bool bme280p = dev.id == BME280_CHIP_ID;
    printf("BMP280: found %s\n", bme280p ? "BME280" : "BMP280");

    float pressure, temperature, humidity;

    while (1)
    {
        vTaskDelay(2000 / portTICK_PERIOD_MS);
        if (bmp280_read_float(&dev, &temperature, &pressure, &humidity) != ESP_OK)
        {
            printf("Temperature/pressure reading failed\n");
            continue;
        }

        /* float is used in printf(). you need non-default configuration in
         * sdkconfig for ESP8266, which is enabled by default for this
         * example. see sdkconfig.defaults.esp8266
         */
        printf("Pressure: %.2f Pa, Temperature: %.2f C", pressure, temperature);
        if (bme280p)
            printf(", Humidity: %.2f\n", humidity);
        else
            printf("\n");
    }
    vTaskDelete(NULL);
}

void app_main()
{
	xTaskCreate(&Prueba,"Prueba",1024,NULL,1,NULL);
	xTaskCreate(bmp280_test, "bmp280_test", configMINIMAL_STACK_SIZE * 8, NULL, 5, NULL);
}

This is log compile:

Python requirements from C:/esp-idf\requirements.txt are satisfied.
[1/4] cmd.exe /C "cd /D D:\Eclipse\Trabajo\OLED_BMP280_Ver_1.1\build\esp-idf\partition_table && "C:\Users\C. Alejandro\.espressif\tools\cmake\3.13.4\bin\cmake.exe" -E echo "Partition table binary generated. Contents:" && "C:\Users\C. Alejandro\.espressif\tools\cmake\3.13.4\bin\cmake.exe" -E echo ******************************************************************************* && python C:/esp-idf/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 4MB D:/Eclipse/Trabajo/OLED_BMP280_Ver_1.1/build/partition_table/partition-table.bin && "C:\Users\C. Alejandro\.espressif\tools\cmake\3.13.4\bin\cmake.exe" -E echo *******************************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/4] Performing build step for 'bootloader'
ninja: no work to do.
esptool.py -p COM3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 HelloWorld.bin
esptool.py v2.8
Serial port COM3
Connecting........__
Detecting chip type... ESP32
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Single Core, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:85:fc:58
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 3072 bytes to 103...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 2240.3 kbit/s)...
Hash of data verified.
Compressed 25296 bytes to 14936...

Writing at 0x00001000... (100 %)
Wrote 25296 bytes (14936 compressed) at 0x00001000 in 0.3 seconds (effective 589.9 kbit/s)...
Hash of data verified.
Compressed 182848 bytes to 90556...

Writing at 0x00010000... (16 %)
Writing at 0x00014000... (33 %)
Writing at 0x00018000... (50 %)
Writing at 0x0001c000... (66 %)
Writing at 0x00020000... (83 %)
Writing at 0x00024000... (100 %)
Wrote 182848 bytes (90556 compressed) at 0x00010000 in 2.2 seconds (effective 650.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Checking Python dependencies...
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running ninja in directory d:\eclipse\trabajo\oled_bmp280_ver_1.1\build
Executing "ninja all"...
Executing action: flash
Running esptool.py in directory d:\eclipse\trabajo\oled_bmp280_ver_1.1\build
Executing "'C:\Users\C. Alejandro\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe' C:/esp-idf\components/esptool_py/esptool/esptool.py -p COM3 -b 460800 --before default_reset --after hard_reset write_flash @flash_project_args"...
Done

This is in terminal:

I (71) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot: ESP-IDF v4.0.1-dirty 2nd stage bootloader
I (39) boot: compile time 16:19:09
I (39) boot: Enabling RNG early entropy source...
I (44) boot: SPI Speed      : 40MHz
I (48) boot: SPI Mode       : DIO
I (52) boot: SPI Flash Size : 4MB
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (90) boot: End of partition table
I (94) boot_comm: chip revision: 1, min. application chip revision: 0
I (101) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0afcc ( 45004) map
I (126) esp_image: segment 1: paddr=0x0001aff4 vaddr=0x3ffb0000 size=0x018d4 (  6356) load
I (129) esp_image: segment 2: paddr=0x0001c8d0 vaddr=0x40080000 size=0x00400 (  1024) load
I (133) esp_image: segment 3: paddr=0x0001ccd8 vaddr=0x40080400 size=0x03338 ( 13112) load
I (147) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x15b74 ( 88948) map
I (183) esp_image: segment 5: paddr=0x00035b94 vaddr=0x40083738 size=0x06e88 ( 28296) load
I (201) boot: Loaded app from partition at offset 0x10000
I (201) boot: Disabling RNG early entropy source...
I (202) cpu_start: Pro cpu up.
I (205) cpu_start: Application information:
I (210) cpu_start: Project name:     HelloWorld
I (215) cpu_start: App version:      1
I (220) cpu_start: Compile time:     Dec  7 2020 23:49:17
I (226) cpu_start: ELF file SHA256:  f670c16c0a7aaac5...
I (232) cpu_start: ESP-IDF:          v4.0.1-dirty
I (237) cpu_start: Single core mode
I (242) heap_init: Initializing. RAM available for dynamic allocation:
I (249) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (255) heap_init: At 3FFB2840 len 0002D7C0 (181 KiB): DRAM
I (261) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
I (267) heap_init: At 40078000 len 00008000 (32 KiB): IRAM
I (273) heap_init: At 4008A5C0 len 00015A40 (86 KiB): IRAM
I (280) cpu_start: Pro cpu start user code
I (297) spi_flash: detected chip: gd
I (298) spi_flash: flash io: dio
I (298) cpu_start: Starting scheduler on PRO CPU.
Ready...
BMP280: found BMP280
BUS Init lookin good...
Drawing.
Done!
C:/esp-idf/components/freertos/queue.c:1448 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40088925 on core 0

ELF file SHA256: f670c16c0a7aaac5

Backtrace: 0x40085785:0x3ffb6330 0x40085b05:0x3ffb6350 0x40088925:0x3ffb6370 0x400d4d2f:0x3ffb63b0 0x400d4f11:0x3ffb63e0 0x400d521d:0x3ffb6420 0x400d3b40:0x3ffb6450 0x40087f6d:0x3ffb64e0

CPU halted.

I don't know where is the error. Someone can help me?
Thanks in advance

Francois-Belanger, 2020-12-08:
Hi @aledoors, please look at line:
C:/esp-idf/components/freertos/queue.c:1448 (xQueueGenericReceive)- assert failed!

Probably missing an init.  I don't see any "i2c_param_config" or "i2c_driver_install"

ginkgm, 2020-12-08:
Hi @aledoors , I can't tell which function caused the abort() without a backtrace.

Could you provide the elf, or use idf_monitor (idf.py monitor) to decode the backtrace?

and BTW, the code you provided does not directly call any IDF functions, I wonder if you are using an 3rd party lib?

aledoors, 2020-12-08:
Hi @ginkgm, i'm using Eclipse IDE. I'm new to programming on FreeRTOS based systems and using ESP-IDF and Eclipse IDE. My knowledge is basic but I am learning. I ask you if you can guide me on what steps I should take to find the problem.

On the other hand, I did the compilation through ESP-IDF and I got the following result:

D:\Eclipse\Trabajo\OLED_BMP280_Ver_1.1>idf.py build flash monitor
Checking Python dependencies...
Python requirements from C:\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory d:\eclipse\trabajo\oled_bmp280_ver_1.1\build
Executing "ninja all"...
[1/4] cmd.exe /C "cd /D D:\Eclipse\Trabajo\OLED_BMP280_Ver...*********************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/4] Performing build step for 'bootloader'
ninja: no work to do.
Executing action: flash
Choosing default port b'COM3' (use '-p PORT' option to set a specific serial port)
Running esptool.py in directory d:\eclipse\trabajo\oled_bmp280_ver_1.1\build
Executing "'C:\Users\C. Alejandro\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe' C:\esp-idf\components/esptool_py/esptool/esptool.py -p COM3 -b 460800 --before default_reset --after hard_reset write_flash @flash_project_args"...
esptool.py -p COM3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 HelloWorld.bin
esptool.py v2.8
Serial port COM3
Traceback (most recent call last):
  File "C:\esp-idf\components/esptool_py/esptool/esptool.py", line 3201, in <module>
    _main()
  File "C:\esp-idf\components/esptool_py/esptool/esptool.py", line 3194, in _main
    main()
  File "C:\esp-idf\components/esptool_py/esptool/esptool.py", line 2883, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "C:\esp-idf\components/esptool_py/esptool/esptool.py", line 273, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "C:\esp-idf\components/esptool_py/esptool/esptool.py", line 237, in __init__
    self._port = serial.serial_for_url(port)
  File "C:\Users\C. Alejandro\.espressif\python_env\idf4.0_py3.8_env\lib\site-packages\serial\__init__.py", line 88, in serial_for_url
    instance.open()
  File "C:\Users\C. Alejandro\.espressif\python_env\idf4.0_py3.8_env\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM3': PermissionError(13, 'Acceso denegado.', None, 5)
esptool.py failed with exit code 1

D:\Eclipse\Trabajo\OLED_BMP280_Ver_1.1>idf.py build flash monitor
Checking Python dependencies...
Python requirements from C:\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory d:\eclipse\trabajo\oled_bmp280_ver_1.1\build
Executing "ninja all"...
[1/4] cmd.exe /C "cd /D D:\Eclipse\Trabajo\OLED_BMP280_Ver...*********************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/4] Performing build step for 'bootloader'
ninja: no work to do.
Executing action: flash
Choosing default port b'COM3' (use '-p PORT' option to set a specific serial port)
Running esptool.py in directory d:\eclipse\trabajo\oled_bmp280_ver_1.1\build
Executing "'C:\Users\C. Alejandro\.espressif\python_env\idf4.0_py3.8_env\Scripts\python.exe' C:\esp-idf\components/esptool_py/esptool/esptool.py -p COM3 -b 460800 --before default_reset --after hard_reset write_flash @flash_project_args"...
esptool.py -p COM3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 HelloWorld.bin
esptool.py v2.8
Serial port COM3
[... 91 lines cut ...]
Done!
C:/esp-idf/components/freertos/queue.c:1448 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40088925 on core 0
0x40088925: xQueueGenericReceive at C:/esp-idf/components/freertos/queue.c:1448 (discriminator 1)

ELF file SHA256: f670c16c0a7aaac5

Backtrace: 0x40085785:0x3ffb6330 0x40085b05:0x3ffb6350 0x40088925:0x3ffb6370 0x400d4d2f:0x3ffb63b0 0x400d4f11:0x3ffb63e0 0x400d521d:0x3ffb6420 0x400d3b40:0x3ffb6450 0x40087f6d:0x3ffb64e0
0x40085785: invoke_abort at C:/esp-idf/components/esp32/panic.c:155

0x40085b05: abort at C:/esp-idf/components/esp32/panic.c:172

0x40088925: xQueueGenericReceive at C:/esp-idf/components/freertos/queue.c:1448 (discriminator 1)

0x400d4d2f: i2c_dev_take_mutex at D:/esp-idf-lib-master/components/i2cdev/i2cdev.c:111

0x400d4f11: bmp280_read_fixed at D:/esp-idf-lib-master/components/bmp280/bmp280.c:347

0x400d521d: bmp280_read_float at D:/esp-idf-lib-master/components/bmp280/bmp280.c:379 (discriminator 4)

0x400d3b40: bmp280_test at D:\Eclipse\Trabajo\OLED_BMP280_Ver_1.1\build/../main/main.c:77

0x40087f6d: vPortTaskWrapper at C:/esp-idf/components/freertos/port.c:143

CPU halted.

aledoors, 2020-12-08:
Adding information, this is my CMakeLists.txt:

**# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

set(EXTRA_COMPONENT_DIRS D:/esp-idf-lib-master/components)

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(OLED_BMP280_Ver_1.1)**

I needed to add an external library to my code and by editing this file I found out how to do it.

MrCurio, 2020-12-12:
Hello @aledoors 
I quite often have this issue and it's usually the same mistake.

**You're accessing a NULL pointer**. That's what the assert is doing. 
I dont know if you're using any queues in your program, but if you do, remember to allocate them before you write/read on them. 
If you're not using queues, there might be some initialization/allocation missing. I see you use your own lib for I2C, doesn't it have an initialization function before using? You may be accessing a NULL mutex causing an abort.

The backtrace shows the error in the i2c_dev file. Review if you're missing an initialization.

Thanks!

aledoors, 2020-12-14:
Hello @MrCurio

Thanks for your help. I'm not using queues in my program, what I did was use two programs that work perfectly and put together a single program, separated into two tasks. Even if I cancel one task or the other the program works fine with the task enabled.
I have the problem when both tasks work simultaneously, the SSD1306 display task uses two different SCL and SDA pins than the SCL and SDA pins of the BMP280 sensor task. What I do is use two I2C channels because I don't know how to do it from the same channel. I am newly programmed FreeRTOS and I am learning and making mistakes.

MrCurio, 2020-12-14:
Hello @aledoors 

As i said you are **trying to access a NULL pointer.** Let me explain more in detail. 

As i assume, you are using a third party library called ```esp-idf-lib``` by UncleRus. This library creates a thread safe API for reading I2C devices by using mutexes (semaphores). Let's say that your header file ```bmp280.h ``` belongs to the esp-idf-lib repo. If this is it,  you need to initialize those mutexes.

For doing so, the ```bmp280.h ``` provides some initialization function you seem to not use. Review what i am talking about cause i am quite sure this is the problem you are having. Below i leave some of the functions i am talking about:

```
esp_err_t bmp280_init_desc(bmp280_t *dev, uint8_t addr, i2c_port_t port, gpio_num_t sda_gpio, gpio_num_t scl_gpio);

esp_err_t bmp280_init(bmp280_t *dev, bmp280_params_t *params);

esp_err_t bmp280_init_default_params(bmp280_params_t *params);
```

It should be noted that if you use the ``` esp-idf-lib ``` repo, it does not matter if you use 2 different ports or not. **Mutexes must be initialize!**

Thanks!

aledoors, 2020-12-16:
Hello @MrCurio 

Thank you for all of your help, we were able to solve the problem and now the program works correctly.

I add these two lines:

    ESP_ERROR_CHECK(bmp280_init_desc(&dev, BMP280_I2C_ADDRESS_0, 0, SDA_GPIO, SCL_GPIO));
    ESP_ERROR_CHECK(bmp280_init(&dev, &params));

but the error still continued

Then modify this line and change the 0 to 1 and the program started working:
   ESP_ERROR_CHECK(bmp280_init_desc(&dev, BMP280_I2C_ADDRESS_0, **1**, SDA_GPIO, SCL_GPIO));

I'm not entirely sure what changed, but now the program works correctly.

MrCurio, 2020-12-16:
Hello @aledoors 
 You're welcome !
I'll close this issue

closed 2020-12-16 by aledoors