espressif/esp-idf #6225: Heap Corruption everytime esp_http_read is called multiple times

rjam-xav, 2020-12-08:
Hello, 
I am trying to read data from a server that sends non chunked response of the size of ~800kb. I am trying to read this as below. However, there is always heap corruption on the second read/while loop iteration.
Code:
  

> while (total_read_bytes_tts <= content_length_tts) {
>             ESP_LOGI(PTAG, "setting bytes to 0");
>             memset(buffer_tts, 0, MAX_HTTP_RECV_BUFFER_TTS + 1);
>             ESP_LOGI(PTAG, "reading more bytes");
>             read_bytes_tts = esp_http_client_read(client_tts, buffer_tts, MAX_HTTP_RECV_BUFFER_TTS);
>             if (read_bytes_tts <= 0) {
>                 ESP_LOGI(PTAG, "ERROR READING TTS DATA");
>                 esp_periph_set_destroy(set);
>                 free(buffer_tts);
>                 return 0;
>             }
>             total_read_bytes_tts += read_bytes_tts;
>             ESP_LOGI(PTAG, "total bytes read: %d", total_read_bytes_tts);
>             //ESP_LOGI(PTAG, "bytes read: %s", buf);
>           //  free(buffer_tts);
>         }

and the error:

>   while (total_read_bytes_tts <= content_length_tts) {
>             ESP_LOGI(PTAG, "setting bytes to 0");
>             memset(buffer_tts, 0, MAX_HTTP_RECV_BUFFER_TTS + 1);
>             ESP_LOGI(PTAG, "reading more bytes");
>             read_bytes_tts = esp_http_client_read(client_tts, buffer_tts, MAX_HTTP_RECV_BUFFER_TTS);
>             if (read_bytes_tts <= 0) {
>                 ESP_LOGI(PTAG, "ERROR READING TTS DATA");
>                 esp_periph_set_destroy(set);
>                 free(buffer_tts);
>                 return 0;
>             }
>           //  int bytes_wav = fwrite(buffer_tts_two, sizeof(char), read_bytes_tts, file_wav);
>           //  if (bytes_wav < read_bytes_tts || bytes_wav == 0) {
>            //     ESP_LOGE(PTAG, "Error dumping response to wav file");
>            //     fclose(file_wav);
>            //     esp_http_client_close(client_tts);
>            //     esp_http_client_cleanup(client_tts);
>             //    esp_periph_set_destroy(set);
>            // }
>             total_read_bytes_tts += read_bytes_tts;
>             ESP_LOGI(PTAG, "total bytes read: %d", total_read_bytes_tts);
>         }
>

istokm, 2020-12-08:
There are large chunks of information missing from your issue, please first look at the [bug report template](https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/.github/ISSUE_TEMPLATE/bug_report.md) and provide all essential information (idf & tool versions, etc). You also provided a very limited snippet of your code, without how you actually allocated the buffer, how large it is, how you established connection, any log output outside of the error, and other maybe seemingly-unimportant details. I can confirm that for me, esp_http_read *currently* works correctly on the master version of ESP-IDF, as I do a very similar thing in my OTA code.

As a side note, please use the code formatting option when posting snppets, and not the quote formatting. There is a large difference in readability.

*Insert code* option (using: _\``` ***newline*** code_ ***newline*** \```)
```c
while (total_read_bytes_tts <= content_length_tts) {
```
*Insert quote* option (using: _\> quote_)
> while (total_read_bytes_tts <= content_length_tts) {

AdityaHPatwardhan, 2021-01-18:
@rjam-xav can you please provide some small test code which I can try at my end and reproduce the issue ?
The missing information suggested by @istokm will indeed help to reproduce and debug the issue. 
If the problem is already resolved on your end can you please close the issue ?

mahavirj, 2021-01-27:
Closing due to lack of update. Please feel free to reopen with more information as requested.

closed 2021-01-27 by mahavirj