espressif/esp-idf #6227: ESP_Ethernet W5500 error with SSL

huybk213, 2020-12-08:
Dear all,

After running example SPI-Ethernet with w5500, i found problems due to memory of W5500.

1. Chip ESP32S2
2. Branch master, commit 178b122c145c19e94ac896197a3a4a9d379cd618
3. Code : Ethernet example + aws iot sdk. I also test with "https" url, problems is the same.
4. Log 
**w5500-mac: emac_w5500_transmit(511): free size (0) < send length (42)**

I think the problems is W5500 does not have enough memory for TX buffer, because SSL use a lot of memory for TX buffer (typical 16kb)

Could you please review it?

Thanks

david-cermak, 2021-05-05:
Hi @huybk213 

We've recently fixed some issues in w5500 driver.
* https://github.com/espressif/esp-idf/commit/dfe9580967a4b35871913ef3fdc47d8c6798f476
* https://github.com/espressif/esp-idf/commit/7af358ed93bf2f165bfc84341d1fb08f49f28573

Plus, we have one fix in progress:
[esp_eth-w5500-Improve-GPIO-interrupt-processing.patch.txt](https://github.com/espressif/esp-idf/files/6427173/esp_eth-w5500-Improve-GPIO-interrupt-processing.patch.txt)

I have checked some simple `https` and `mqtts` scenarios and they seem to work as expected. Could you please also test if this works for you?  if you agree with this assessment and the above patch works in your setup, I would mark the WIP fix to close this issue.
Thanks.

huybk213, 2021-05-08:
Ok, i will stress test and report ASAP

closed 2021-09-14 by Alvin1Zhang