espressif/esp-idf #6228: ESP32-S2 is reset 1 second after entering lightsleep, reason 0x10 RTCWDT_RTC_RST

yannpom, 2020-12-08:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-S2-WROOM
- IDF version: v4.3-dev-1901-g178b122c1
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     macOS
- Using an IDE?:       No
- Power Supply:         external 3.3V

## Problem Description

When I run the example provided in `example/system/light_sleep`, the ESP32-S2 reboots 1 second after entering lightsleep with reboot reason 0x10 RTCWDT_RTC_RST.

### Expected Behavior

The ESP32-S2 should lightsleep and resume after 2 seconds (or a GPIO0 falling edge).

### Actual Behavior

The ESP32-S2 is reset after 1 second with reason 0x10 RTCWDT_RTC_RST.

### Steps to reproduce

idf.py set-target esp32s2
idf.py build
idf.py -p /dev/cu.usbserial-FT4521QV flash monitor

## Debug Logs

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x180c
load:0x4004c000,len:0x9d8
load:0x40050000,len:0x2e34
entry 0x4004c1ec
I (46) boot: ESP-IDF v4.3-dev-1901-g178b122c1 2nd stage bootloader
I (46) boot: compile time 14:47:31
I (46) boot: chip revision: 0
I (49) boot.esp32s2: SPI Speed      : 80MHz
I (54) boot.esp32s2: SPI Mode       : DIO
I (59) boot.esp32s2: SPI Flash Size : 2MB
I (63) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  2 factory          factory app      00 00 00010000 00100000
I (102) boot: End of partition table
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x06428 ( 25640) map
I (121) esp_image: segment 1: paddr=0x00016450 vaddr=0x3ffbd480 size=0x025e8 (  9704) load
I (126) esp_image: segment 2: paddr=0x00018a40 vaddr=0x40022000 size=0x00404 (  1028) load
I (133) esp_image: segment 3: paddr=0x00018e4c vaddr=0x40022404 size=0x071cc ( 29132) load
I (150) esp_image: segment 4: paddr=0x00020020 vaddr=0x40080020 size=0x12094 ( 73876) map
I (166) esp_image: segment 5: paddr=0x000320bc vaddr=0x400295d0 size=0x03ea4 ( 16036) load
I (177) boot: Loaded app from partition at offset 0x10000
I (177) boot: Disabling RNG early entropy source...
I (189) cache: Instruction cache    : size 8KB, 4Ways, cache line size 32Byte
I (189) cpu_start: Pro cpu up.
I (243) cpu_start: Pro cpu start user code
I (243) cpu_start: cpu freq: 160000000
I (243) cpu_start: Application information:
I (245) cpu_start: Project name:     light_sleep_example
I (251) cpu_start: App version:      v4.3-dev-1901-g178b122c1
I (258) cpu_start: Compile time:     Dec  8 2020 14:47:24
I (264) cpu_start: ELF file SHA256:  6d821b2ae047bc07...
I (270) cpu_start: ESP-IDF:          v4.3-dev-1901-g178b122c1
I (276) heap_init: Initializing. RAM available for dynamic allocation:
I (284) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (290) heap_init: At 3FFC01F8 len 0003BE08 (239 KiB): DRAM
I (296) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (303) spi_flash: detected chip: generic
I (307) spi_flash: flash io: dio
W (311) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (324) cpu_start: Starting scheduler on PRO CPU.
I (330) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
Entering light sleep
##################################################################
ESP32S2 SLEEPS FOR AROUND 1 SECOND HERE, BEFORE BEING RESET.
##################################################################
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x180c
load:0x4004c000,len:0x9d8
load:0x40050000,len:0x2e34
entry 0x4004c1ec
I (46) boot: ESP-IDF v4.3-dev-1901-g178b122c1 2nd stage bootloader
I (46) boot: compile time 14:47:31
I (47) boot: chip revision: 0
I (50) boot.esp32s2: SPI Speed      : 80MHz
I (55) boot.esp32s2: SPI Mode       : DIO
I (59) boot.esp32s2: SPI Flash Size : 2MB
I (64) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 00100000
I (103) boot: End of partition table
I (107) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x06428 ( 25640) map
I (121) esp_image: segment 1: paddr=0x00016450 vaddr=0x3ffbd480 size=0x025e8 (  9704) load
I (127) esp_image: segment 2: paddr=0x00018a40 vaddr=0x40022000 size=0x00404 (  1028) load
I (134) esp_image: segment 3: paddr=0x00018e4c vaddr=0x40022404 size=0x071cc ( 29132) load
I (150) esp_image: segment 4: paddr=0x00020020 vaddr=0x40080020 size=0x12094 ( 73876) map
I (167) esp_image: segment 5: paddr=0x000320bc vaddr=0x400295d0 size=0x03ea4 ( 16036) load
I (178) boot: Loaded app from partition at offset 0x10000
I (178) boot: Disabling RNG early entropy source...
I (190) cache: Instruction cache    : size 8KB, 4Ways, cache line size 32Byte
I (190) cpu_start: Pro cpu up.
I (242) cpu_start: Pro cpu start user code
I (242) cpu_start: cpu freq: 160000000
I (242) cpu_start: Application information:
I (244) cpu_start: Project name:     light_sleep_example
I (250) cpu_start: App version:      v4.3-dev-1901-g178b122c1
I (257) cpu_start: Compile time:     Dec  8 2020 14:47:24
I (263) cpu_start: ELF file SHA256:  6d821b2ae047bc07...
I (269) cpu_start: ESP-IDF:          v4.3-dev-1901-g178b122c1
I (275) heap_init: Initializing. RAM available for dynamic allocation:
I (283) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (289) heap_init: At 3FFC01F8 len 0003BE08 (239 KiB): DRAM
I (295) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (302) spi_flash: detected chip: generic
I (306) spi_flash: flash io: dio
W (310) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (323) cpu_start: Starting scheduler on PRO CPU.
I (329) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
Entering light sleep
##################################################################
ESP32S2 SLEEPS FOR AROUND 1 SECOND HERE, BEFORE BEING RESET.
##################################################################
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
```

## Other items if possible

If I externally toggle GPIO0 within the first second. The ESP32-S2 exits the lightsleep as expected, then re-enters it. But if I wait more than 1 second, the WDT resets the ESP32-S2.

If I comment line `light_sleep_example_main.c:45`: `esp_sleep_enable_timer_wakeup(200000);`. The ESP32-S2 does not reset after 1 second. It stays in lightsleep and wakes up with a falling GPIO0. The problem seems to be related to the activation of the wakeup timer for lightsleep.

Alvin1Zhang, 2020-12-09:
Thanks for the detailed report and letting us know, we will look into.

maalllooooo, 2021-07-02:
I've the same issue with an ESP-32-S2-WROVER (saola board).
I'm using the esp-idf extension for vs-code (version 1.1.0), the idf version is 4.2.
the error appear every time I try to use esp_sleep_enable_timer_wakeup(), this include the light_sleep example.
if anyone know someting about it I'll be happy to hear it!

Pupil2013, 2021-08-20:
I got the same issue using esp32s2,idf4.3.
Don't know the reason why it restarts.
rst:0x10 (RTCWDT_RTC_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Also working in light sleep mode

jack0c, 2022-04-24:
I have no idea about the reason. Can't reproduce the issue. 
However, I find `esp_sleep_enable_timer_wakeup()` some how related with the issue.
So I suspect it is related with the flash power down feature.
@yannpom @maalllooooo @Pupil2013 Can your tell if you enable flash power down option?
If yes, can you test with this option disabled?

lennyJL, 2022-04-24:
Hi @yannpom. This issue is related with the flash power down feature. I think it has been fixed in the latest release/v4.3. If you need to use light sleep feature, you can rebase your code to the latest release/v4.3.

lennyJL, 2022-04-24:
Hi @maalllooooo. You can rebase your code to the latest release/v4.2.

closed 2022-04-25 by Alvin1Zhang