espressif/esp-idf #6230: WiFi stops working after one or two days of operation

itayperl, 2020-12-08:
## Environment

- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.3-dev-1901-g178b122c1
- Power Supply:    external 5V

WiFi is configured with sleep disabled (WIFI_PS_NONE). BLE is also enabled.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5660825/sdkconfig.txt)

## Problem Description

WiFi stops working after the device has been running for a long time (> 24 hours).

Some observations:
1. Cannot ping the device's IP, and the device cannot make outbound connections
2. Nothing is written to the monitor logs when this happens
3. esp_wifi_sta_get_ap_info returns the correct AP record, so ESP still thinks it's connected
4. If I try to reconnect, or connect to a different network, the connection fails:
```
I (112624324) app_prov_handler: WiFi Credentials Received :
        ssid ****
        password *****
I (112624514) app_prov_handler: WiFi Credentials Applied
I (112624634) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (112624636) wifi:state: init -> auth (b0)
I (112624640) wifi:state: auth -> assoc (0)
I (112624646) wifi:state: assoc -> run (10)
I (112624684) app_prov_handler: Connecting state
I (112625358) app_prov_handler: Connecting state
I (112626033) app_prov_handler: Connecting state
I (112626708) app_prov_handler: Connecting state
I (112627383) app_prov_handler: Connecting state
I (112628058) app_prov_handler: Connecting state
I (112628690) wifi:state: run -> init (2a0)
I (112628691) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (112628692) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
E (112628704) app_prov: STA Disconnected
E (112628704) app_prov: Disconnect reason : 2
I (112628715) app_prov: STA Auth Error
I (112628734) app_prov_handler: Disconnected state
```

Alvin1Zhang, 2020-12-09:
Thanks for reporting the issue, and sorry for the inconvenience, we will look into.

shahpiyushv, 2020-12-09:
@itayperl , the query is not clear to me. Can you answer these questions

1) Is this with some standard esp-idf example?
2) Was the device connected to the correct AP and then lost access after 2 days?
3) After the device lost connection, what operation did  you perform on the device side to re-start provisioning?

itayperl, 2020-12-09:
Hi @shahpiyushv, thanks for the quick reply.

1. No, I'm running my own app.
2. The device was connected for about a day with no issues, then at some point it disappeared from the network. I had this happen several times.
3. I have a custom command over BLE that calls `esp_wifi_disconnect` followed by `app_prov_start_ble_provisioning`. From there I just use the protocomm API. I made minor changes to app_prov so I can stop the prov task when done without disabling BLE, so I can restart it later. It works really well in general. It only failed in the scenario in this issue.

YouDONG-ESP, 2020-12-10:
Hi @Alvin1Zhang, the reason code `I (112628690) wifi:state: run -> init (2a0)`shows that sta had received a disassosiation frame from AP with reason AUTH_EXPIRE. This frame would be send by an AP in two case:
1. The ssid entering the PSK is wrong. 
2. Client left BSSID without sending a deauthentication frame. It should not happen because esp_wifi_sta_get_ap_info can return, it shows that from the sta side, sta is still connected. So it should be a problem from the AP.

I have three questions:
1. If you restart esp32, can it reconnect?
2. which authenticate mode did you use?
3. Dose it happen with others type of AP?

itayperl, 2020-12-10:
Hi @YouDONG-ESP,

1. I was able to connect with no issues after restarting the ESP32.
2. WPA2-PSK
3.  I'm not sure what you mean by other types of AP.  Before I restarted the device, I tried to connect to a different AP (I started an AP on my laptop via wpa_supplicant) with a similar result:
```
WiFi Credentials Received : 
	ssid *****
	password *****
I (115207343) app_prov_handler: WiFi Credentials Applied
I (115207506) app_prov_handler: Connecting state
I (115208068) wifi:new:<6,0>, old:<11,0>, ap:<255,255>, sta:<6,0>, prof:1
I (115208481) wifi:state: init -> auth (b0)
I (115208486) app_prov_handler: Connecting state
I (115208494) wifi:state: auth -> assoc (0)
I (115208508) wifi:state: assoc -> run (10)
I (115209171) app_prov_handler: Connecting state
I (115209846) app_prov_handler: Connecting state
I (115210521) app_prov_handler: Connecting state
I (115211196) app_prov_handler: Connecting state
I (115211616) wifi:state: run -> init (2c0)
I (115211617) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (115211618) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
E (115211630) app_prov: STA Disconnected
E (115211631) app_prov: Disconnect reason : 2
I (115211641) app_prov: STA Auth Error
I (115211871) app_prov_handler: Disconnected state
```
This other AP also connected without issue after I restarted the ESP.

Thanks.

YouDONG-ESP, 2020-12-17:
@itayperl It's hard to reproduce it in our side, could you help us to debug this issue? You could do a packet capture for the wifi packet when it happens and add a esp_get_free_heap_size() command to see if it's a memory leak problem.
BTW when it happen, will it cause problem to bluetooth part?

itayperl, 2020-12-19:
Hi @YouDONG-ESP,

1. esp_get_free_heap_size returned 23608 when the issue reproduced
2. Packet captures: [wifi_capture.zip](https://github.com/espressif/esp-idf/files/5719449/wifi_capture.zip). the ZIP contains a capture when connecting to an AP after the issue occurred (wifi_bad.pcap) and then immediately after rebooting the device (wifi_good.pcap). 
3. Bluetooth works flawlessly even after WiFi stops working.

Please let me know if there's any other information I can provide that might help resolve this.
Thanks!

YouDONG-ESP, 2020-12-21:
@itayperl the packet you post is wired packets, It shows that esp32 didn't reply eapol frame for authentication. I need wireless packet to see if esp32's hardware had recived them(by sending ACK/BA frame), please follow this tutorial for wireless packet capture, thanks:

The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets:
https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

itayperl, 2021-01-07:
@YouDONG-ESP, I reproduced the issue again and captured in monitor mode as you requested.
[esp-idf-issue-6230.zip](https://github.com/espressif/esp-idf/files/5780782/esp-idf-issue-6230.zip)

YouDONG-ESP, 2021-01-07:
Thanks for the packets @itayperl. 
When it happens, does it make a difference if you connect it to an open router(no password)?

YouDONG-ESP, 2021-01-21:
@itayperl It should be a memory leak in wifi lib. Couldy you add command `esp_wifi_statis_dump(WIFI_STATIS_ALL)` to print the wifi buffer info?

itayperl, 2021-01-24:
Here's the output after WiFi stopped working:
```
I (116808459) wifi: BUFFER:
I (116808459) wifi: t  max alloc total_alloc total_free fail_fc  fail_oom
I (116808462) wifi: 1  32  0     5375        5375       0        0       
I (116808468) wifi: 2  32  0     51651       51651      0        0       
I (116808475) wifi: 3  32  0     8297        8297       0        0       
I (116808481) wifi: 4  32  0     0           0          0        0       
I (116808488) wifi: 5  0   0     4           4          0        0       
I (116808494) wifi: 6  0   0     0           0          0        0       
I (116808501) wifi: 7  64  64    103014      102950     0        19399   
I (116808507) wifi: 8  64  0     1243305     1243305    0        0       
I (116808514) wifi: 9  0   0     5375        5407       0        0       
I (116808520) wifi:
I (116808522) wifi: LMAC TX:
I (116808525) wifi: tx_all   lrc src age to 
I (116808529) wifi: 61750    0   2   0   0  
I (116808533) wifi: 0        0   0   0   0  
I (116808537) wifi: 3573     0   0   0   0  
I (116808541) wifi: 0        0   0   0   0  
I (116808545) wifi: 0        0   0   0   0  
I (116808549) wifi:
I (116808550) wifi: LMAC RX:
I (116808553) wifi: ampdu    mdpu     mgmt     ctrl     frag
I (116808558) wifi: 4084     118106   1351624  4347     0   
I (116808564) wifi:
I (116808566) wifi: HW RX:
I (116808568) wifi: rx_suc   rx_end   full#1   full#2  
I (116808573) wifi: 168e11   83a9c80  6        0       
I (116808578) wifi:
I (116808580) wifi: Hardware:
I (116808583) wifi: rxbufbk  rxcckerr ofdmerr  rx_agc   rxsf     rx_abort rx_fcs   fifofull
I (116808591) wifi: 1        55d45c0  55d45f1  0        13e5a    1372c24  1f04b4   0       
I (116808599) wifi: afull    rx_oth   tkiperr  samebm   ack_int  rts_int  cts_int  rifs_int
I (116808607) wifi: 0        bffd63   0        3b       553f     6c       2c8      0       
I (116808615) wifi: blockerr hoperr   txrts    txcts    track    trcts    trigger  txhung   panic   
I (116808624) wifi: 0        0        56b      0        8cdf     6c       41a      5fa      0       
I (116808633) wifi:
I (116808634) wifi: DIAG:
I (116808637) wifi: diag0    diag1    diag2    diag3    diag4    diag5    diag6   
I (116808644) wifi: e00002   0        fa00a08  0        1        a2082d   0       
I (116808651) wifi: diag7    diag8    diag9    diag10   diag12   diagsel 
I (116808658) wifi: 10       0        880000   40       0        0       
I (116808664) wifi:
I (116808666) wifi: Power save:
I (116808669) wifi: tbtt     beacon   timset   dtimset  tbttdrm  actto    txwake   null1    null0    bcndly   sleep    wake     sleept   waket    totalt   percent 
I (116808683) wifi: 903272   902538   0        0        0        9248     0        0        4318     822      0        0        0        0        0        nan   %
I (116808698) wifi:
I (116808699) wifi: HMAC RX:
I (116808702) wifi: ifx all      data     ctl    mgmt   bcn/probe
I (116808708) wifi: sta 1346326  103012   0      63137  51189
I (116808713) wifi: ap  0        0        0      0      0    
I (116808719) wifi:
I (116808721) wifi: ifx auth     assoc    deauth action amsdu
I (116808726) wifi: sta 148      134      134    12     0    
I (116808731) wifi: ap  0        0        0      0      0    
I (116808737) wifi:
I (116808739) wifi: HMAC TX:
I (116808741) wifi: sta_tx   ap_tx    psq_tx psq_mc psq_uc ps_uc1 reorder oos   amsdu
I (116808749) wifi: 5376     0        0      0      0      0       1    
```
The serial output shows that it's trying to reconnect and fails every time:
```
I (117139272) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (117139273) wifi:state: init -> auth (b0)
I (117139275) wifi:state: auth -> assoc (0)
I (117139285) wifi:state: assoc -> run (10)
I (117143334) wifi:state: run -> init (2a0)
I (117143334) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (117143334) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (117143340) app: WiFi connection failed, reason=2
I (117145392) app: WiFi connection failed, reason=205
```

`esp_get_free_heap_size` returns 42272.

I'm running ESP-IDF revision 526f682397a8cfb74698c601fd2c5b30e1433837.

YouDONG-ESP, 2021-01-26:
@itayperl , thanks, `I (116808501) wifi: 7  64  64    103014      102950     0        19399`    shows that the buffer to recieve eapol frame is full, going to check why it wasn't release.

YouDONG-ESP, 2021-01-28:
@itayperl please use cmd `xtensa-esp32-elf-gcc -v` under example path to see the tool chain version. If it's not the latest  `gcc version 8.4.0 (crosstool-NG esp-2020r3)` it may cause this issue.

itayperl, 2021-01-28:
Hi @YouDONG-ESP, that does not seem to be issue:
```
$ xtensa-esp32-elf-gcc -v |& grep 'gcc version'
gcc version 8.4.0 (crosstool-NG esp-2020r3)
```

itayperl, 2021-01-28:
Hi again,

A few things that might be relevant:

1. I see the following logs at the moment wifi disconnects (just before it starts reconnecting in a loop)
```
I (6559092) wifi:state: run -> init (2a0)
I (6559092) wifi:pm stop, total sleep time: 0 us / 6555834439 us

W (6559092) wifi:<ba-del>idx
W (6559093) wifi:<ba-del>idx
I (6559096) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (6559103) app: WiFi connection failed, reason=2
```

2. Note that I call `esp_wifi_set_ps(WIFI_PS_NONE)` at startup.

3. It seems that the leak only happens **after** wifi gets disconnected, while stuck in the reconnect loop. I tried calling the dump function after running for a while, but before the issue reproduced, and there were no allocated buffers at all.

YouDONG-ESP, 2021-01-29:
@itayperl Do you mean that` I (116808501) wifi: 7  64  64    103014      102950     0        19399   ` the third term is zero before before the issue reproduced? Could you describ the scenario of your station? Dose it run tcp/udp or others connection, dose it  keep sending/recieveing pakect? We need more detail to located the cause. We will probably give you a patch with some log to dump the packet blocked inside the issue buffer.

itayperl, 2021-01-29:
> @itayperl Do you mean that`I (116808501) wifi: 7 64 64 103014 102950 0 19399 ` the third term is zero before the issue reproduced?

Yes, it looks like this:

```
I (78777837) wifi: t  max alloc total_alloc total_free fail_fc  fail_oom
I (78777837) wifi: 1  32  1     4905        4904       0        0       
I (78777841) wifi: 2  32  0     2           2          0        0       
I (78777847) wifi: 3  32  0     3665        3665       0        0       
I (78777853) wifi: 4  32  0     0           0          0        0       
I (78777860) wifi: 5  0   0     4           4          0        0       
I (78777866) wifi: 6  0   0     0           0          0        0       
I (78777873) wifi: 7  64  0     105566      105566     0        0       
I (78777879) wifi: 8  64  0     764384      764384     0        0       
I (78777886) wifi: 9  0   0     4905        4937       0        0       
```
It seems that the time it takes for the issue to occur varies quite a lot. In a previous attempt it took less than 2 hours, and here it's been running for over 20 hours without issue.

> Could you describ the scenario of your station? Dose it run tcp/udp or others connection, dose it keep sending/recieveing pakect? 

The app runs a UDP server based on the asio library. The traffic is pretty low, I'd say <1000 requests per hour on average.
There's also an httpd server running but it gets close to zero traffic.

itayperl, 2021-01-29:
WiFi died just a few minutes after I posted that reply:
```
I (80145997) wifi: BUFFER:
I (80145998) wifi: t  max alloc total_alloc total_free fail_fc  fail_oom
I (80145998) wifi: 1  32  0     5038        5038       0        0       
I (80146002) wifi: 2  32  0     2           2          0        0       
I (80146008) wifi: 3  32  0     3729        3729       0        0       
I (80146015) wifi: 4  32  0     0           0          0        0       
I (80146021) wifi: 5  0   0     4           4          0        0       
I (80146027) wifi: 6  0   0     0           0          0        0       
I (80146034) wifi: 7  64  64    107614      107550     0        264     
I (80146040) wifi: 8  64  0     777725      777725     0        0       
I (80146047) wifi: 9  0   0     5038        5070       0        0       
I (80146053) wifi:
I (80146055) wifi: LMAC TX:
I (80146057) wifi: tx_all   lrc src age to 
I (80146061) wifi: 5302     0   2   0   0  
I (80146065) wifi: 0        0   0   0   0  
I (80146069) wifi: 3467     0   0   0   0  
I (80146073) wifi: 0        0   0   0   0  
I (80146077) wifi: 0        0   0   0   0  
I (80146081) wifi:
I (80146083) wifi: LMAC RX:
I (80146085) wifi: ampdu    mdpu     mgmt     ctrl     frag
I (80146090) wifi: 3630     103952   831421   4356     0   
I (80146096) wifi:
I (80146097) wifi: HW RX:
I (80146100) wifi: rx_suc   rx_end   full#1   full#2  
I (80146105) wifi: e6501    aa94113  1        0       
I (80146109) wifi:
I (80146111) wifi: Hardware:
I (80146114) wifi: rxbufbk  rxcckerr ofdmerr  rx_agc   rxsf     rx_abort rx_fcs   fifofull
I (80146122) wifi: 1        512ed56  512ed55  0        9ab7     364bb9c  589834   0       
I (80146130) wifi: afull    rx_oth   tkiperr  samebm   ack_int  rts_int  cts_int  rifs_int
I (80146138) wifi: 0        262ce0e  0        55       18ec     95       2c0      0       
I (80146146) wifi: blockerr hoperr   txrts    txcts    track    trcts    trigger  txhung   panic   
I (80146155) wifi: 0        0        76e      0        20       95       2c2      30f      0       
I (80146163) wifi:
I (80146165) wifi: DIAG:
I (80146167) wifi: diag0    diag1    diag2    diag3    diag4    diag5    diag6   
I (80146174) wifi: 500003   0        fa00405  20       2b700001 a60825   0       
I (80146182) wifi: diag7    diag8    diag9    diag10   diag12   diagsel 
I (80146188) wifi: 810      0        8a4000   40       0        0       
I (80146195) wifi:
I (80146196) wifi: Power save:
I (80146199) wifi: tbtt     beacon   timset   dtimset  tbttdrm  actto    txwake   null1    null0    bcndly   sleep    wake     sleept   waket    totalt   percent 
I (80146213) wifi: 782645   777721   0        0        0        8013     0        0        3723     923      0        0        0        0        0        nan   %
I (80146227) wifi:
I (80146229) wifi: HMAC RX:
I (80146232) wifi: ifx all      data     ctl    mgmt   bcn/probe
I (80146237) wifi: sta 885342   107610   0      56831  56826
I (80146243) wifi: ap  0        0        0      0      0    
I (80146248) wifi:
I (80146250) wifi: ifx auth     assoc    deauth action amsdu
I (80146255) wifi: sta 1        1        0      3      0    
I (80146261) wifi: ap  0        0        0      0      0    
I (80146266) wifi:
I (80146268) wifi: HMAC TX:
I (80146270) wifi: sta_tx   ap_tx    psq_tx psq_mc psq_uc ps_uc1 reorder oos   amsdu
I (80146278) wifi: 5038     0        0      0      0      0       1    
```
There's roughly 20 minutes between this dump and the previous one. It looks like something happens that makes that pool go from 0 to full in a fairly short time

itayperl, 2021-02-11:
@YouDONG-ESP, is there any update?
Thanks.

Pablopabota, 2021-02-24:
Hi, is there any update on this issue? 

I think I have a similar problem, after some time (it seems to be random) the ESP32 becomes unreachable through wifi, and stops working.

The only message I receive is **"wifi: f m auth"** not sure why, and it happens every once a couple days (not an easy to reproduce problem). The device does not respond to pings and based on my router it should still be connected.

I'm using an **ESP WROVER-B** module, programmed with the latest **4.2 stable** release IDF version. GCC is also **8.4.0 crosstool-NG esp-2020r3**.

The ESP32 is running a custom data logger sending data over MQTT through wifi.

Hope this is of help to find the problem,

Best regards.

espxiehang, 2021-04-14:
Hi @itayperl , @Pablopabota  Can you verify this issue with https://github.com/espressif/esp-idf/commit/1cb31e50943bb757966ca91ed7f4852692a5b0ed version?

ozanerturk, 2021-06-02:
I also have this problem, I use idf 4.2. Both STA and AP mode enabled. Device connectes to a wifi and  communicates with MQTT over network. I also run web server in my ESP. As @itayperl mentioned, After a day, ESP WIFI unable to re-connect, stops monitor log and returns esp_wifi_sta_get_ap_info 'true' for connection info and stucks.

espxiehang, 2021-07-21:
The problem is that buf is used to its maximum value?
 wifi: 7  64  64    107614      107550     0        264 
You can see from the log that there are 64 bufs for "type=7", so this type of buf cannot be requested.

AxelLin, 2021-07-21:
> The problem is that buf is used to its maximum value.
> wifi: 7 64 64 107614 107550 0 264
> You can see from the log that there are 64 bufs for "type=7", so this type of buf cannot be requested.

That was what @YouDONG-ESP  tried to figure out. https://github.com/espressif/esp-idf/issues/6230#issuecomment-767338472

AxelLin, 2021-11-13:
@espxiehang @YouDONG-ESP 
Any update?
Or any additional information required?

xueyunfei998, 2021-12-20:
hi  @itayperl 

Please upgrade to the latest version of 4.3 to test this problem.

You can help me to provide a simple reappearance demo to help me reproduce this problem locally, because I can't reproduce this problem with idf's example.

AxelLin, 2021-12-31:
Hi @itayperl  @Pablopabota  @ozanerturk 

Any chance to reproduce the issue with recent esp-idf version? 
Please share the updated esp-idf version again if you can reproduce the issue.

ozanerturk, 2021-12-31:
Hello Alex,
We went to defensive solutions to work around. We implemented retry
mechanism and if all retries fails, device restarts itself . And we also
implement daily restart to our devices just in case. After that we do not
face a wifi issue.
And watchdogs should be enabled in this scopes to be sure device is not
getting in to stuck

AxelLin, 2021-12-31:
> Hello Alex, We went to defensive solutions to work around. We implemented retry mechanism and if all retries fails, device restarts itself . 

Hi @ozanerturk

So it means this issue is still reproducable.
What is your esp-idf version (by running git describe --tags) ?

Fix the root cause is better than a workaround.
Any chance to verify latest esp-idf and provide log if still hit the issue?

AxelLin, 2022-02-15:
@itayperl 
Is this still an issue? Do you still hit the issue with recent esp-idf version? If yes, please also update the esp-idf version you use (git describe --tags).

ozanerturk, 2022-02-16:
for an update, we are still facing this issue from time to time. Retry and daily resets solve some of it but sometimes we just ask our client to turn the device off and on again. So an awful user experience. :/

ozanerturk, 2022-02-16:
```json
{         
          "Compile-Time": "15:08:54",
          "Compile-Date": "Sep 23 2021",
          "ESP-IDF Version": "v4.2.2",
          "WiFi-RSSI": -63,
          "Last Reset Reason": "POWERON",
          "Memory:": {
               "Free": 76168,
               "Total-alloc": 190272,
               "Min-free": 66492,
               "Largest-free": 74716
          },
          "Runtime(ms)": 13630,
          "Last-disconnect": "STARTUP"
     }
```
here is the version and last reset reason from the device on the field facing this issue

AxelLin, 2022-02-16:
@ozanerturk 
Can you provide device log (with verbose build) and wireshark log when the issue happened?

ozanerturk, 2022-02-16:
> @ozanerturk Can you provide device log (with verbose build) and wireshark log when the issue happened?

I will prep a test environment and collect the required information. I will share it here ASAP.

ozanerturk, 2022-03-07:
I've built a test environment, I plugged a raspi as a master computer directly monitoring the idf application and at the same time, collecting network logs with tshark. It's been a day. Nothing has happened yet.  I'll share the logs once I got them.

venulife9sys, 2022-03-18:
I am also facing same issue, after sending 10 to 15 mqtt messages WiFi is disconnecting. please help me to solve the problem.

ozanerturk, 2022-03-20:
dear @AxelLin  It took me a lot of time to set up a proper environment. Even I collected a lot of data. I can not reproduce the same problem with my home network and the other 2 setups network seem quite fine. In summary, I do not have a case yet. Once I have it, I'll let you know. For now, this is an only ghostbuster :/   I believe the best option, for now, is to close this issue. No need to confuse other developers. People would jump on that title even they had a basic configuration mistake. I wish I would come up with a solid case.

MaxwellAlan, 2022-03-23:
@venulife9sys Can you share the log when WiFi disconnect? And provide some more detail about how to reproduce your issue?

venulife9sys, 2022-03-24:
@MaxwellAlan please go through this log. this log is based on WiFi dongle connection. 

event->id=5,
Free Heap size : 168112 
Minimum free heap size : 160184  
I (54998270) Life9-IoT-Station: MQTT_EVENT_PUBLISHED, msg_id=14316
I (55019190) wifi:state: run -> init (4a0)
I (55019190) wifi:pm stop, total sleep time: 163900093 us / 239880677 us

I (55019190) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (55019200) Life9-IoT-Station: connect to the AP fail
E (55019200) TRANS_TCP: tcp_poll_read select error 113, errno = Software caused connection abort, fd = 54
E (55019210) MQTT_CLIENT: Poll read error: 0, aborting connection
event->id=2,
I (55019220) Life9-IoT-Station: MQTT_EVENT_DISCONNECTED
Free Heap size : 168988 
Minimum free heap size : 160184 .

Am using WiFi dongle as AP. esp32 station  device successfully connect to the AP, I can able to send maximum 10 -15 mqtt messages. and also i tried with WiFi router, Here disconnect failure case is very less, I can able to send 500 mqtt messages. here only possible to disconnect based on WiFi rssi strength.

MaxwellAlan, 2022-03-24:
hi @venulife9sys 
> I (55019190) wifi:state: run -> init (4a0)
error code: 4a0 means that ESP32 Station receive disassociation from AP and reason code is 4 means assoc expire.
Seems like that, when station connect to AP, but the dongle AP does not send EAPOL 1/4 pkt, and station does not send any pkt, and then disassociated by AP.
Questions:
- WiFi dongle as AP, which auth mode you are using? Can you try with Open mode and try WPA2 mode.

AxelLin, 2022-08-01:
Hi @Alvin1Zhang @MaxwellAlan @espxiehang @YouDONG-ESP 
This issue is marked as "Resolution: Done", could you share which commit fixed this issue?
Thanks.

MaxwellAlan, 2022-08-02:
Hi @AxelLin 

Actually, @itayperl's  issue we didn't find the root cause(do not have the fix commit) and feel free to reopen when @itayperl reply. 
While @venulife9sys's issue , the root cause was the WiFi Dongle.
Since this issue contains some different root cause issue, suggest close this and open new independent issue.

Thanks.

ciniml, 2024-04-30:
Hi.

I encountered a similar issue where the receive buffer of the WiFi driver was depleting on my end as well. 
Eventually, it was discovered that a pbuf leak was occurring in the handler of the udp_recv function in the upper lwIP callback API, and addressing this resolved the problem. 

To determine if the pbuf leak is the cause, it's advisable to enable CONFIG_LWIP_L2_TO_L3_COPY. 
If the pbuf leak is indeed the cause, enabling CONFIG_LWIP_L2_TO_L3_COPY will prevent the depletion of the WiFi driver's receive buffer (though it will eventually lead to heap depletion instead).

Hope this helps.