espressif/esp-idf #6232: Time/timezone handling via setenv/getlocaltime_r/strftime works in 3.x broken in 4.1

bill1davis, 2020-12-08:
The linux like timezone/timestamp handling code is broken in 4.1 after working in the 3.x tree.  See source at the bottom of the file. hello-world example was modified to set the time zone and then use localtime_r and strftime to format and display a known fixed epoch time_t.  The two os versions give different answers. 3.x matches expected behavior duplicated by compiling/running code below on ubuntu.

**3.3.2**
Hello world!
chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 3, 16MB external flash
2020-12-08 16:48:06

**4.1**
Hello world!
chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 3, 16MB external flash
2020-12-08 21:48:06

**Example slightly modified to run on Linux**
2020-12-08 16:48:06

**Code To Reproduce**
#include <stdio.h>
#include <stdlib.h>
#include <time.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

time_t   now = 0;
struct tm timeinfo = {0};
char buf[100];
`
void app_main(void)
{
    printf("Hello world!\n");

    /* Print chip information */
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    printf("chip with %d CPU cores, WiFi%s%s, ",
            chip_info.cores,
            (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
            (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

    printf("silicon revision %d, ", chip_info.revision);

    printf("%dMB %s flash\n", spi_flash_get_chip_size() / (1024 * 1024),
            (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    now = (time_t)1607464086; // 2020-12-08 16:48:06 EST / 2020-12-08 21:48:06 PM UTC
    setenv("TZ", "EST5EDT,M3.2.0,M11.1.0", 1);
    localtime_r(&now, &timeinfo);
    strftime(buf, sizeof(buf), "%Y-%m-%d %H:%M:%S", &timeinfo);
    printf("%s\n", buf);

    while (true) vTaskDelay(1000 / portTICK_PERIOD_MS);
}
`

bill1davis, 2020-12-08:
**Build Details**
Chip is ESP32-Wrover module v3.  Not a dev kit.
v4.1 and v3.3.2 esp idfs
4.1 tools : xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
3.3.2 tools : xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
    IDF version (run git describe --tags to find it): // v3.2-dev-1148-g96cd3b75c

OS: Linux (ubuntu)
Build System
3.3.2: make
4.1: idp.py

No IDE
Power is 5V charging circuitry + Battery

igrr, 2020-12-09:
@bill1davis does your program include the call to `tzset()` after setting the TZ variable?
e.g. in the SNTP the example:
https://github.com/espressif/esp-idf/blob/178b122c145c19e94ac896197a3a4a9d379cd618/examples/protocols/sntp/main/sntp_example_main.c#L90

bill1davis, 2020-12-09:
@igrr No.  The test program does not and it starts working if I add it.  The original program that demonstrated the problem does do tzset, but it must be finding a way not to call it in every code path or calling setenv with a bad timezone string.  Thanks for debugging my code for me! 8)

How does it work?  It seems like it must persist across boots to fool my test....

closed 2020-12-09 by bill1davis