espressif/esp-idf #6233: W5500 Ethernet driver bug

huybk213, 2020-12-09:
Dear all,

When i unplug Ethernet cable and replug cable, Ethernet link down event was not fired in esp-ethernet driver, so task watchdog timer was not feed on time.

It happens 100% every times i re-plug ethernet cable.

After take a look inside esp-idf, file **esp_eth_mac_w5500.c**,  i found the problems is in function **emac_w5500_transmit**, device never break the loop inside that function.

Here

```
    uint8_t status = 0;
    do {
        MAC_CHECK(w5500_read(emac, W5500_REG_SOCK_IR(0), &status, sizeof(status)) == ESP_OK, "read SOCK0 IR failed", err, ESP_FAIL);
    } while (!(status & W5500_SIR_SEND));  // **Bug here, device never exit loop**
```

Could you please take a look? If you think i am right, please fix it in next release of esp-idf.

Thanks
Full code
```
static esp_err_t emac_w5500_transmit(esp_eth_mac_t *mac, uint8_t *buf, uint32_t length)
{
    esp_err_t ret = ESP_OK;
    emac_w5500_t *emac = __containerof(mac, emac_w5500_t, parent);
    uint16_t offset = 0;

    // check if there're free memory to store this packet
    uint16_t free_size = 0;
    MAC_CHECK(w5500_get_tx_free_size(emac, &free_size) == ESP_OK, "get free size failed", err, ESP_FAIL);
    MAC_CHECK(length <= free_size, "free size (%d) < send length (%d)", err, ESP_FAIL, free_size, length);
    // get current write pointer
    MAC_CHECK(w5500_read(emac, W5500_REG_SOCK_TX_WR(0), &offset, sizeof(offset)) == ESP_OK, "read TX WR failed", err, ESP_FAIL);
    offset = __builtin_bswap16(offset);
    // copy data to tx memory
    MAC_CHECK(w5500_write_buffer(emac, buf, length, offset) == ESP_OK, "write frame failed", err, ESP_FAIL);
    // update write pointer
    offset += length;
    offset = __builtin_bswap16(offset);
    MAC_CHECK(w5500_write(emac, W5500_REG_SOCK_TX_WR(0), &offset, sizeof(offset)) == ESP_OK, "write TX WR failed", err, ESP_FAIL);
    // issue SEND command
    MAC_CHECK(w5500_send_command(emac, W5500_SCR_SEND, 100) == ESP_OK, "issue SEND command failed", err, ESP_FAIL);
#if 1   // original
    // pooling the TX done event
    uint8_t status = 0;
    do {
        MAC_CHECK(w5500_read(emac, W5500_REG_SOCK_IR(0), &status, sizeof(status)) == ESP_OK, "read SOCK0 IR failed", err, ESP_FAIL);
    } while (!(status & W5500_SIR_SEND));  // **Bug here, device never exit loop**
#else   // HuyTV : Dirty workaround
    // pooling the TX done event
    uint8_t status = 0;
    volatile int32_t count = 0;
    do {
        MAC_CHECK(w5500_read(emac, W5500_REG_SOCK_IR(0), &status, sizeof(status)) == ESP_OK, "read SOCK0 IR failed", err, ESP_FAIL);
    }
    while (!(status & W5500_SIR_SEND) && (count++ < 5000));
    if (count >= 5000 && !(status & W5500_SIR_SEND))
    {
        ESP_LOGE(TAG, "Failed cnt %d, status %d\r\n", count, status);       // HuyTV
        ret = ESP_FAIL;
        goto err;
    }
#endif

    // ESP_LOGE(TAG, "count %d\r\n", count);
    // clear the event bit
    status  = W5500_SIR_SEND;
    MAC_CHECK(w5500_write(emac, W5500_REG_SOCK_IR(0), &status, sizeof(status)) == ESP_OK, "write SOCK0 IR failed", err, ESP_FAIL);

err:
    return ret;
}
```

zod6, 2021-01-27:
Having same issue with latest master v4.3-dev-2586-g526f68239.
ETH Wiz click and ESP-WROOM-32 devkit configured as follows:
```
(1) SPI Host Number
(18) SPI SCLK GPIO number
(23) SPI MOSI GPIO number
(19) SPI MISO GPIO number
(5) SPI CS GPIO number
(16) SPI clock speed (MHz)
(4) Interrupt GPIO number
(16) PHY Reset GPIO number
(1) PHY Address
```

examples/ethernet/basic gets ip and I can ping it, but when cable is disconnected it stays in loop until watchdog restarts it.
This happens every time. Tried exactly same pinout with DM9051 and it detected connections/disconnections properly.
```
I (3874) eth_example: Ethernet Got IP Address
I (3874) eth_example: ~~~~~~~~~~~
I (3874) eth_example: ETHIP:192.168.1.211
I (3884) eth_example: ETHMASK:255.255.255.0
I (3884) eth_example: ETHGW:192.168.1.1
I (3894) eth_example: ~~~~~~~~~~~
E (13394) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (13394) task_wdt:  - IDLE (CPU 1)
E (13394) task_wdt: Tasks currently running:
E (13394) task_wdt: CPU 0: IDLE
E (13394) task_wdt: CPU 1: tiT
E (13394) task_wdt: Print CPU 0 (current core) backtrace
Backtrace:0x400e6798:0x3FFB0650 0x4008195D:0x3FFB0670 0x401014EF:0x3FFB7950 0x400E64EB:0x3FFB7970 0x40085B05:0x3FFB7990 0x40087BB5:0x3FFB79B0
E (13394) task_wdt: Print CPU 1 backtrace
```
When turning on LWIP debugging it stays in "ethernet_output: sending packet 0x3ffaed3c" and after a while whatchdog intervenes.

markus-fischbacher, 2021-02-17:
I'm dealing with the same topic.
All working except disconnect and detecting link down.
Reconnect does not work either.
Would be nice if that gets fixed soon.

huybk213, 2021-02-17:
@markus-fischbacher  You can use my dirty fix comment in first post.

markus-fischbacher, 2021-02-17:
@huybk213 thanks, I'll have a look!

markus-fischbacher, 2021-02-17:
The interrupt register is 0 if the Ethernet link is down.
The following code is working for me:
```
    // pooling the TX done event
    uint8_t status = 0;
    do {
        MAC_CHECK(w5500_read(emac, W5500_REG_SOCK_IR(0), &status, sizeof(status)) == ESP_OK, "read SOCK0 IR failed", err, ESP_FAIL);
    } while (!(status & W5500_SIR_SEND) && status != 0);
    if ((status & W5500_SIR_SEND) == W5500_SIR_SEND)
    {
        // just clear the event bit
        status |= W5500_SIR_SEND;
    }
    else
    {
        // do not clear other bits, maybe they are needed on other places
        status = 0;
    }
```

markus-fischbacher, 2021-02-17:
Unfortunately it's not really stable.
```
I (96407) eth_example: Ethernet Link Down
I (98407) eth_example: Ethernet Link Up
I (98407) eth_example: Ethernet HW Addr 02:00:00:12:34:56
I (99377) esp_netif_handlers: eth ip: 192.168.8.150, mask: 255.255.255.0, gw: 192.168.8.1
I (99377) eth_example: Ethernet Got IP Address
I (99377) eth_example: ~~~~~~~~~~~
I (99377) eth_example: ETHIP:192.168.8.150
I (99387) eth_example: ETHMASK:255.255.255.0
I (99387) eth_example: ETHGW:192.168.8.1
I (99397) eth_example: ~~~~~~~~~~~
E (168427) w5500-mac: emac_w5500_transmit(510): free size (0) < send length (42)
I (168427) eth_example: Ethernet Link Down
E (206417) w5500-mac: emac_w5500_transmit(510): free size (0) < send length (350)
I (206417) eth_example: Ethernet Link Up
I (206417) eth_example: Ethernet HW Addr 02:00:00:12:34:56
E (206877) w5500-mac: emac_w5500_transmit(510): free size (0) < send length (350)
E (207877) w5500-mac: emac_w5500_transmit(510): free size (0) < send length (350)
E (209877) w5500-mac: emac_w5500_transmit(510): free size (0) < send length (350)
I (212407) eth_example: Ethernet Link Down
E (214407) w5500-mac: emac_w5500_transmit(510): free size (192) < send length (350)
I (214407) eth_example: Ethernet Link Up
I (214407) eth_example: Ethernet HW Addr 02:00:00:12:34:56
E (214877) w5500-mac: emac_w5500_transmit(510): free size (0) < send length (350)
E (215877) w5500-mac: emac_w5500_transmit(510): free size (0) < send length (350)
```
I think, I'll change to DM9051.
It to risky for production use.

david-cermak, 2021-04-08:
Hi all and sorry for the late reply,

Thanks for reporting this issue and suggesting fixes! I've combined both methods into this patch 
[esp_eth-Fix-w5500-to-break-from-potentially-infinite.patch.txt](https://github.com/espressif/esp-idf/files/6277433/esp_eth-Fix-w5500-to-break-from-potentially-infinite.patch.txt)
The original *dirty fix* by @huybk213 is I think okay, as it just adds a retry timeout, the one by @markus-fischbacher could be potentially dangerous, because the `SOCKn_IR` could be zero event if link up (to cause packet drops on slower networks)

As for the stack overflow, I think it's related to your debug logs in the code. The default stack size of the timer task is 2048, which has about 300+ watermark (based on my measurements). It could be also related to the linked issue, e.g. by incorrect reads of the offset register. But, I am going to test the driver thoroughly first, seems there might be other issues.

zod6, 2021-04-08:
quick hint what else to test: two flood ping's (ping -f) to w5500 caused esp32 to reset after ~10sec. Unfortunately I'm not having hw near at moment to retest.

david-cermak, 2021-04-09:
It looks like this fix and the one from #6579 are quite efficient, at least for the flood pings, running for half a day already.

There's still one issue of potentially missing GPIO interrupts, happening for short, quickly repeated payloads (causes the device to stop responding, not to crash). Here is a quick patch to recover from it:
[esp_eth-Recover-the-w5500-driver-from-missed-io-inte.patch.txt](https://github.com/espressif/esp-idf/files/6288104/esp_eth-Recover-the-w5500-driver-from-missed-io-inte.patch.txt)

huybk213, 2021-04-14:
Close

closed 2021-04-14 by huybk213