espressif/esp-idf #6234: esp_http_client to support chunked request

galah92, 2020-12-09:
**Is your feature request related to a problem? Please describe.**
I'm trying to steam data over a single HTTP request using chunked requests (PUT request).

**Describe the solution you'd like**
Current HTTP-client-API has `esp_http_client_open()` which takes the total size of the request. This prevents from defining a chunked requests and `esp_http_client_write()`ing as much as I want.

Alvin1Zhang, 2020-12-10:
Thanks for raising this feature request.

galah92, 2021-01-10:
Just to elaborate - it seems that using a negative `write_len` will add `chunked` "Transfer-Encoding" header, but will also override the method to be `POST`. My use-case is for chunked requests with `PUT` method.

shubhamkulkarni97, 2021-03-24:
Hi @galah92,

Sorry for slow turnaround. I got the issue where HTTP method gets overwritten to POST while sending chunked request.

Can you try applying patch attached below and check if it works for your use case?

[0001-esp_http_client-Set-HTTP-method-to-POST-only-if-meth.patch.zip](https://github.com/espressif/esp-idf/files/6195671/0001-esp_http_client-Set-HTTP-method-to-POST-only-if-meth.patch.zip)

galah92, 2021-03-26:
Hi @shubhamkulkarni97. The patch is working, thanks.
Can this be merged to v4.3?

closed 2021-04-01 by espressif-bot

galah92, 2021-06-28:
@shubhamkulkarni97 will this be included as part of v4.4? Is there an ETA for that?

galah92, 2021-07-25:
> @shubhamkulkarni97 @Alvin1Zhang will this be included as part of v4.4? Is there an ETA for that?

Any update? Thanks

galah92, 2021-08-17:
@shubhamkulkarni97 @Alvin1Zhang bumping it again. This bugfix is of great importance for us, thank you.

shubhamkulkarni97, 2021-08-19:
Hi @galah92,

This fix will be a part of v4.4 release. We are aiming to release IDF v4.4 by end of 2021.