espressif/esp-idf #6235: Increase the size of the size of BLE GAP

ftoyon, 2020-12-09:
We using esp IDF 4.1

**Is your feature request related to a problem? Please describe.**

We need more than 12 addresses in the BLE GAP whitelist while BLE scan.

**Describe the solution you'd like**

Adding a field in `esp_ble_scan_params_t` to set the size limit.

Alvin1Zhang, 2020-12-10:
Thanks for raising this feature request.

ftoyon, 2021-01-07:
Is there a chance to have this feature.

xiewenxiang, 2021-01-08:
Hi @ftoyon , 

If we put this configuration in menuconfig， does it can meet your needs?

ftoyon, 2021-01-08:
Hi @xiewenxiang ,

It's a possibility.

It would be a good idea to put this limit of 12 in the documentation.

Emill, 2021-03-12:
I also think the limit of 12 is quite low, especially compared to other chips. Each entry is just an address in RAM of 49 bits. Should be possible to configure this to the application's need, up to 255.

xiewenxiang, 2021-03-16:
@Emill 
Currently, it can only support up to 12.

ftoyon, 2021-06-30:
Hi ! 
Are there any plans to support this issue in a nearer version?

xiewenxiang, 2021-07-01:
@ftoyon 
OK, we will fix it as soon as possible, and give you feedback when it is ready.