espressif/esp-idf #6236: v4.2 secure boot + secure encryption doesn't boot

skattarallo, 2020-12-09:
## Environment
- Development Kit:   None
- Kit version N/A
- Module or chip used: esp32 eco v3
- IDF version (run ``git describe --tags`` to find it): v4.2
- Build System:        idf.py
- Compiler version :  xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         Battery

## Problem Description

We are using v4.2 tag from https://github.com/espressif/esp-idf/releases/tag/v4.2 with secure boot v2 and flash encryption (see attached below our sdkconfig.defaults)

To flash the device we sign v2 offline the bootloader and the app:

```
espsecure.py sign_data --version 2 --keyfile my_secure_boot_signing_key.pem --output bootloader_signed.bin bootloader.bin

espsecure.py sign_data --version 2 --keyfile my_secure_boot_signing_key.pem --output app_signed.bin app.bin
```

```
esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 460800 --before default_reset --after no_reset --no-stub write_flash --flash_mode dio --flash_freq 40m --flash_size 4MB  -u 0x1000 bootloader_signed.bin

esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 460800 --before default_reset --after no_reset --no-stub write_flash --flash_mode dio --flash_freq 40m --flash_size keep 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 app_signed.bin
```

However it seems the app doesn't boot.

If we check the efuses now:

```
espefuse.py summary

Connecting....
Detecting chip type... ESP32
espefuse.py v3.0
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):              BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                       Voltage reference calibration                      = 1093 R/- (0b10001)
 
Config fuses:
XPD_SDIO_FORCE (BLOCK0):                 Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                   If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                  If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                     8MHz clock freq override                           = 53 R/W (0x35)
SPI_PAD_CONFIG_CLK (BLOCK0):             Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):               Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):               Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):              Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):             Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):              Disable SDIO host                                  = False R/W (0b0)
 
Efuse fuses:
WR_DIS (BLOCK0):                         Efuse write disable mask                           = 385 R/W (0x0181)
RD_DIS (BLOCK0):                         Efuse read disable mask                            = 1 R/- (0x1)
CODING_SCHEME (BLOCK0):                  Efuse variable block length scheme                
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False R/W (0b0)
 
Identity fuses:
MAC (BLOCK0):                            Factory MAC Address                               
   = 24:6f:28:8f:67:54 (CRC 0xcb OK) R/W 
MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 203 R/W (0xcb)
CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = True R/W (0b1)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 0 R/W (0b000)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)
 
Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 0 R/W (0b0000000)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = True R/W (0b1)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = True R/W (0b1)
BLOCK1 (BLOCK1):                         Flash encryption key                              
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/- 
BLOCK2 (BLOCK2):                         Secure boot key                                   
   = 60 f4 ff 0d 18 1b 1f 20 2a 2c 2e 36 3b 64 96 a4 ae bf 72 f7 f8 fe 84 95 c9 d1 d7 53 4f 50 3f 42 R/- 
BLOCK3 (BLOCK3):                         Variable Block 3                                  
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
 
Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```

(BLOCK2 key replaced from real one)

It seems the public key is there. At this point it seems I can't reflash the bootloader to enable logging (reflashing seems to work but i see no logging in `idf.py monitor`)

Are we doing something wrong?

Can I recover the device? If yes, what (and how) can I flash on it now, just the app or both the app and the bootloader?

flashing always seems to work but our app doesn't boot and we see nothing in the monitor. Admittedly we have logging disabled in the initial flashing but we have tried to enabled since at no avail, hence we assume the bootloader can't be changed.

Please advise.

Thanks

## Other items if possible

- [x] sdkconfig file [sdkconfig.defaults.zip](https://github.com/espressif/esp-idf/files/5667156/sdkconfig.defaults.zip)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-12-10:
Thanks for the detailed report and letting us know, we will look into.

projectgus, 2020-12-10:
Hi @skattarallo,

Yes, thanks for providing such detailed information. I tried to reproduce, but was able to use your sdkconfig and the ESP-IDF v4.2 release to boot successfully and confirm flash encryption and secure boot had enabled.

I agree with your assessment that something has gone wrong in the first boot and because the log level is set to None, you haven't seen any error message. The efuse state looks like Secure Boot has enabled, and Flash Encryption has started enabling (the key and the configuration eFuses are burned) but hasn't finished (`FLASH_CRYPT_CNT` efuse is still zero).

A common scenario that results in these exact symptoms is that the ESP32 was reset before it finished self-encrypting on the first boot (if your device has a development-board-style automatic reset circuit, this can happen when you connect to it to monitor the output.) Is it possible this happened? If so, try reflashing everything (partition table, app, bootloader, etc) without resetting, reset the device once, then wait a couple of minutes for it to self-encrypt before trying to interact with it again.

Otherwise, with Secure Boot V2 it shouldn't be a problem to flash a new (signed) bootloader with a higher log level configured, as long as flash encryption remains disabled in the hardware (ie FLASH_CRYPT_CNT is zero, once it becomes 1 then flash encryption is enabled and you can't reflash via serial - but if you get to that point then it should be working). There are a couple of things to keep in mind:

* The sdkconfig has the [bootloader log level set to None](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-bootloader-log-level) and the  [UART console disabled](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-esp-console-uart) - the latter option controls both bootloader and app. Would need to set both back in order to see any output.
* Once output is re-enabled, the bootloader may no longer fit with the default partition table offset 0x8000. [You can read more about this problem here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/security/secure-boot-v1.html#secure-boot-bootloader-size).

If none of this seems to be helping, can you please share the output you see on the UART when trying to boot the current bootloader?

skattarallo, 2020-12-10:
We haven't seen any logging even trying to flash a new signed bootloader/app/partition table etc with the uart set and logging in the boot loader and app enabled to info. For what is worth we GPIO15 pulled down so we have disabled rom logging and we can't change this. Flashing a new device with v4.2 without secure boot/flash works for what is worth (and we can see logging if we enable it via uart/+info)

Should we flash the bootloader and app in two separate esptool write flashes or in one command? Could it be that when the app is flashed it does a --before default_reset and that's wrong?

KaeLL, 2020-12-10:
Btw, since you touched on this subject, which pin should be strapped to disable ROM log? 13 or 15? Docs say 13, I've seen people saying 15. Which one is it?

projectgus, 2020-12-15:
Hi @skattarallo ,

Sorry for the slow reply

> For what is worth we GPIO15 pulled down so we have disabled rom logging and we can't change this. Flashing a new device with v4.2 without secure boot/flash works for what is worth (and we can see logging if we enable it via uart/+info)

Ah OK, in this case yes that explains that.

It seems like someone from your company may have contacted us internally, I'll see if it's possible to get your exact bootloader binary for comparison.

> Should we flash the bootloader and app in two separate esptool write flashes or in one command? Could it be that when the app is flashed it does a --before default_reset and that's wrong?

It should be safe to do it in either, but one is generally easier and safer. If doing in two passes, it's safest to flash the other files before the bootloader just in case - same sequence that you're doing now.

The "--before" reset shouldn't matter, as long as the option "--after no_reset" is passed then the ESP32 won't reset immediately after flashing, and esptool should be able to connect again.

> since the subject has been brought up, which pin should be strapped to disable ROM log? 13 or 15? Docs say 13, I've seen people saying 15. Which one is it?

@KaeLL The pin is MTDO which is GPIO15. Where did you see pin 13 in the docs?