espressif/esp-idf #6238: nimble: sending large buffers from Android into ESP fails

DiegoActionItem, 2020-12-10:
I have been sending large BLE packets using NimBLE/ESP4.2 from Android. The packets basically a protobuff data, and the data sent is generally around 200 bytes (201 - 230). If we reach those sizes - the Android App sees error 17 when writing the characteristic.

I traced it to this line:
https://github.com/apache/mynewt-nimble/blob/4c2a72064dba17f10cd1143b9e505e33a429b9eb/apps/btshell/src/gatt_svr.c#L315
https://github.com/apache/mynewt-nimble/blob/d11729a60aaee277a9516d93d8fb8fe6be78c5ee/nimble/host/include/host/ble_att.h#L58

What happens is that the BLE system cannot allocate memory for it. If I am not mistaking - the code failing is this:
https://github.com/apache/mynewt-nimble/blob/4c2a72064dba17f10cd1143b9e505e33a429b9eb/porting/nimble/src/os_mbuf.c#L276 

Is this a known limitation of NimBLE - how to overcome it?

dhrishi, 2020-12-11:
Hi @DiegoActionItem Thanks for reporting the issue. Can you please provide additional details as per the issue template:

1. Which example are you trying out? If it is btshell, then have you ported that on ESP32? The source code of your example or a patch based on any of the existing ESP-IDF NimBLE example would be really helpful.
2. What is the ESP-IDF version (release/ branch / GIT head) that you are using?
3. Assuming that ESP32 is BLE peripheral (from the description above), what is the MTU value that is negotiated after the connection?
4. Does this fail during the first iteration of sending large data itself or after a few iterations?

DiegoActionItem, 2020-12-12:
1. This is production code, based on https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/bleprph/main/gatt_svr.c - we have this gatt server, which read the stream and de-serialize this using protobuff. Not really relevant - since my code is not called at all.

2. 4.2 - rc1. But can reproduce with beta as well.
3. Dunno, I will update this when I have more info :)
4. On the first iteration of sending data.

sweetymhaiske, 2020-12-14:
Hi @DiegoActionItem, 

1. **Solution for android app error 17**

- Either increase MTU value.
- Or increase MYS1_BLOCK_COUNT from menuconfig ( `make menuconfig --> Component config --> Bluetooth --> NImBLE Options --> MSYS_1 Block Count` ) 

2. The error you pointed out https://github.com/apache/mynewt-nimble/blob/4c2a72064dba17f10cd1143b9e505e33a429b9eb/apps/btshell/src/gatt_svr.c#L315 => Here, The min_len or max_len value should be configured by the user depending upon use case.

DiegoActionItem, 2020-12-16:
Thanks, increasing the block count helped.

closed 2020-12-16 by DiegoActionItem